react-native-camera/react-native-camera #2478: Use all featurse of device's camera

valdman40, 2019-09-10:
Is there a way to let RNcamera use all of the device's features for camera? 
I want the user to have all the features his phone capable of

And if there isn't a proper way of doing that, I'd like to know how you implented features such as pinch zoom in (tried using this code [1282](https://github.com/react-native-community/react-native-camera/issues/1282#issuecomment-392731643) but it is really slow and functions poorly), burst mode, resolution and so..

fabriziobertoglio1987, 2019-10-15:
I also need to implement [1282](https://github.com/react-native-community/react-native-camera/issues/1282#issuecomment-392731643) and reading that it functions poorly makes me very sad, as the zoom feature is very important for my app.

How can we build a Camera App without zoom functionalities? I am pretty sure that Instangram has perfectly working zoom functionalities... :-(

fabriziobertoglio1987, 2019-10-15:
Maybe not using react-native-camera and instead [just writing an intent to open the default camera application](https://stackoverflow.com/a/2729382/7295772)

[The following app](https://synced.to/) uses this solution, all the camera functionalities work perfectly.

Could you re-open the issue as it seems to not be solved? 
Developers may need to review all the open issues to estimate the project deadlines.
Thanks

[image]

valdman40, 2019-10-16:
> Maybe not using react-native-camera and instead [just writing an intent to open the default camera application](https://stackoverflow.com/a/2729382/7295772)
> 
> [The following app](https://synced.to/) uses this solution, all the camera functionalities work perfectly.
> 
> [image]

valdman40, 2019-10-16:
> Maybe not using react-native-camera and instead [just writing an intent to open the default camera application](https://stackoverflow.com/a/2729382/7295772)
> 
> [The following app](https://synced.to/) uses this solution, all the camera functionalities work perfectly.
> 
> [image]

the solution you sent is written with android studio.
how can i use this intent in react native?

fabriziobertoglio1987, 2019-10-17:
You can read [this example](https://github.com/petterh/react-native-android-activity) on how to create and trigger a native android activity or ios view within a react-native app

But I don't believe that it is a good solution to trigger an intent to use the default camera application, I believe we should work intensively with the Native Camera api.
 
I believe `Stackoverflow` would be a more appropriate place for all Github related discussions, while here developers should only come to download the sourcecode and read the instructions.

Many talented developers will never review issues on github, because it is just too messy and disorganized and there is no incentive to solve a problem on Github. 

So maybe the problem is not the missing functionalities, but how the OpenSource project is handled, but how it is organized.

If we had a clear example on Stackoverflow of how to implement the zoom functionalities with react-native-camera and also a showcase application or gif that displays how smooth/laggy it is, it would be much better then arguing and trying to put together pieces of codes from differents posts.

Then we can argue if we need or not an integrated zoom functionality in the react-native-camera api.

fabriziobertoglio1987, 2019-10-18:
Thanks @cristianoccazinsp for your [very complete and clear solution](https://github.com/react-native-community/react-native-camera/issues/1282#issuecomment-543188018). I look forward to contribute to this amazing project

closed 2019-12-24 by stale[bot]

babitakapoor112, 2020-08-19:
Is there a way to use your mobile native camera in react-native application?