react-native-camera/react-native-camera #2485: Error takePictureAsync code E_CAMERA_BAD_VIEWTAG

BrayanArrieta, 2019-09-13:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android: 9
- iOS: N/A
- react-native-camera: 3.4.0
- react-native: 0.59.10
- react: 16.8.3
  

**Description/Current Behaviour**
I cannot take a picture was produced an error in the click event for the camera. In the logs I can see this. The error was produced here.
```
const options = { quality: 0.5, base64: true };
const data = await this.camera.takePictureAsync(options);
``` 
Logs
```
09-13 08:25:06.657  6321  6356 I ReactNativeJS: 'Picture error:', { [Error: takePicture failed]
09-13 08:25:06.657  6321  6356 I ReactNativeJS:   framesToPop: 1,
09-13 08:25:06.657  6321  6356 I ReactNativeJS:   nativeStackAndroid: [],
09-13 08:25:06.657  6321  6356 I ReactNativeJS:   userInfo: null,
09-13 08:25:06.657  6321  6356 I ReactNativeJS:   code: 'E_CAMERA_BAD_VIEWTAG',
09-13 08:25:06.657  6321  6356 I ReactNativeJS:   line: 2108,
09-13 08:25:06.657  6321  6356 I ReactNativeJS:   column: 26,
09-13 08:25:06.657  6321  6356 I ReactNativeJS:   sourceURL: 'http://10.0.2.2:8081/index.delta?platform=android&dev=true&minify=false' }
```

**Expected Behaviour**
Take a photo

**Steps to Reproduce**
Just the steps in the documentation. I wasn't added anything more.

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2019-09-19:
The rejected promise should also have the error, do you have that one? You should definitely always  try/catch that await takePictureAsync and handle any errors appropriately.

codervss, 2019-10-10:
@BrayanArrieta Have you come up with any solution?

yosagar, 2019-10-15:
@BrayanArrieta @codervss Any solution for this error ? I am getting same error.

tsVestor, 2019-10-15:
@BrayanArrieta  I'm also getting  "[Error: takePicture failed]"

cristianoccazinsp, 2019-10-15:
The error seems to be coming from here: https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/CameraModule.java#L277

Have you tried with the latest version? Do you have any additional error log from the JS side? The error message only says that the internal call to `takePicture` has failed, but there's no stack trace to tell whether it failed on the Camera1 call, the view call, etc.

yosagar, 2019-10-15:
@cristianoccazinsp the stringified error object looks like
```
{
 "framesToPop": 1,
 "nativeStackAndroid": [],
 "userInfo": null,
 "message": "takePicture failed",
 "code": "E_CAMERA_BAD_VIEWTAG",
 "line": 2189,
 "column": 45,
 "sourceURL": "http://192.168.31.164:8081/index.bundle?platform=android&dev=true&minify=false"
}
```

Jarrio, 2019-10-25:
Okay guys I ran into this issue today and literally spent all day stuck on this. After exhausting all options I ended up just experimenting with things. Here's what I did to figure this out for my situation:

**The Thought Process** (might help anyone debug issues on there own in the future I dunno)
1. I opened android studio to see if that was reporting any thing different. I was also getting this unhelpful `E_CAMERA_BAD_VIEWTAG` error with no real hint
2. Android studio logs showed the following error `Cannot take picture without preview enabled`
3. I couldn't find anything related to this error either a part from a stale issue that got auto-closed
4. Started searching all over the rn docs for any hint to what "preview" could be related to

**My Solution** 
5. On the `RNCamera` page there's a style called `preview` I kid you not this is where things are going.
6. Adding the following to the camera component fixed the issue: 
```haxe
{
	flex: 1,
	justifyContent: "flex-end",
	alignItems: "center",
}
```

cristianoccazinsp, 2019-10-25:
`Cannot take picture without preview enabled` should mean just that, your camera preview wasn't running. Wasn't that the issue? Was the issue really a style issue?

Jarrio, 2019-10-25:
For me, yes. That error gives a fairly clear idea of what to do, it's directly telling you that "hey, you need to configure a viewfinder somewhere" I had thought it was some kind of prop and it might be managed by the component or the component would inherit the style from the parent view.

The main issue was the `E_CAMERA_BAD_VIEWTAG` and lack of information on google about it. But I guess the error kind of makes sense in hindsight

closed 2019-12-31 by stale[bot]