react-native-camera/react-native-camera #2486: Incompatibilite with version of firebase

RaulCorreia, 2019-09-13:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 8.0
- iOS: 
- react-native-camera: 3.4.0
- react-native: 0.59.10
- react: 16.8.3
  

**Description/Current Behaviour**

When I compile a release, i'm get the incompatibility error with some libraries, this only happens when I add the RNCamera library, but in debug the app is installed normally.

`> The library com.google.firebase:firebase-iid is being requested by various other libraries at [[18.0.0,18.0.0]], but resolves to 17.0.3. Disable the plugin and check your dependencies tree using ./gradlew :app:dependencies.`

**Expected Behaviour**

when compiling a release, generate an apk

**Steps to Reproduce**

cd android && ./gradlew assemblereleasek

**Config used**
in app/build.gradle

```
defaultConfig {
   ...
  missingDimensionStrategy 'react-native-camera', 'general'
}
```

and i use 

```
implementation "com.google.firebase:firebase-core:16.0.9"
implementation "com.google.firebase:firebase-messaging:18.0.0"
implementation "com.google.android.gms:play-services-base:16.1.0"
```

in build.gradle i add
```
googlePlayServicesVisionVersion = "17.0.2"
```

sibelius, 2019-09-13:
you can fix this with gradle resolutions

RaulCorreia, 2019-09-17:
How could i do this? I've been researching and haven't found anything that could help me, I've tried everything I found in the gradle documentation, could you give me a light to solve this?

bobbyrinaldy, 2019-10-02:
is there is solution about this ?
im getting same error too here

closed 2019-12-21 by stale[bot]