react-native-camera/react-native-camera #2489: Receiving an audible push notification causes audio to stop recording

xHeinrich, 2019-09-17:
# Bug Report

**To Do First**

- [yes ] Did you try latest release?
- [yes ] Did you try master?
- [ yes] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

12.4.*

**Description/Current Behaviour**
When a recording session is active and an audible push notification is received the videos audio cuts out and does not return

**Expected Behaviour**

Audio should continue recording

**Steps to Reproduce**

Begin recording video
Receive audible push notification

**Additionals**

xHeinrich, 2019-10-01:
It appears this is caused by this patch - https://github.com/react-native-community/react-native-camera/pull/2207

MateusAndrade, 2019-10-01:
HI @xHeinrich did you tested this without the changes at ```isRecordingInterrupted```?

xHeinrich, 2019-10-01:
@MateusAndrade yes I did, the result was that the audio cuts out when the notification comes in then doesnt resume when the notification is gone

xHeinrich, 2019-10-01:
The issue and possible fixes:
https://stackoverflow.com/questions/19559238/resume-audio-recording-after-incoming-call

xHeinrich, 2019-10-01:
I think i found the core of the issue, the code in the above patch basically only works for phone calls or app backgrounding and cuts off the session and does not resume it. The resulting file will only contains the video from when the app got backgrounded and before. 

http://www.gdcl.co.uk/2013/03/22/iPhone-Resume-Recording.html

```        

[[NSNotificationCenter defaultCenter] addObserver:self
                                                 selector:@selector(bridgeDidForeground:)
                                                     name:UIApplicationDidBecomeActiveNotification
                                                   object:nil];

        [[NSNotificationCenter defaultCenter] addObserver:self
                                                 selector:@selector(bridgeDidBackground:)
                                                     name:UIApplicationWillResignActiveNotification
                                                   object:nil];

```

I believe this code will handle when notifications get foregrounded.

MateusAndrade, 2019-10-02:
WOW! Thanks for helping with that info @xHeinrich. Would you mind to send a PR with this? Contributions are always welcome. :smiley:

xHeinrich, 2019-10-15:
I've identified another issue around this, playing a sound(music, chime or something similar) and attempting to call recordAsync can cause the file output to have no audio.

closed 2019-12-21 by stale[bot]