react-native-camera/react-native-camera #2490: Failed to list versions for org.webkit:android-jsc.

oqnunes, 2019-09-17:
Hi everyone.

I'm trying to install react-native-camera in my project, so i follow all steps from but when i set the command react-native run-android i get this error. 

`* What went wrong:
Could not determine the dependencies of task ':app:preDebugBuild'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve org.webkit:android-jsc:+.
     Required by:
         project :app
      > Failed to list versions for org.webkit:android-jsc.
         > Unable to load Maven meta-data from https://jitpack.io/org/webkit/android-jsc/maven-metadata.xml.
            > Could not get resource 'https://jitpack.io/org/webkit/android-jsc/maven-metadata.xml'.
               > Could not GET 'https://jitpack.io/org/webkit/android-jsc/maven-metadata.xml'. Received status code 522 from server: Origin Connection Time-out`

So anyone can help me? i've been two days looking for some resolution but without success.

wouterds, 2019-09-18:
Out of nowhere (no configuration changes, no new packages) we started seeing the exact same issue since yesterday as well in our React Native 0.60.5 project. I've been trying to build for 24 hours.

```
FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:lintVitalRegularRelease'.
> Could not resolve all task dependencies for configuration ':app:regularReleaseRuntimeClasspath'.
   > Could not resolve org.webkit:android-jsc:+.
     Required by:
         project :app
      > Failed to list versions for org.webkit:android-jsc.
         > Unable to load Maven meta-data from https://jitpack.io/org/webkit/android-jsc/maven-metadata.xml.
            > Could not get resource 'https://jitpack.io/org/webkit/android-jsc/maven-metadata.xml'.
               > Could not GET 'https://jitpack.io/org/webkit/android-jsc/maven-metadata.xml'. Received status code 522 from server: Origin Connection Time-out
   > Could not resolve com.github.droibit.customtabslauncher:launcher:1.0.8.
     Required by:
         project :app > project :react-native-custom-tabs
      > Skipped due to earlier error
```

yurkagon, 2019-09-18:
Got the same issue with `react-native-image-crop-picker`, fixed by

`maven { url "https://www.jitpack.io" }`
(Added `www`)

I dont know why, but it works

oqnunes, 2019-09-18:
> Got the same issue with `react-native-image-crop-picker`, fixed by
> 
> `maven { url "https://www.jitpack.io" }`
> (Added `www`)
> 
> I dont know why, but it works

I'm trying to read barcode in my app, does react-native-image-crop-picker can do that? i dont try yet.

oqnunes, 2019-09-18:
@yurkagon do you know if barcode work on in react-native-image-crop-picker? i'm trying to read a barcode...

yurkagon, 2019-09-18:
> @yurkagon do you know if barcode work on in react-native-image-crop-picker? i'm trying to read a barcode...

that package is for picking images from files or from camera. I guess its not that case you need. But it got the same issue with `jitpack.io`, so I've just said how to fix the build process

wouterds, 2019-09-20:
@yurkagon: That works, thanks!

rockia, 2019-09-23:
I also had this problem with React Native 0.60.5. I submitted a PR to amend the official installation guide: https://github.com/react-native-community/react-native-camera/pull/2497

juslin03, 2019-09-25:
> Got the same issue with `react-native-image-crop-picker`, fixed by
> 
> `maven { url "https://www.jitpack.io" }`
> (Added `www`)
> 
> I dont know why, but it works

Thank you  yurkagon, your method worked for me. Just add "www" to maven { url "https://jitpack.io" } in "android/build.gradle" file

Summerian-Winter, 2019-09-26:
@yurkagon: I was " " <--- this close to giving my laptop a lethal headbutt, so I guess my laptop owes you a favour

closed 2019-12-02 by stale[bot]

Armen6964, 2021-03-15:
Just turn off the internet and retry.

RamanathanN, 2021-03-15:
I am facing this issue now. Already my "build.gradle" file has "maven { url 'https://www.jitpack.io' }" only. Is there any other solution?

amanfcp, 2021-03-15:
> Just turn off the internet and retry.

This solution worked. Seems to be some temporary issue

fnadj94, 2021-03-15:
> 
> 
> I am facing this issue now. Already my "build.gradle" file has "maven { url 'https://www.jitpack.io' }" only. Is there any other solution?

> 
> 
> > Just turn off the internet and retry.
> 
> This solution worked. Seems to be some temporary issue

https://status.bintray.com/incidents/ctv4bdfz08bg

JCenter is down. This seems to be causing this issue

christhorp, 2021-03-16:
I still facing this problem, tried restarting internet and computer. Any other ideas?

hu-qi, 2021-03-16:
The same, JCenter.
[image]

[image]

hu-qi, 2021-03-16:
> > Just turn off the internet and retry.
> 
> This solution worked. Seems to be some temporary issue

Hello，How did you solve it？Thanks！

KieranCarden, 2021-03-16:
Same here, down as well. However, started working a moment ago with no change by me. Good news? But has me worried about the state of things concerning JCenter.

jihokim2, 2021-03-16:
I found workaround. When you build, turn off wifi. It is working. I think it is temporary solution until Jcenter works fine

hu-qi, 2021-03-16:
> I found workaround. When you build, turn off wifi. It is working. I think it is temporary solution until Jcenter works fine

Nice! When I run, turn off wifi, it works！But the build failed！

NewtonOngko, 2021-03-16:
> I found workaround. When you build, turn off wifi. It is working. I think it is temporary solution until Jcenter works fine

after build done i turn on wifi to do login and other its work fine for me

EvertonCosmo, 2021-03-16:
Need run in offline, Jcenter apparently is down

mcvictormurillo, 2021-03-16:
The same problem also happens to me. thanks for your help

FladioArmandika, 2021-03-16:
i just turn off the internet but the build failed.

TienTungs295, 2021-03-16:
@yurkagon thank you, your solution worked for me

unicornRainbows, 2021-03-16:
haha turning internet off and doing the build works.. but only if you didn't clean your android build before.. Got to 💌 computers some times hey.. 😉

chengsokdara, 2021-03-16:
Just wait til Jcenter is up again. Any workaround? I already clean the build cache.
it is currently down
https://status.bintray.com/

bblack1225, 2021-03-16:
I just turn off the wifi when building my app and its work for me!

mcvictormurillo, 2021-03-16:
**Jcenter is working now!!!**

ekkam04, 2021-09-16:
Facing this issue today and the app only builds by disconnecting from the internet. Is Jcenter down again?

shaurya2612, 2021-09-16:
> Facing this issue today and the app only builds by disconnecting from the internet. Is Jcenter down again?

Came here just to say this, facing the same problem. Try running offline.

nolife08021, 2021-09-16:
> > Facing this issue today and the app only builds by disconnecting from the internet. Is Jcenter down again?
> 
> Came here just to say this, facing the same problem. Try running offline.

did it works? I dont know why I still getting the same error

ekkam04, 2021-09-16:
> > Facing this issue today and the app only builds by disconnecting from the internet. Is Jcenter down again?
> 
> Came here just to say this, facing the same problem. Try running offline.

Yes running offline and then once it builds switching to online works for now. So is it a worldwide issue as of now?

shaurya2612, 2021-09-16:
> Yes running offline and then once it builds switching to online works for now. So is it a worldwide issue as of now?

Yep

troZee, 2021-09-16:
For targetSDK 30 it does not work at all:

Before:
```
    ext {
        buildToolsVersion = "30.0.2"
        minSdkVersion = 21
        compileSdkVersion = 30
        targetSdkVersion = 30
        ndkVersion = "20.1.5948944"
    }
```

After: 
```
    ext {
        buildToolsVersion = "29.0.2"
        minSdkVersion = 21
        compileSdkVersion = 29
        targetSdkVersion = 29
        supportLibVersion = "28.0.0"
        ndkVersion = "22.1.7171670"
    }
  ```

Adeel984, 2021-09-16:
> Just turn off the internet and retry.

OMG That works!

dipras, 2021-09-16:
How to build react native on Offline?

denis0324, 2021-09-16:
Still same issue

relativelyrehan, 2021-09-16:
Having the same issue, some other dependencies in my project need an internet connection. Any other solution?

Arslan106, 2021-09-16:
still same issue

ekkam04, 2021-09-16:
> How to build react native on Offline?

Just disconnect your computer from the internet and build

Hemistone, 2021-09-16:
Is here TG? everybody talking this issue as live haha 🤣

redvelvet-fan, 2021-09-16:
I Have same issue now T.T

mohammed-sec2010, 2021-09-16:
I Have same issue now

youngdocho, 2021-09-16:
> > Just turn off the internet and retry.
> 
> OMG That works!

It works for me 👍

MaxJadav, 2021-09-16:
Any one have alternate option?
Bintray services will no longer be available. JCenter will no longer be available for non-Artifactory clients. This means by the end of March 2021, publishing new packages to Bintray will no longer be allowed. Existing packages can still be downloaded until February 1st 2022.
Ref: https://testfairy.com/blog/jcenter-and-bintray-is-shutting-down-what-to-do/

ortegarenzy, 2021-09-16:
> Just turn off the internet and retry.

This definitely works right now!

xoapit, 2021-09-16:
I meet this issue a couple of hours ago, there are no solutions for now. I cannot turn off the internet because I need to build on the server. It's very crazy when Google Service updates something.

ThomasMcDonnell, 2021-09-16:
I have the encountered the same issue and have been stumped for the last few hours! Turning off the internet is not a solution for me either as it needs to build on a server.

huzaifaaak, 2021-09-16:
Turning off internet isn't working for me either. Good thing I have local build installed in the emulator. Can run it from the local server.

https://jcenter.bintray.com/ is giving 403 forbidden

huzaifaaak, 2021-09-16:
Turning off internet isn't working for me either

https://jcenter.bintray.com/ is giving 403 forbidden

imarem, 2021-09-16:
Jcenter apparently is down https://status.jitpack.io/

shoaib90, 2021-09-16:
I guess, Need to wait for Jcenter to start working again

Rushikesh1212, 2021-09-16:
For me also its not working. If anyone find the solution please let us know.

choukrus, 2021-09-16:
Oh god .. I thought I was the only one facing this problem now as the status of servers doesn't show any problem...

Feeling better now because I thought I changed something in my app ( I didn't, yesterday everything worked fine, and now it doesn't )

Can't switch internet off because it's an react native app..

kishoreAwdar, 2021-09-16:
OMG. I did lot of things...
Thank god otherwise I will delete all of them

maddeha, 2021-09-16:
glad that's a common problem

LoknathPandit, 2021-09-16:
Yes also i am facing same issue

Jscripter-pk, 2021-09-16:
@LoknathPandit  Turn your internet off and try to run your project

khaledJabari-90, 2021-09-16:
I'm facing the same issue. 
jitpack is down. 
https://status.jitpack.io/776679744

meliodev, 2021-09-16:
@shoaib90 Yeah it's good reason to take some rest !

[40 more replies]