react-native-camera/react-native-camera #2496: How to mock RNCamera.Constants.VideoQuality?

PointCodeZero, 2019-09-20:
I'm writing tests for the RNCamera, but I'm getting the error below:

[image]

I mocked the library and followed the instructions as per the documentation and added to the Constants the VideoQuality, but even though the tests fail:

[image]

When I delete the VideoQuality from my code all tests that are not related to the record passes.

Please, someone can shine some light into this?

n1ru4l, 2019-09-23:
Needs more context. Do you have a reproduction repository?

PointCodeZero, 2019-09-25:
Unfortunately, I cannot share the repo, but below you can find my component and dependencies I'm using.

Please let me know if you need more information.

```
class VideoRecorder extends React.PureComponent<IVideoRecordProps, IVideoRecordState> {
  private camera: RNCamera;
  private toggleButton: boolean = true;

  public static defaultProps = {
    videoQuality: '720p',
    videoBitrate: 5
  };

  public static propTypes = {
    videoQuality: PropTypes.oneOf(VideoQualityTypes).isRequired,
    videoBitrate: PropTypes.number.isRequired,
    onUpload: PropTypes.func.isRequired,
    t: PropTypes.func
  }

  public constructor(props: IVideoRecordProps) {
    super(props);
    this.state = {
      data: null,
      onCamera: false,
      useFrontCamera: false,
      media: '',
      isPhotoTaken: false,
      isRecording: false,
      recordOptions: {
        maxDuration: 3600,
        maxFileSize: 104857600,
        quality: RNCamera.Constants.VideoQuality[props.videoQuality],
      },
      isVideoDone: false,
      thumbnail: '',
      selectedColor: 'secondary',
      unSelectedColor: 'transparent',
      timer: null,
      minutes_Counter: '00',
      seconds_Counter: '00',
      startDisable: false,
      focusedScreen: null,
      hasCameraPermission: null
    };
  }

  private toggleCamera = () => {
    const { useFrontCamera } = this.state;
    this.setState({ useFrontCamera: !useFrontCamera });
  }

  private handleColorChange = () => {
    this.toggleButton = !this.toggleButton;
    this.setState({
      selectedColor: this.toggleButton ? 'secondary' : 'transparent',
      unSelectedColor: this.toggleButton ? 'transparent' : 'secondary'
    });
  }

  // Duration Video Timer - Begin
  private onTimerStart = () => {
    const timer = setInterval(() => {
      let num = (Number(this.state.seconds_Counter) + 1).toString(),
        count = this.state.minutes_Counter;

      if (Number(this.state.seconds_Counter) == 59) {
        count = (Number(this.state.minutes_Counter) + 1).toString();
        num = '00';
[... 255 lines cut ...]
  await wrapper
    .find('#picture_button')
    // .at(1)
    .props()
    .onPress();
  await wrapper
    .find('#upload_photo_button')
    // .at(1)
    .props()
    .onPress();
  expect(onUpload.mock.calls.length).toBe(1);
});
```

Here are my dependencies: 

```
"react": "16.8.6",
"react-native": "0.60.4",
"react-native-camera": "^3.4.0",
"babel-jest": "^23.4.2",
"enzyme": "^3.10.0",
"enzyme-to-json": "^3.3.5",
"jest": "^24.8.0",
```

closed 2019-12-01 by stale[bot]