react-native-camera/react-native-camera #2498: Starting video recording failed. Another recording might be in progress.

FaisalAli19, 2019-09-23:
# Bug Report

**To Do First**

- [Y] Did you try latest release?
- [Y] Did you try master?
- [Y] Did you look for existing matching issues?

**Platforms**
Android

**Versions**
- Android: 9
- react-native-camera: 3.4.0
- react-native: 0.60.5
- react: 16.8.6
- react-navigation: 3.12.0

**Description/Current Behaviour**
RecordAsync throws an error
`Error: Starting video recording failed. Another recording might be in progress.
    at createErrorFromErrorData (NativeModules.js:152)
    at NativeModules.js:104
    at MessageQueue.__invokeCallback (MessageQueue.js:442)
    at MessageQueue.js:127
    at MessageQueue.__guard (MessageQueue.js:343)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (MessageQueue.js:126)
    at debuggerWorker.js:80`

I have all the permissions in my AndroidManifest.xml 
```
<uses-permission android:name="android.permission.RECORD_AUDIO"/>
<uses-permission android:name="android.permission.CAMERA" />
 <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
 <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
```

I am also checking all the permission before rendering the camera with react-native-permission and the response are `authorized` for the camera, audio, and storage.

**Expected Behaviour**
It should have resolved the promise with video details like Uri, codec, etc.

**Steps to Reproduce**
Click on the start recording button and record async throws an error in console.

Sample code.
```
startRecording = async () => {
 if (this.camera) {
      try {
        const options = {
          quality: 2,
          videoBitrate: 8000000,
          maxDuration: 30
        };
        const promise = this.camera.recordAsync(options);
        if (promise) {
          this.setState({ recording: true });
          const data = await promise;
          this.setState({ recording: false });
          console.log(data)
         }
      } catch (error) {
        console.log(error);
      }
    }
}

<RNCamera
            style={cameraStyle}
            ref={ref => {
              this.camera = ref;
            }}
            type={cameraType}
            androidCameraPermissionOptions={{
              title: 'Permission to use camera',
              message: 'We need your permission to use your camera phone',
              buttonPositive: 'Ok',
              buttonNegative: 'Cancel',
            }}
            androidRecordAudioPermissionOptions={{
              title: 'Permission to use audio recording',
              message: 'We need your permission to use your audio',
              buttonPositive: 'Ok',
              buttonNegative: 'Cancel',
            }}
           captureAudio={true}
          />
```
Thanks in advance.

PFA android logs.
[Android_error.log](https://github.com/react-native-community/react-native-camera/files/3643394/Android_error.log)

cristianoccazinsp, 2019-09-24:
Can you get the logs from Logcat/Android Studio? Quite a generic error, but I believe the actual Java error is logged.

FaisalAli19, 2019-09-24:
Here the logs from android studio.

```
--------- beginning of system
2019-09-24 13:18:17.560 17732-17755/com.wocal D/DecorView: onWindowFocusChangedFromViewRoot hasFocus: true, DecorView@3969f35[MainActivity]
2019-09-24 13:18:17.574 17732-17732/com.wocal E/SensorManager: registerListenerImpl sensorName:BMI160 Accelerometer,isWakeUpSensor:false,callingApp: com.wocal,callingPid:17732,callingUid:10462
2019-09-24 13:18:17.592 17732-17732/com.wocal E/libc: Access denied finding property "vendor.camera.hal1.packagelist"
2019-09-24 13:18:17.936 17732-17732/com.wocal D/Camera: CAMERA_MSG_STATE_CALLBACK
2019-09-24 13:18:17.936 17732-17732/com.wocal D/Camera: mCameraStateCallback is null
2019-09-24 13:18:18.160 17732-18000/com.wocal W/MapperHal: buffer descriptor with invalid usage bits 0x2000
2019-09-24 13:18:18.582 17732-17732/com.wocal D/Camera: CAMERA_MSG_STATE_CALLBACK
2019-09-24 13:18:18.582 17732-17732/com.wocal D/Camera: mCameraStateCallback is null
2019-09-24 13:18:19.114 17732-18057/com.wocal W/MapperHal: buffer descriptor with invalid usage bits 0x2000
2019-09-24 13:18:19.124 17732-17732/com.wocal D/Camera: CAMERA_MSG_STATE_CALLBACK
2019-09-24 13:18:19.124 17732-17732/com.wocal D/Camera: mCameraStateCallback is null
2019-09-24 13:18:19.725 17732-18000/com.wocal W/MapperHal: buffer descriptor with invalid usage bits 0x2000
2019-09-24 13:18:20.202 17732-17732/com.wocal D/ReactExoplayerView: onStateChanged: playWhenReady=false, playbackState=ready
2019-09-24 13:18:20.204 17732-17858/com.wocal D/MediaCodecInfo: NoSupport [sizeAndRate.support, 750x1000x-1.0] [OMX.qcom.video.decoder.mpeg4, video/mp4v-es] [OnePlus5T, ONEPLUS A5010, OnePlus, 28]
2019-09-24 13:18:20.208 17732-17858/com.wocal I/chatty: uid=10462(com.wocal) ExoPlayerImplIn identical 4 lines
2019-09-24 13:18:20.208 17732-17858/com.wocal D/MediaCodecInfo: NoSupport [sizeAndRate.support, 750x1000x-1.0] [OMX.qcom.video.decoder.mpeg4, video/mp4v-es] [OnePlus5T, ONEPLUS A5010, OnePlus, 28]
2019-09-24 13:18:20.497 17732-17732/com.wocal D/Camera: CAMERA_MSG_STATE_CALLBACK
2019-09-24 13:18:20.497 17732-17732/com.wocal D/Camera: mCameraStateCallback is null
2019-09-24 13:18:20.923 17732-18057/com.wocal W/MapperHal: buffer descriptor with invalid usage bits 0x2000
2019-09-24 13:18:21.144 17732-17732/com.wocal D/Camera: CAMERA_MSG_STATE_CALLBACK
2019-09-24 13:18:21.144 17732-17732/com.wocal D/Camera: mCameraStateCallback is null
2019-09-24 13:18:21.228 17732-17732/com.wocal D/Camera: CAMERA_MSG_STATE_CALLBACK
2019-09-24 13:18:21.228 17732-17732/com.wocal D/Camera: mCameraStateCallback is null
2019-09-24 13:18:21.392 17732-17799/com.wocal W/ReactNativeJS: Warning: componentWillReceiveProps is deprecated and will be removed in the next major version. Use static getDerivedStateFromProps instead.
2019-09-24 13:18:21.440 17732-17732/com.wocal E/SensorManager: registerListenerImpl sensorName:BMI160 Accelerometer,isWakeUpSensor:false,callingApp: com.wocal,callingPid:17732,callingUid:10462
2019-09-24 13:18:21.446 17732-17732/com.wocal I/MediaRecorder: param-use-64bit-offset=1
2019-09-24 13:18:21.498 17732-17732/com.wocal E/MediaRecorder: start 0 m_rotation
2019-09-24 13:18:21.599 17732-17732/com.wocal E/MediaRecorder: start failed: -2147483648
2019-09-24 13:18:21.601 17732-17732/com.wocal E/CAMERA_1::: Record start failed
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.native_start(Native Method)
        at android.media.MediaRecorder.start(MediaRecorder.java:1143)
        at com.google.android.cameraview.Camera1.record(Camera1.java:597)
        at com.google.android.cameraview.CameraView.record(CameraView.java:553)
        at org.reactnative.camera.RNCameraView.record(RNCameraView.java:276)
        at org.reactnative.camera.CameraModule$6.execute(CameraModule.java:294)
        at com.facebook.react.uimanager.UIViewOperationQueue$UIBlockOperation.execute(UIViewOperationQueue.java:530)
        at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:844)
        at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:952)
        at com.facebook.react.uimanager.UIViewOperationQueue.access$2200(UIViewOperationQueue.java:44)
        at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:1012)
        at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:29)
        at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:172)
        at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:84)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1002)
        at android.view.Choreographer.doCallbacks(Choreographer.java:816)
        at android.view.Choreographer.doFrame(Choreographer.java:748)
        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:990)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6898)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:537)
[... cut ...]
2019-09-24 13:18:28.191 17732-17732/com.wocal E/SensorManager: unregisterListenerImpl callingApp: com.wocal,callingPid:17732,callingUid:10462
2019-09-24 13:18:33.593 17732-17732/com.wocal D/ReactExoplayerView: onStateChanged: playWhenReady=false, playbackState=ready
2019-09-24 13:18:37.855 17732-17732/com.wocal D/ReactExoplayerView: onStateChanged: playWhenReady=false, playbackState=ready
2019-09-24 13:18:40.679 17732-17732/com.wocal D/ReactExoplayerView: onStateChanged: playWhenReady=false, playbackState=ready

```

FaisalAli19, 2019-09-25:
Hi,

I have tried restarting the device. It is still giving the same error with the same logs. I have also tried it on Samsung and its working fine.

I will test this on other devices and will share the logs if the same error occurred.

Thanks.

Deepak-merahkee, 2019-09-25:
Try this, here is the clear explanation regarding RNCamera video reacording [https://stackoverflow.com/a/58099743/8746415](https://stackoverflow.com/a/58099743/8746415)

FaisalAli19, 2019-09-25:
Hi,

Thanks, Deepak I will surely check that.

FYI, recording works fine if I don't pass any options to recordAsync but fails with options. I will check it on other devices apart from Oneplus and Samsung to confirm this issue.

cristianoccazinsp, 2019-09-25:
Can you pinpoint what option is causing issues? I've been using record with
a low quality setting (480p and 720p, never 2k) and it works fine on the
devices I've tested.

FaisalAli19, 2019-09-27:
Hi,

It is a quality issue, changing the quality to 720p doesn't throw the error.

Is there a way to check the quality on the device.

Thanks.

cristianoccazinsp, 2019-09-27:
Not that I'm aware of, but it would be good for the code to fail with a nicer / less generic error in that case.

FaisalAli19, 2019-09-27:
Yes, that would be great.

FaisalAli19, 2019-09-30:
Hi,

So after testing on multiple devices. I found a strange bug if I record a video and post the same to the server and start recording the second video. Same error popup.

Any thought on why recording the video fails the second time.

Thanks in advance.

jrartiga, 2019-10-19:
I have the same error, 
is this common or just happen with the last version of react-native-camera ?

FaisalAli19, 2019-10-19:
Hey jrartiga,

I am still unable to figure out the issue. 
When I change the video quality to 720p it worked for some time and then started to get the same error while recording the video 2nd time.

JakubPod, 2019-10-26:
Hey,
for me, this issue appeared when I updated the react-native from 0.59.8 to 0.61.2. I also updated react-native-camera from 2.9.0 to the current 3.8.0 but this issue comes up regardless.

Tweaking with video quality and orientation does nothing for me.

JakubPod, 2019-10-28:
I found that the camera starts to work after the following scenario:
1. Load the app, go to the camera, try to record - it throws an error.
2. Kill the app.
3. Go to settings and clear the app's cache.
4. Open the app again, the camera is recording OK.

Just killing the app without clearing the cache also seems to work but not all the time.
I'm testing this on Samsung Galaxy J5. On the emulated Pixel 3 (API 28) there was no issue.

JakubPod, 2019-10-29:
[image]

I got this screen while starting the app today. Is it possible that the Camera is created twice and therefore while trying to record a video the other Camera's instance is "holding the lock"?

I've looked around my dependencies and I don't see anything else using react-native-camera, only me, directly.

FaisalAli19, 2019-10-29:
Hi JakubPod,

I am not getting this error. But I can confirm that the camera error is happening mostly in development, in production its working fine.

cristianoccazinsp, 2019-10-29:
Probably unrelated, but if the camera is created twice, maybe the changes here can help: https://github.com/react-native-community/react-native-camera/pull/2560

Can you try with that PR? Adds some race condition checks and also some missing cleanup on camera destroy which might be related.

JakubPod, 2019-10-29:
Hey Cristiano, I've tried with your PR, but sadly the issue still appears.

cristianoccazinsp, 2019-10-29:
@JakubPod  if this is very device specific, and related to the media recording settings, I suggest manually debugging and playing around with the Java code to see which settings are causing issues.

dnl-jst, 2019-11-04:
Having the same error on two different devices:

- Huawei Y5 2019 with Android 9
- Samsung Galaxy A5 (2016) with Android 7.0

On the Huawei it worked sometimes, but now I don't get it to work anymore. I tried to restart the phones, cleared their data folders and reinstalled the apps completely. Nothing helped.

[image]

cristianoccazinsp, 2019-11-04:
@dnl-jst  can you get Android logs? Is it the same `at android.media.MediaRecorder.start` fail as the logs posted above?

dnl-jst, 2019-11-04:
Does that help in any way?

Output produced with:

`adb logcat | grep -i camera`

```11-04 22:31:02.394 18986 18986 I PermissionManager: camera remind result:true
11-04 22:31:02.394 18986 18986 I HwMediaPermWrapper: confirmCameraPermission, blocked:false
11-04 22:31:02.509   621 20384 I CameraSource: Using encoder format: 0x22
11-04 22:31:02.510   621 20384 I CameraSource: Using encoder data space: 0x104
11-04 22:31:02.518   599  2504 E HwCameraServiceEx: Failed to load CameraService sounds: /product/media/audio/ui/VideoRecord.ogg
11-04 22:31:02.817   637 20331 I BWC     : Profile_Change:[BWCPT_CAMERA_PREVIEW]:OFF,current concurrency is 0x10,DDR=,EMI scenario=BWL_SCN_ICFP OFF
11-04 22:31:02.820   599  4752 I Camera3-IOStreamBase: returnAnyBufferLocked: Stream 3: All buffers returned; now idle
11-04 22:31:02.820   599  4752 I Camera3-IOStreamBase: returnAnyBufferLocked: Stream 4: All buffers returned; now idle
11-04 22:31:02.821   599 20265 I Camera3-Status: Camera device is now idle
11-04 22:31:02.821   599  2504 I Camera3-Device: deleteStream: Camera 1: Deleting stream 4
11-04 22:31:02.821   599  2504 I Camera3-Device: deleteStream(), delete stream: 4, 0xf21f9480
11-04 22:31:02.821   599  2504 I Camera3-Stream: disconnect: Stream 4, 0xf21f9480: Disconnecting...
11-04 22:31:02.821   599  2504 I Camera3-OutputStream: disconnectLocked: disconnecting stream 4 from native window, for stream 0xf21f9480
11-04 22:31:02.822   599  2504 I CameraLatencyHistogram: Stream 4 dequeueBuffer latency histogram (195) samples:
11-04 22:31:02.822   599  2504 I CameraLatencyHistogram:         5     10     15     20     25     30     35     40     45    inf (max ms)
11-04 22:31:02.822   599  2504 I CameraLatencyHistogram:      100.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00   0.00 (%)
11-04 22:31:02.822   599  2504 I Camera3-Device: deleteStream: Camera 1: Deleting stream 4, return: 0
11-04 22:31:02.823   599  2504 I Camera3-Stream: Camera3Stream(), create stream 0xf18a1f00, 1280 x 720, format: 0x22
11-04 22:31:02.892   637 20260 W mtkcam-AppStreamMgr: [1-ConfigHandler::checkStream] framework stream dataspace:0x08c20000(V0_JFIF|STANDARD_BT601_625|TRANSFER_SMPTE_170M|RANGE_FULL) {.v3_2 = {.id = 1, .streamType = OUTPUT, .width = 2560, .height = 1920, .format = BLOB, .usage = CPU_READ_NEVER | CPU_READ_RARELY | CPU_READ_OFTEN | CPU_WRITE_NEVER (0x3), .dataSpace = UNKNOWN | STANDARD_UNSPECIFIED | STANDARD_BT601_625 | TRANSFER_UNSPECIFIED | TRANSFER_LINEAR | TRANSFER_SRGB | TRANSFER_SMPTE_170M | RANGE_UNSPECIFIED | RANGE_FULL | V0_JFIF (0x8c20000), .rotation = ROTATION_0}, .physicalCameraId = "", .bufferSize = 5898240}
11-04 22:31:02.892   637 20260 W mtkcam-AppStreamMgr: [1-ConfigHandler::checkStream] framework stream dataspace:0x00000104(BT709) {.v3_2 = {.id = 5, .streamType = OUTPUT, .width = 1280, .height = 720, .format = IMPLEMENTATION_DEFINED, .usage = CPU_READ_NEVER | CPU_WRITE_NEVER | VIDEO_ENCODER (0x10000), .dataSpace = UNKNOWN | STANDARD_UNSPECIFIED | TRANSFER_UNSPECIFIED | RANGE_UNSPECIFIED | BT709 (0x104), .rotation = ROTATION_0}, .physicalCameraId = "", .bufferSize = 0}
11-04 22:31:02.895   637 20260 I BWC     : Profile_Change:[BWCPT_CAMERA_CAPTURE]:OFF,current concurrency is 0x0,DDR=,EMI scenario=BWL_SCN_ICFP OFF
11-04 22:31:02.923   637 20260 I mtkcam-mtk_scenario_mgr: [mtk_get_features_table_by_scenario] find features for scenario(1 : MTK_CAMERA_SCENARIO_CAPTURE_NORMAL)
11-04 22:31:02.923   637 20260 I mtkcam-FeatureSettingPolicy: [evaluateCaptureConfiguration] scenario(MTK_CAMERA_SCENARIO_CAPTURE_NORMAL) support feature:TP_FEATURE_HDR(0x100000000), feature combination:MTK_FEATURE_COMBINATION_HDR(0x9000000f8)
11-04 22:31:02.923   637 20260 I mtkcam-FeatureSettingPolicy: [evaluateCaptureConfiguration] scenario(MTK_CAMERA_SCENARIO_CAPTURE_NORMAL) support feature:MTK_FEATURE_AINR(0x2000), feature combination:MTK_FEATURE_COMBINATION_AINR(0x8000020f8)
11-04 22:31:02.923   637 20260 I mtkcam-FeatureSettingPolicy: [evaluateCaptureConfiguration] scenario(MTK_CAMERA_SCENARIO_CAPTURE_NORMAL) support feature:MTK_FEATURE_MFNR(0x1), feature combination:MTK_FEATURE_COMBINATION_MFNR(0x8000000f9)
11-04 22:31:02.923   637 20260 I mtkcam-FeatureSettingPolicy: [evaluateCaptureConfiguration] scenario(MTK_CAMERA_SCENARIO_CAPTURE_NORMAL) support feature:MTK_FEATURE_REMOSAIC(0x4), feature combination:MTK_FEATURE_COMBINATION_REMOSAIC(0x8000000fc)
11-04 22:31:02.923   637 20260 I mtkcam-FeatureSettingPolicy: [evaluateCaptureConfiguration] scenario(MTK_CAMERA_SCENARIO_CAPTURE_NORMAL) support feature:NO_FEATURE_NORMAL(0), feature combination:MTK_FEATURE_COMBINATION_SINGLE(0x8000000f8)
11-04 22:31:02.923   637 20260 I mtkcam-mtk_scenario_mgr: [mtk_get_features_table_by_scenario] find features for scenario(51 : MTK_CAMERA_SCENARIO_STREAMING_NORMAL)
11-04 22:31:02.924   637 20260 I mtkcam-FeatureSettingPolicy: [evaluateStreamConfiguration] scenario(MTK_CAMERA_SCENARIO_STREAMING_NORMAL) support feature:NO_FEATURE_NORMAL(0), feature combination:MTK_FEATURE_COMBINATION_VIDEO_NORMAL(0xc0)
11-04 22:31:02.925   637 20260 I MtkCam/HwInfoHelper: [getImgoFmt] debug.camera.rawunpak = -1,  useUnpakFmt = 0
11-04 22:31:02.932   637 20260 I MtkCam/ppl_context: [dump]     [APP     ] 01 2560x1920 OUT t:0/r:0 maxBuffers:5 d/s:0x08c20000(V0_JFIF|STANDARD_BT601_625|TRANSFER_SMPTE_170M|RANGE_FULL) Hal-Client-usage:0x3(0|SW_READ_OFTEN) Real:0x2300() Request:0x21(BLOB) Override:0x21(BLOB) Hal-usage:0x20033(0|SW_READ_OFTEN|SW_WRITE_OFTEN|HW_CAME [... cut ...]
11-04 22:31:21.911   599  4752 I Camera3-Device: Camera 1: notifyError: HAL error, frame 750, stream 0: 2
11-04 22:31:21.911   599  4752 W Camera2Client: notifyError: Received recoverable error 3 from HAL - ignoring, requestId 20000004
11-04 22:31:21.912   599  4752 W Camera3-OutputStream: returnBufferCheckedLocked: A frame is dropped for stream 3 due to buffer error.
11-04 22:31:21.912   599  4752 W Camera3-OutputStream: returnBufferCheckedLocked: A frame is dropped for stream 5 due to buffer error.```

Thank you for your support!

cristianoccazinsp, 2019-11-04:
@dnl-jst  you probably need to filter by your app first, and also while the error actually happens.  From the above logs, I can't really see anything related to RNCamera

dnl-jst, 2019-11-04:
```
11-04 22:54:44.785 22933 22933 I PermissionManager: camera remind result:true
11-04 22:54:44.785 22933 22933 I HwMediaPermWrapper: confirmCameraPermission, blocked:false
11-04 22:54:44.801 22933 22933 I MediaRecorder: enter in file frameworks/av/media/libmedia/mediarecorder.cpp, function prepare, line 461
11-04 22:54:44.801 22933 22933 I IMediaRecorder: prepare (BpMediaRecorder client) in file frameworks/av/media/libmedia/IMediaRecorder.cpp, function prepare, line 253
11-04 22:54:44.801 22933 22933 I MediaRecorder: exit in file frameworks/av/media/libmedia/mediarecorder.cpp, function prepare, line 495
11-04 22:54:44.801 22933 22933 I MediaRecorder: start
11-04 22:54:45.280 22933 22933 I HwMediaRecorderImpl: sendStateChangedIntent, state=3
11-04 22:54:45.387 22933 24263 E ReactNativeJS: Error: Starting video recording failed. Another recording might be in progress.
11-04 22:54:45.461 22933 22933 I HwPhoneWindow: updateLayoutParamsColor false mSpecialSet=false, mForcedNavigationBarColor=false, navigationBarColor=0, mNavBarShow=false, mIsFloating=false
11-04 22:54:45.464 22933 22933 I HwPhoneWindow: updateLayoutParamsColor true mSpecialSet=true, mForcedNavigationBarColor=false, navigationBarColor=fffcfcfc, mNavBarShow=false, mIsFloating=false
11-04 22:54:45.549 22933 23014 D OpenGLRenderer:   HWUI Binary is  enabled
11-04 22:54:45.575 22933 24509 W HiTouch_HiTouchSensor: depended package hiTouch does n't exist!
11-04 22:54:45.575 22933 24509 I HiTouch_HiTouchSensor: HiTouch restricted: system app HiTouch don't exist.
11-04 22:54:45.576 22933 24509 D HiTouch_PressGestureDetector: onAttached, package=de.jostdevops.rumapatientapp, windowType=2, mHiTouchRestricted=true
11-04 22:54:45.580 22933 22933 I HwPhoneWindow: updateLayoutParamsColor true mSpecialSet=true, mForcedNavigationBarColor=false, navigationBarColor=fffcfcfc, mNavBarShow=true, mIsFloating=false
11-04 22:54:45.718 22933 23014 D OpenGLRenderer:   HWUI Binary is  enabled
11-04 22:54:45.720 22933 22933 W InputMethodManager: startInputReason = 1
11-04 22:54:45.760 22933 22933 I ViewRootImpl: jank_removeInvalidNode jank list is null
```

dnl-jst, 2019-11-04:
I think I've found my problem: I started video recording automatically after scanning a qr code - the problem is, the onBarCodeRead event can trigger multiple times, so it tried to start the video recording multiple times. So this seems to be a problem on my side. ;-)

Sorry for trouble. ;-)

cristianoccazinsp, 2019-11-04:
Well, that might help other people. 

Just in case, I don't really see anything wrong with the above logs neither. Looks like it fails right before `11-04 22:54:44.801 22933 22933 I MediaRecorder: start`, which is similar to the logs posted above where it would fail in the same place.

Perhaps we need a better/nicer error when the video fails to start due to media recorder errors.

JakubPod, 2019-11-05:
I've got some more useful logs that I'm currently investigating.

> 11-05 11:06:25.656  2611  2951 D ExynosCamera1Parameters: [CAM_ID(0)][ParametersBack]-DEBUG(checkSetfileYuvRange[5216]):m_cameraId(0) : general[setfile(0) YUV range(0)] : reprocesing[setfile(5) YUV range(0)]
> 11-05 11:06:25.656  2611  2951 D ExynosCamera3Interface: DEBUG:duration time(    2 msec):(HAL_camera_device_set_parameters)
> 11-05 11:06:25.657  2611  2951 D ExynosCamera3Interface: DEBUG:duration time(    0 msec):(HAL_camera_device_get_parameters)
> 11-05 11:06:25.657  2611  2951 D ExynosCamera3Interface: DEBUG:duration time(    0 msec):(HAL_camera_device_put_parameters)
**> 11-05 11:06:25.658  2611  2951 D ExynosCamera3Interface: DEBUG:duration time(    0 msec):(HAL_camera_device_store_meta_data_in_buffers)
> 11-05 11:06:25.706  4068 32317 W CameraClient: attempt to use a locked camera from a different process (old pid 4081, new pid 2789)**
> 11-05 11:06:25.707  2611  2951 D ExynosCamera3Interface: DEBUG:duration time(    0 msec):(HAL_camera_device_recording_enabled)
> 11-05 11:06:25.710  2789  2789 E CAMERA_1:::    at com.google.android.cameraview.Camera1.record(Came
> ra1.java:640)
> 11-05 11:06:25.710  2789  2789 E CAMERA_1:::    at com.google.android.cameraview.CameraView.record(C
> ameraView.java:584)
> 11-05 11:06:25.710  2789  2789 E CAMERA_1:::    at org.reactnative.camera.RNCameraView.record(RNCame
> raView.java:279)
> 11-05 11:06:25.710  2789  2789 E CAMERA_1:::    at org.reactnative.camera.CameraModule$6.execute(Cam
> eraModule.java:297)
----------------------------
> 2019-11-05 11:46:17.466 4068-32317/? E/CameraService: checkTemperature (0)
> 2019-11-05 11:46:17.477 4068-32317/? E/CameraService: setCompanionDisableSysFs : cc.disable property set to 0
> 2019-11-05 11:46:17.491 4068-32317/? E/SecCameraCoreManager: initialize1
> 2019-11-05 11:46:17.507 2611-4037/? E/ExynosCamera: Build Date is (__DATE__) (__TIME__)
> 2019-11-05 11:46:17.546 4068-32317/? E/SecCameraCoreManager: initialize2
> 2019-11-05 11:46:17.596 17259-17259/com.teachsnap E/Camera1: startCameraPreview
> 2019-11-05 11:46:17.606 4068-4068/? E/SecCameraCoreManager: checkTemperature (0)
> 2019-11-05 11:46:17.622 2611-20887/? E/ExynosCameraBufferManager: [CAM_ID(0)][SCP_BUF]-ERR(m_alloc[1945]):m_reqBufCount(12)
> **2019-11-05 11:46:20.459 4068-4068/? E/CameraClient: setVideoBufferMode: 692: videoBufferMode 2 is not supported.**
> 2019-11-05 11:46:20.611 17259-17259/com.teachsnap E/CAMERA_1::: Record start failed
>     java.lang.RuntimeException: start failed.
>         at android.media.MediaRecorder._start(Native Method)
>         at android.media.MediaRecorder.start(MediaRecorder.java:1370)

cristianoccazinsp, 2019-11-05:
Can you play around with these settings here? https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L1384

Looks like some codec/buffer settings are not supported by your device?

JakubPod, 2019-11-06:
I solved my issue. The react-native-camera was working as expected all this time and the issue was fully on the side of my app. 

The case was: Another part of my app has a feed (kind of like Instagram) with multiple videos. Every instance of a loaded video obviously needs to use a decoder. These videos were all kept in memory after navigating to the camera and the decoder was not released, so when I tried to record a video, the camera could not access the busy decoder. 

The reason why I could not replicate this issue on another device was that different devices have different limits on how many tasks a hardware decoder can do simultaneously.

Another new thing for me was that when reloading the app, the componentWillUnmount method is not called, so even when reloading the decoders were not being released. 

TLDR: The error message is actually helpful and I'm just slow-witted.

kyle-belle, 2020-01-16:
ok so in my case it was because i thought that the "path" property in the options object took only the folder's path and that it would assign the file name itself. turns out you also have to include the file name in path. don't know if this will help anyone buttttt shoudn't hurt either

fabriziobertoglio1987, 2020-03-02:
as mentioned in [this comment][1] from JakubPod 

>I think my issue is with the app memory usage, which is pretty high overall and the camera cannot allocate enough memory for buffering. I set the video quality to lowest...

I experienced this bug while developing a new functionality for the camera, which allows users to set the frame per second. My phone (Oppo CPH1729) would not have enough memory to record at 40fps per second. The yellow box message `Starting video recording failed. Another recording might be in progress` would show up. Recording at 30fps would work witout problems. Thanks

[1]: https://github.com/react-native-community/react-native-camera/issues/2498#issuecomment-549770110

zvs001, 2020-09-03:
test lib: react-native-camera 3.38.0
test device: Xiomi with android 8

Reproduced this issue during development.
1) I don't know exactly why it happened. I recorded videos many time.
2) When error appeared, it continue throwing on every try. Even after app close. But default camera app works ok.
3) I deleted and installed app again, but it didn't help
4) Then I restarted system and issue was finally solved.

closed 2021-01-13 by stale[bot]

Gurpreet31-cpu, 2021-02-12:
Try this in Android manifest file->   android:hardwareAccelerated="true"

ShakeelAhmad-dev, 2021-02-16:
Facing this Issue on Android 10!

Hamza6244, 2021-05-27:
I'm still facing this issue in android

lucasluca, 2021-06-15:
I just fixed fps to 30 and resolved my problem. Like this:

```
const options: RecordOptions = {
          quality: '480p',
          maxDuration: 15,
          fps: 30,
          mute: false,
        };
```

funbook-memes, 2021-07-12:
@lucasluca No it didn't work for me.

Anyone have solution for this issue??

shankar81, 2021-07-12:
@lucasluca No it didn't work for me.

Anyone have solution for this issue??

lucasluca, 2021-07-14:
@shankar81 do you have this permission on your Android Manifest?

 <uses-permission android:name="android.permission.CAMERA" />

I'm using these options when recording: 

` const options: RecordOptions = {
          quality: '480p',
          maxDuration: 15,
          fps: 30,
          codec: 'H264',
          mute: false,
        };
        const data: RecordResponse = await cameraRef?.current?.recordAsync(
          options,
        );`

And these when taking picture:

` const options = {
          quality: 0.4,
          skipProcessing: true,
          autoFocus: true,
          base64: true,
          doNotSave: true,
        };
        cameraRef.current
          .takePictureAsync(options)
          .then((data: any) => {...`

My component looks like this:

`<RNCamera
              ref={cameraRef}
              style={{flex: 1}}
              // captureQuality={Camera.constants.CaptureQuality['720p']}
              zoom={zoom}
              captureAudio={true}
              maxZoom={MAX_ZOOM}
              type={RNCamera.Constants.Type.back}
              autoFocus={RNCamera.Constants.AutoFocus.on}
              flashMode={RNCamera.Constants.FlashMode.off}
            />`

Hope it helps

vicky123saini, 2021-11-01:
Remove these attr from AndroidManifest.

android:hardwareAccelerated="false"
android:largeHeap="true"