react-native-camera/react-native-camera #2503: Recognize text on the picture?

from74, 2019-09-24:
Hello
Is there a way to recognize text on the picture (after make a photo), not in real time mode?

simistern, 2019-09-25:
Yes, if you check the example code you can see an implementation: 
[Text and Facial Recognition example](https://github.com/react-native-community/react-native-camera/blob/master/examples/mlkit/App.js)

closed 2019-09-25 by simistern

from74, 2019-09-25:
> Yes, if you check the example code you can see an implementation
Where? Only the recognition function in live mode is implemented here, with button that turn on the recognition, or not?

simistern, 2019-09-25:
Ah I had misunderstood, I don't think so since once you've taken the picture you'd use different tools to render/save it, and can implement text recognition from there. You can read here for more info: [React Native Camera Docs and API](https://react-native-community.github.io/react-native-camera/docs/rncamera)