react-native-camera/react-native-camera #2506: How to get live frames from a camera preview??

shreyakamoji96, 2019-09-25:
I tried implementing this in my project to recognize fruits from camera preview in real-time using TensorFlow. Instead of grabbing the frames and converting it to jpeg format or base64 string and passing it to javascript, I grabbed the frames via TextureView.getBitmap() from CameraView and feed it to TensorFlow for image recognition and passed the results to Javascript. It's fast and could process frames in real-time. If you're going to convert the byte[] data from onFramePreview to other formats, it may take a while to pass it to javascript.

How to capture live frames in javascript

ShaharyarMaroof, 2019-09-25:
Hey @shreyakamoji96 , can you please let me know how you achieved this? Can you please share the code to pass frames from live preview to tensorflow? Thanks.

sibelius, 2019-09-25:
Check this https://github.com/react-native-community/react-native-camera/pull/2505

shreyakamoji96, 2019-09-26:
> Hey @shreyakamoji96 , can you please let me know how you achieved this? Can you please share the code to pass frames from live preview to tensorflow? Thanks.

I have just implemented camera live preview, not grabbing the live frames. Actually even I wanted the live frames that was the question. If you get any solution please let me know too. Thanks

awais-devden, 2019-11-19:
@shreyakamoji96 can you please tell me how did you implemented this camera live preview :/ ?

closed 2020-01-25 by stale[bot]

rafaelcavalcante, 2020-09-29:
> @shreyakamoji96 can you please tell me how did you implemented this camera live preview :/ ?

Have you found any solution, @awaishere? I've opened an issue ( #2984 ) but still no response.

awaishere, 2020-10-05:
Nope, i haven't found it yet.

rafaelcavalcante, 2020-10-05:
> Nope, i haven't found it yet.

I'm using [this fork](https://github.com/react-native-community/react-native-camera/pull/2505) and taking pictures in a specific interval, then i use the taken images.

AdarshJais, 2021-04-26:
> Nope, i haven't found it yet.

Still ?

klayverxd, 2022-05-25:
> Nope, i haven't found it yet.

Still?

wallride, 2022-11-24:
Still. :(

shahnupur1901, 2023-02-10:
Still?