react-native-camera/react-native-camera #2507: Pause/Play Functionality

BaderSerhan, 2019-09-25:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- iOS: 12.3.1
- react-native-camera: 3.6.0
- react-native: 0.60.5
- react: 16.9.0

**Expected Behaviour**

I want to be able to pause the video recording, then proceed within the same video. From the docs, I read that we can provide the `options` object with a `path` prop to achieve this.

> - path (file path on disk). Specifies the path on disk to record the video to. You can use the same uri returned to continue recording across start/stops

**Steps to Reproduce**

So, what I did was start a video recording with `recordAsync`. When I press the pause button, I stop the recording and save the `uri` gotten. When I press play again, I call `recordAsync` again, this time with the `path` prop set to the saved `uri`. Then when I press stop, I display the final `uri`.

**Description/Current Behaviour**

The final `uri` was of the format `file:///file:/var/...` while it should be `file:///var/...`.
Upon removing the extra `file:/` programmatically, the video display is the one recording after pressing the pause and play button, i.e. the functionality is not achieved.

Any help in how to achieve the pause/play functionality is highly appreciated.

michaelfreund, 2019-12-15:
@BaderSerhan We're facing the same problem. Have you found a solution?

BaderSerhan, 2019-12-16:
@michaelfreund I have checked the issue you referenced, it is indeed the same problem that I am facing. Unfortunately, I did not find a solution and decided to postpone this feature for now.

matiborr, 2020-04-03:
Does anyone have a solution to this problem?

closed 2020-06-09 by stale[bot]

happyfloat, 2021-02-04:
Please reopen... I'm also facing the same issue. I'm not able to upload or play the resulting path after multiple start/stops.

But I think the capture works because if I set a video length limit, it adds up nicely and stops the video after multiple start/stops because of the length.

riveraj33, 2021-06-06:
Any solution to this? Is this not something we can do with react native? I know flutter allows this. Really need a solution for react native though.

adrienfloor, 2021-07-28:
Hey guys ! Did any of you find a workaround for this ? Thanks a lot !