react-native-camera/react-native-camera #2508: Neither onCameraReady nor onMountError is called although camera preview has started.

gaurav321garg, 2019-09-25:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS: 12.3
- react-native-camera: master
- react-native: 0.60.4
- react: 16.8.6
  

**Description/Current Behaviour**
Neither onCameraReady nor onMountError is called although camera preview has started.

**Expected Behaviour**
If camera is mounted and showing preview then onCameraReady function should have been executed already and otherwise if there is some issue with mounting in that case onMountError should have been called. 

**Steps to Reproduce**
Frequently mount and unmount the RNCamera.

Everything happens as expected except if we open and close a View containing RNCamera frequently like 30-40 times within few minutes then camera shows pendingAuthorizationView for a while and then camera preview starts, in this case onCameraReady is not triggered.

**Additionals**

 Before version 3.4.0 our app used to crash in production which was fixed in this issue #2056 
but now the app does not crash but without status change I am unable to call takePictureAsync on the camera instance 

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2019-10-01:
@sibelius  this looks like an iOS issue, not Android.

@gaurav321garg does it only happen after opening the camera about 30-40 times? It works fine initially? I haven't seen it happen.

gaurav321garg, 2019-10-17:
@cristianoccazinsp yes this happen after 30-40 times and work fine initially.
Surprisingly this does not replicate if I disable face detection from the code.

closed 2019-12-23 by stale[bot]