react-native-camera/react-native-camera #2509: takePictureAsync() returns undefined

BitterDoneWork, 2019-09-25:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [x ] Did you try master?
- [ x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 10.0 emulated device
- iOS: n/a
- react-native-camera: 3.4.0
- react-native: 0.60.4
- react: 16.8.6
  

**Description/Current Behaviour**

Calling this.camera.takePictureAsync() via onPress makes the shutter sound, appears to freeze the camera, and returns an undefined object

**Expected Behaviour**

Calling this.camera.takePictureAsync() via onPress takes a picture and returns the uri of the resulting file
Camera's view should continue to show a live feed of the emulated room

**Steps to Reproduce**

Create a new basic react-native-camera app by following the doc:
https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md
Update slightly to mirror the code below
Run the code in an Android emulator, Android version 10, 4 GB ram
Click the SNAP button and observe the bug behavior

Checking Expo is a WIP, unsure as of now

**Additionals**

`import React, { useCallback,  } from 'react';
import { Navigation } from 'react-native-navigation';
import {
  StyleSheet,
  View,
  Text,
  TouchableOpacity,
} from 'react-native';
import { RNCamera } from 'react-native-camera';`

`  takePicture = async() => {
    if (this.camera) {
      const options = { quality: 0.5, base64: true };
      const data = await this.camera.takePictureAsync() // options)
        .then(data => console.log('data', data))
        .catch(error => console.log('eror', error));
      console.log('Exiting takePicture()');
    }
  };`

`export function CaptureImage({ componentId }) {
  const handlePushScreen = useCallback(() => {
    Navigation.push(componentId, {
      component: {
        name: 'Home',
      },
    });
  }, [componentId]);
  console.log('Capturing image');

  return (
      <View style={styles.body}>
        <RNCamera
          ref={ref => { this.camera = ref; }}
          style={{ flex: 1, width: '100%', }}
          androidCameraPermissionOptions={{
            title: 'Permission to use camera',
            message: 'We need your permission to use your camera',
            buttonPositive: 'Sure',
            buttonNegative: 'Cancel',
          }}
        >
        </RNCamera>
        <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
          <TouchableOpacity onPress={this.takePicture.bind(this)} style={styles.capture}>
            <Text style={{ fontSize: 14, backgroundColor: 'white', textAlign: 'center', }}> SNAP </Text>
          </TouchableOpacity>
        </View>
      </View>
  );
}`
> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

BitterDoneWork, 2019-09-25:
I'll update react-native-camera from 3.4 to 3.6 in the morning and try that. Leaving this here for visibility

BitterDoneWork, 2019-09-26:
The behavior on react-native-camera v3.6.0 is the same as v3.4.0. I'm attempting a basic Expo app with RN Camera to see if the issue is there as well

sibelius, 2019-09-26:
in which version this broke?

BitterDoneWork, 2019-09-26:
I'm not sure which piece you're asking about, but here's all the versions I can think of that are being used:

Android: 10.0 emulated device
iOS: n/a
react-native-camera: 3.6.0 (tested 3.6 this morning)
react-native: 0.60.4
react: 16.8.6

BitterDoneWork, 2019-09-26:
Apparently I can't upload a .js file, so here's the contents of my App.js:
`import React, { useState, useEffect } from "react";
import { StyleSheet, Text, View, TouchableOpacity } from 'react-native';
import * as Permissions from 'expo-permissions';
import { Camera } from 'expo-camera';

const App = () => {
  const [type, setType] = useState(Camera.Constants.Type.back);
  const [hasCameraPermission, setHasCameraPermission] = useState(true);

  useEffect(() => {
    Permissions.askAsync(Permissions.CAMERA)
      .then(data => {
        console.log('data');
        console.log(data);
        return data.status;
      })
      .then(status => {
        console.log('status === granted', status === 'granted');
        setHasCameraPermission(status === 'granted');
      })
      .catch(error => console.log('eror'));
  }, []);

  const flipCamera = (type) =>
    setType(type === Camera.Constants.Type.back
      ? Camera.Constants.Type.front
      : Camera.Constants.Type.back);

  if (hasCameraPermission === null) {
    console.log('Null camera permission');
    return <View style={styles.container}><Text>Null camera permissions</Text></View>;
  } else if (hasCameraPermission === false) {
    console.log('No access to camera');
    return <View style={styles.container}><Text>No access to camera</Text></View>;
  } else {
    console.log('Rendering camera w type ', type);
    return (
      <View style={{ flex: 1 }}>
        {console.log('Camera type: ', type)}
        <Camera
          style={{ flex: 1 }}
          type={type}
        >
          <View
            style={{
              flex: 1,
              backgroundColor: 'transparent',
              flexDirection: 'row',
            }}>
            <TouchableOpacity
              style={{
                flex: 0.1,
                alignSelf: 'flex-end',
                alignItems: 'center',
              }}
              onPress={() => flipCamera(type)}>
              <Text style={{ fontSize: 18, marginBottom: 10, color: 'white' }}> Flip </Text>
            </TouchableOpacity>
          </View>
        </Camera>
      </View>
    );
  }  
};

const styles = StyleSheet.create({
  container: {
    flex: 1,
    backgroundColor: '#fff',
    alignItems: 'center',
    justifyContent: 'center',
  },
});

export default App;`

brunosantanarj, 2019-09-27:
Same issue

MateusAndrade, 2019-09-30:
@brunosantanarj have you tried with the example project? I tested it here and it works fine. You can provide your code to help with that?

BitterDoneWork, 2019-09-30:
Happy update: I no longer have any issue with RN Camera

I'm guessing my issue was some configuration with my AVD, where the app couldn't save the picture to local storage, and thus returned an undefined uri. My workplace hasn't provided a development phone, but I was able to deploy to my personal device and see the console.log(data.uri) work.

From there, it was a matter of using RNFS to move the picture to DCIM/myAppName/ and see it in the My Files app.

@brunosantanar , @MateusAndrade, I'm closing this ticket since it's working on my end. Thank you all for the replies!

HappyCMD, 2019-10-02:
@BitterDoneWork 

How did you move it with RNFS, I'm getting source file not found when I try to move the cached image, the following displays the image:

                    <Image 
                        source={{uri: data.uri}}
                    />

However, removing the file:// from the data.uri says source file not found..

MateusAndrade, 2019-10-02:
@HappyCMD are you trying to move the picture take? You will need the ```file://``` because this is acessing the internal storage on your application.

BitterDoneWork, 2019-10-02:
@HappyCMD 
I have this method attached to a "Take Picture" button:
```
const takePicture = async() => {
  if (this.camera) {
    const options = { quality: 0.5, base64: true };
    const data = await this.camera.takePictureAsync(options)
      .then(data => {
        console.log('data uri:' + data.uri);
        saveImage(data.uri);
      });
  }
};
```
then saveImage makes the file name and calls the RNFS move method:
```
const saveImage = async filePath => {
  try {
    const newImageName = `${moment().format('DDMMYY_HHmmSSS')}.jpg`;
    const newFilepath = `${dirHome}/${newImageName}`;
    console.log('Saving file as ', newImageName);
    console.log('At path ', newFilepath);
    // move and save image to new filepath
    const imageMoved = await moveAttachment(filePath, newFilepath);
    console.log('image moved', imageMoved);
  } catch (error) {
    console.log(error);
  }
};
```
```
const moveAttachment = async (filePath, newFilepath) => {
  return new Promise((resolve, reject) => {
    console.log('mkdir dirHome then')
    RNFS.moveFile(filePath, newFilepath)
      .then(() => {
        console.log('FILE MOVED', filePath, newFilepath);
        resolve(true);
      })
      .catch(error => {
        console.log('moveFile error', error);
        reject(error);
      });
  });
};
```

BitterDoneWork, 2019-10-02:
I just noticed I left out an import- this is my dirStorage.js. I'm not sure if I needed to, but I manually created the AwesomeApp folder in DCIM.
```
import { Platform } from 'react-native';
const RNFS = require('react-native-fs');

const loc = '/DCIM/AwesomeApp';
export const dirHome = Platform.select({
  ios: `${RNFS.DocumentDirectoryPath}`.concat(loc),
  android: `${RNFS.ExternalStorageDirectoryPath}`.concat(loc)
});

export const dirPicutures = `${dirHome}/Pictures`;
// export const dirAudio = `${dirHome}/Audio`;
```

BitterDoneWork, 2019-10-02:
Also, I've only tested on my Android device. I don't have the setup for iOS yet

BitterDoneWork, 2019-10-03:
2nd also, this code isn't mine- I took/adapted from Aman Agarwal's Medium post:
https://medium.com/@masochist.aman/capturing-images-with-react-native-203e24f93eb9

grkemtneri, 2019-10-11:
Same issue, i'm using "react-native": "0.61.2", and  "react-native-camera": "^3.7.1","react-native-fs": "^2.14.1".

```
import React, {PureComponent} from 'react';
import {StyleSheet, Text, TouchableOpacity, View} from 'react-native';
import {RNCamera} from 'react-native-camera';
var RNFS = require('react-native-fs');

import {dirPicutures} from './src/utils/dirStorage';

const moveAttachment = async (filePath, newFilepath) => {
  return new Promise((resolve, reject) => {
    RNFS.mkdir(dirPicutures)
      .then(() => {
        RNFS.moveFile(filePath, newFilepath)
          .then(() => resolve(true))
          .catch(error => reject(error));
      })
      .catch(err => reject(err));
  });
};

export default class App extends PureComponent {
  saveImage = async filePath => {
    try {
      // set new image name and filepath
      const newImageName = `merhabalar.jpg`;
      const newFilepath = `${dirPicutures}/${newImageName}`;
      const imageMoved = await moveAttachment(filePath, newFilepath);
      console.log('image moved', imageMoved);
    } catch (error) {
      console.log(error);
    }
  };

  takePicture = async () => {
    if (this.camera) {
      const options = {
        quality: 0.5,
        base64: true,
        forceUpOrientation: true,
        fixOrientation: true,
      };
      const {uri} = await this.camera.takePictureAsync(options);
      console.log(uri);
    }
  };

  render() {
    return (
      <View style={styles.container}>
        <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          captureAudio={false}
          style={styles.preview}
          androidCameraPermissionOptions={{
            title: 'Permission to use camera',
            message: 'We need your permission to use your camera',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
        />
        <View style={{flex: 0, flexDirection: 'row', justifyContent: 'center'}}>
          <TouchableOpacity
            onPress={this.takePicture.bind(this)}
            style={styles.capture}></TouchableOpacity>
        </View>
      </View>
    );
  }
}
```

teamhide, 2020-07-30:
Any update here?

closed 2020-10-09 by stale[bot]