react-native-camera/react-native-camera #2511: Recording with an invalid codec causes a hard crash

xHeinrich, 2019-09-27:
# Bug Report

**To Do First**

- [yes ] Did you try latest release?
- [yes] Did you try master?
- [yes] Did you look for existing matching issues?

**Platforms**
iOS

**Versions**
ios/ipados 12.4.1, 13.1

**Description/Current Behaviour**
Recording with the h264 codec on new devices causes a hard crash, hvec must be used. to bypass this issue I created a script which checks what device is being used and flips the codec depending.

**Expected Behaviour**
Default codec should be used

**Steps to Reproduce**

Recording with the h264 codec on new devices causes a hard crash, hvec must be used. to bypass this issue I created a script which checks what device is being used and flips the codec depending.

Feature isnt implemented on Expo Camera

cristianoccazinsp, 2019-10-01:
Are you sure there will be a hard crash? The code seems to check for it here: https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L604

xHeinrich, 2019-10-01:
Yes, it crashes with an NSInvalidArgumentException. Tried h264 on iPhone 8+.

cristianoccazinsp, 2019-10-06:
Hey @xHeinrich . I've been testing with a bunch of iOS devices (iPhone 7, iPhone 7 plus, iPhone X, iPhone 11 Pro Max) and none of them seem to be crashing with h264 turned on. 

Any chance this is an iPhone 8 issue only? Any chance you can try with this fork? https://github.com/cristianoccazinsp/react-native-camera/tree/ios-cameraids-ultrawide

closed 2019-12-12 by stale[bot]