react-native-camera/react-native-camera #2515: react-native 0.60.5 build release failed

AntonBorovoi, 2019-09-30:
I am upgrade react-native from 0.59.8 to 0.60.5
 "react-native-camera": "^3.6.0",
`android.build.gradle`
```
defaultConfig {
       vectorDrawables.useSupportLibrary true
        multiDexEnabled true
        missingDimensionStrategy 'react-native-camera', 'general'
    }
```

Debug works fine, but the release  build are failing `./gradlew assembleRelease`

```
Could not find com.google.firebase:firebase-ml-vision:12.0.1.
     Required by:
         project :react-native-camera
   > Could not find com.google.firebase:firebase-ml-vision-face-model:12.0.1.
     Required by:
         project :react-native-camera
```

How can I resolve it?

MateusAndrade, 2019-10-01:
Are you using any module related to ML Kit on Android?

AntonBorovoi, 2019-10-01:
@MateusAndrade No. I also had tried to remove `react-native-camera` and project complied successfully.

ChineseDavid, 2019-10-01:
Do you have an example of react-native-camera in react-native 0.6+,I just can't run it.
I install it,but there are some faults that I can't resolve.

javastud, 2019-10-02:
@AntonBorovoi debug works fine? I am trying over 2 days cant still get it worked. Can you remember how did you installed it?

And should i "link" it since using 0.6+?

bobbyrinaldy, 2019-10-02:
can someone help this ?
[https://github.com/react-native-community/react-native-camera/issues/2486#issue-493422215](url)

AntonBorovoi, 2019-10-03:
@javastud I upgraded to `3.6.0` and unlink 'by react-native unlink react-native-camera' since autolink is supported.

I am using react-native camera only in context of `react-native-qrcode-scanner`

Jhonte, 2019-11-13:
@AntonBorovoi How does your defaultConfig look like?

I have the same problem :(   Just installed it fresh on RN 0.61.2

Trying to not use ML Kit

`Could not determine the dependencies of task ':react-native-camera:compileMlkitReleaseAidl'.
> Could not resolve all task dependencies for configuration ':react-native-camera:mlkitReleaseCompileClasspath'.
   > Could not find com.google.firebase:firebase-ml-vision:12.0.1.
     Required by:
         project :react-native-camera
   > Could not find com.google.firebase:firebase-ml-vision-face-model:12.0.1.
     Required by:
         project :react-native-camera`

BlackCat1397, 2020-01-16:
Same issue.
```
android {
  defaultConfig {
             missingDimensionStrategy 'react-native-camera', 'general'
  }
}
```

Ok in debug, `Task :react-native-camera:compileMlkitReleaseJavaWithJavac FAILED` in assembleRelease.

abdul-rightapps, 2020-02-05:
> Same issue.
> 
> ```
> android {
>   defaultConfig {
>              missingDimensionStrategy 'react-native-camera', 'general'
>   }
> }
> ```
> 
> Ok in debug, `Task :react-native-camera:compileMlkitReleaseJavaWithJavac FAILED` in assembleRelease.

Same issue, OK in debug, 

Could not determine the dependencies of task ':react-native-camera:compileMlkitReleaseAidl'. > Could not resolve all task dependencies for configuration ':react-native-camera:mlkitReleaseCompileClasspath'. in assembleRelease

closed 2020-04-12 by stale[bot]

Lloyd1229, 2020-07-04:
Could not determine the dependencies of task ':react-native-camera:compileMlkitReleaseJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':react-native-camera:mlkitReleaseCompileClasspath'.