react-native-camera/react-native-camera #2518: Improve onCameraReady and onMountError for iOS

cristianoccazinsp, 2019-10-02:
# Feature-Request

Currently, onCameraReady fires every time the camera is changed on Android (both Camera1 and Camera2). That is, when the facing or id props are changed. On iOS, however, the event is only fired once on session start and never again. It would probably be a good idea to make iOS behave the same. This could be accomplished by updating `initializeCaptureSessionInput` to also fire the event if there's a session already running. However, this change might cause issues to people that relies on the current iOS behaviour.

On the other hand, onMountError doesn't seem to be fired at all on iOS, but is fired on Android if the camera device change failed (because the camera is stopped and started). In addition, iOS makes impossible to detect camera change errors (e.g., due to an unsupported quality or invalid device ID). It might be a good idea to fire onMountError also on `initializeCaptureSessionInput` if the device fails to initialize.

Thoughts?

cristianoccazinsp, 2019-12-13:
Still an issue, haven't really had time to update this though.

cristianoccazinsp, 2019-12-13:
Added PR: https://github.com/react-native-community/react-native-camera/pull/2642

closed 2019-12-13 by sibelius

n1ru4l, 2019-12-13:
:tada: This issue has been resolved in version 3.15.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v3.15.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket: