react-native-camera/react-native-camera #2519: How do I edit exif of photo, image is flipped incorrectly in Android?

HappyCMD, 2019-10-04:
I am having issues with react-native-camera my current settings that semi-work in Android are:

`const options = { quality: 1, writeExif: true, exif: true,};`

But unfortunately, whilst it did fix orientation it now shows the image mirrored horizontally

On inspection, in the exif given shows Orientation": 8, according to the list below, it should be 7

I've tried setting pictureOrientation and deviceOrientation which are currently 1 to any other number to no avail aswell.

1 - 0 degrees – the correct orientation, no adjustment is required.
2 - 0 degrees, mirrored – image has been flipped back-to-front.
3 - 180 degrees – image is upside down.
4 - 180 degrees, mirrored – image is upside down and flipped back-to-front.
5 - 90 degrees – image is on its side.
6 - 90 degrees, mirrored – image is on its side and flipped back-to-front.
7 - 270 degrees – image is on its far side.
8 - 270 degrees, mirrored – image is on its far side and flipped back-to-front.

HappyCMD, 2019-10-04:
I got it working with: const options = { quality: 1, exif: true, writeExif: true, mirrorImage: true, orientation: "portraitUpsideDown"}

closed 2019-10-04 by MateusAndrade

mjmon, 2020-01-07:
> I got it working with: const options = { quality: 1, exif: true, writeExif: true, mirrorImage: true, orientation: "portraitUpsideDown"}

this works for Android(Samsung Galaxy A8) but not on my iPhone

It is literally upside down on iPhone without applying the mirror although it was set to true :(