react-native-camera/react-native-camera #2521: Front camera is not working in android 10

ajitesh-itobuz, 2019-10-04:
The rear and front camera is working properly on below android 10 devices but the front camera is not opening in android 10(Google Pixel 3).
The rear camera works fine.

I use it like below
`        <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={styles.preview}
          type={
            this.state.cameraFacing === 'back'
            ? RNCamera.Constants.Type.back
            : RNCamera.Constants.Type.front
          }
          mirrorImage={this.state.mirrorMode}
          flashMode={
            this.state.flash
            ? RNCamera.Constants.FlashMode.on
            : RNCamera.Constants.FlashMode.off
          }
          androidCameraPermissionOptions={{
            title: 'Permission to use camera',
            message: 'We need your permission to use your camera',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
          captureAudio={false}
        />`

cristianoccazinsp, 2019-10-06:
Have you tried without setting the mirrorImage prop?

I haven't had issues on Android 10 with a Pixel 2, but I guess Pixel 3 might be different.

For debugging only, have you tried using `getCameraIdsAsync` and setting the `cameraId` property to select the front camera instead of using the `type` prop?

You can easily use the above method and prop to loop through all cameras of the devices.

Lastly, have you tried with `useCamera2Api={true}` ?

Please provide some android/logcat logs to see if there's something being rejected internally since it is very hard to return errors on props changes.

noambonnie, 2020-03-11:
I'm observing the problem on Android 10, Samsung Galaxy S10, it only happens with `useCamera2API={true}`. Works fine with default camera API.

Calling getCameraIdsAsync returns:
`[{"id": "0", "type": 0}, {"id": "1", "type": 1}, {"id": "2", "type": 0}, {"id": "3", "type": 1}]`

I tried passing '0'-'3' as cameraId, only '0' works and shows the rear camera. Other IDs show a black preview.

Any thoughts? I'm happy with using the old camera API but I need to mute shutter sound using `playSoundOnCapture` but that prop is only available with camera2 (as far as I can tell).

cristianoccazinsp, 2020-03-11:
Interesting. You are getting ids from 1 to 3 using the camera 2 API?  I believe the Camera 2 API returns strings (more like uuids). Can you verify that you're calling `getCameraIdsAsync` with a Camera2 instance? Can you try calling that method with a short timeout/delay to see if you get the same results?  The reason I ask this, is because I know Camera2 is initialized in an odd way (it first tries with Camera1 and "upgrades" to Camera2), so you might be having a race condition.

However, if not using `cameraId` also fails, it might be something else. I will try to debug once I get some free time. 

In the meantime, you could try adding some breakpoints and debug in here https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera2.java#L725 and https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera2.java#L404

noambonnie, 2020-03-12:
I double checked that I _am_ using camera2 api when getting camera IDs. 

I set a breakpoint as you suggested. I noticed that `chooseCameraIdByFacing` is being called twice - first time it's setting mCameraId to 0, and preview of the rear camera appears in the app. The second time the selected camera is id=1, which I guess is the front facing, and maybe it fails because the other camera is already initialized and there's some race condition where the rear cam has not been release first?

Here are some logcats (filtered for "cam") from launch of the app to trying to access the front camera:

```
2020-03-12 08:36:27.102 1028-2099/? I/CameraService: onUidStateChanged: uid=10031, procState=16
2020-03-12 08:36:27.940 1028-2099/? I/CameraService: onUidStateChanged: uid=10112, procState=12
2020-03-12 08:36:27.940 1028-2099/? I/CameraService: onUidStateChanged: uid=10031, procState=9
2020-03-12 08:36:27.946 1028-27112/? I/CameraService: onUidStateChanged: uid=10112, procState=20
2020-03-12 08:36:29.393 894-17682/? V/APM_AudioPolicyManager: computeVolume : camera shutter sound volume 1.000000
2020-03-12 08:36:29.431 1028-27112/? I/CameraService: UidPolicy: onUidIdle, uid = 10344
2020-03-12 08:36:29.431 1028-27112/? I/CameraService: onUidStateChanged: uid=10344, procState=20
2020-03-12 08:36:29.544 1028-2099/? I/CameraService: UidPolicy: onUidActive, uid = 10344
2020-03-12 08:36:29.544 1028-2099/? I/CameraService: onUidStateChanged: uid=10344, procState=2
2020-03-12 08:36:29.920 1028-27112/? I/CameraService: onUidStateChanged: uid=99753, procState=6
2020-03-12 08:36:29.920 1028-27112/? I/CameraService: onUidStateChanged: uid=99749, procState=6
2020-03-12 08:36:29.920 1028-27112/? I/CameraService: onUidStateChanged: uid=10216, procState=6
2020-03-12 08:36:29.920 1028-27112/? I/CameraService: notifyMonitoredUids
2020-03-12 08:36:29.920 1028-27112/? I/CameraService: onUidStateChanged: uid=10045, procState=6
2020-03-12 08:36:30.031 7899-7976/com.rncaissonsdk W/unknown:ViewManagerPropertyUpdater: Could not find generated setter for class com.lwansbrough.RCTCamera.RCTCameraViewManager
2020-03-12 08:36:30.032 7899-7976/com.rncaissonsdk W/unknown:ViewManagerPropertyUpdater: Could not find generated setter for class org.reactnative.camera.CameraViewManager
2020-03-12 08:36:30.312 1028-2099/? I/CameraService: onUidStateChanged: uid=10031, procState=16
2020-03-12 08:36:32.501 894-9976/? V/APM_AudioPolicyManager: computeVolume : camera shutter sound volume 1.000000
2020-03-12 08:36:33.997 894-9976/? V/APM_AudioPolicyManager: computeVolume : camera shutter sound volume 1.000000
2020-03-12 08:36:34.518 7899-7899/com.rncaissonsdk I/CameraManagerGlobal: Connecting to camera service
2020-03-12 08:36:34.522 1028-2099/? D/CameraService: addListenerHelper: Add listener 0xecfe9940
2020-03-12 08:36:34.522 1028-2099/? D/CameraService: CameraDeviceState for CameraManager
2020-03-12 08:36:34.522 7899-7899/com.rncaissonsdk I/CameraManagerGlobal: postSingleUpdate device: camera id 0 status STATUS_PRESENT
2020-03-12 08:36:34.523 7899-7899/com.rncaissonsdk I/CameraManagerGlobal: postSingleUpdate device: camera id 1 status STATUS_PRESENT
2020-03-12 08:36:34.523 7899-7899/com.rncaissonsdk I/CameraManagerGlobal: postSingleUpdate device: camera id 2 status STATUS_PRESENT
2020-03-12 08:36:34.523 7899-7899/com.rncaissonsdk I/CameraManagerGlobal: postSingleUpdate device: camera id 3 status STATUS_PRESENT
2020-03-12 08:36:34.523 7899-7899/com.rncaissonsdk E/CAMERA_2::: Adjusting exposure is not currently supported for Camera2
2020-03-12 08:36:34.523 7899-7929/com.rncaissonsdk I/CameraManagerGlobal: Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.rncaissonsdk API Level 2
2020-03-12 08:36:34.524 7899-7929/com.rncaissonsdk I/CameraManagerGlobal: Camera 1 facing CAMERA_FACING_FRONT state now CAMERA_STATE_CLOSED for client com.samsung.android.smartface API Level 2
2020-03-12 08:36:34.524 7899-7929/com.rncaissonsdk I/CameraManagerGlobal: Camera 2 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.sec.android.app.sbrowser API Level 2
2020-03-12 08:36:34.524 7899-7929/com.rncaissonsdk I/CameraManagerGlobal: Camera 20 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.sec.android.app.camera API Level 2
2020-03-12 08:36:34.524 7899-7929/com.rncaissonsdk I/CameraManagerGlobal: Camera 21 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client android.system API Level 2
2020-03-12 08:36:34.524 7899-7929/com.rncaissonsdk I/CameraManagerGlobal: Camera 23 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client android.system API Level 2
2020-03-12 08:36:34.524 7899-7929/com.rncaissonsdk I/CameraManagerGlobal: Camera 3 facing CAMERA_FACING_FRONT state now CAMERA_STATE_CLOSED for client android.system API Level 2
2020-03-12 08:36:34.524 7899-7929/com.rncaissonsdk I/CameraManagerGlobal: Camera 40 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client android.system API Level 2
2020-03-12 08:36:34.525 7899-7929/com.rncaissonsdk I/CameraManagerGlobal: Camera 41 facing CAMERA_FACING_FRONT state now CAMERA_STATE_CLOSED for client android.system API Level 2
2020-03-12 08:36:34.525 7899-7929/com.rncaissonsdk I/CameraManagerGlobal: Camera 50 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client android.system API Level 2
[... cut ...]
2020-03-12 08:36:34.576 1028-5843/? I/CameraService: wouldEvictLocked:getCurrentCostLocked()=100, cost=100, totalCost=200, MaxCost=100
2020-03-12 08:36:34.576 1028-5843/? I/CameraService: wouldEvictLocked: owner=7899, priority-score=0, highestPriorityOwner=7899, highestPriority-score=0
2020-03-12 08:36:34.576 1028-5843/? E/CameraService:    Conflicts with: Device 0, client package com.rncaissonsdk (PID 7899, score 0, state 2)
2020-03-12 08:36:34.577 7899-8024/com.rncaissonsdk E/Camera2: onError: 1 (2)
2020-03-12 08:36:35.057 1028-5843/? I/CameraService: onUidStateChanged: uid=10031, procState=16
```

noambonnie, 2020-03-12:
FWIW, it _does_ work on the Pixel 3a, which only has a single rear camera.

cristianoccazinsp, 2020-03-12:
So this is only an issue of the Pixel 3? Just to confirm, the same issue happens regardless of using the `cameraId` prop or just setting the `facing` prop right?  If that's the case, then you are most likely right, the issue is related by the camera switching process. I have tested on a Pixel 2 and I definitely don't see the issue so I can't really test it.

Would it be possible for you to test adding some delay to the camera switch step (in Java) to confirm that's the issue?  Perhaps test some synchronized calls? I'm sorry I can't be of much help!

Lastly, this is NOT an issue with Camera1? Both implementations are very similar... so that's odd.

noambonnie, 2020-03-12:
No, Pixel 3a is fine. Only an issue with Samsung Galaxy S10.  

I'm unsure what 'facing' prop is. I'm using `type={RNCamera.Constants.Type.front}`

It is _not_ an issue with camera1 indeed. 

Any way you can guide me as to where to set the delay?

cristianoccazinsp, 2020-03-12:
Sorry, by facing I meant the `type` prop. 

You could try delaying the start/stop calls to see if that's the problem. For example:

https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera2.java#L383

and https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera2.java#L362

Those two places stop and start the camera when either the ID or Type is changed. 

To add a delay, you may try delaying the `start` call with something like what Camera1 does: https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L1208

Note that the above is just to confirm the issue, adding such delay on the stop/start step would actually be quite bad!

noambonnie, 2020-03-12:
For all I could tell that code is not being called. I will try to dig deeper into that tomorrow. I'm pretty sure it has to do with the destruction/construction order. Maybe something about this phone's camera array is different.

noambonnie, 2020-03-13:
I was able to track down the problem, not yet the solution (and any guidance/help would be welcomed!)

The flow as it happens is:
- `setUsingCamera2Api` is being called and in turn calling `onRestoreInstanceState()` which then calls setFacing() with the default (0 / Rear camera)
- `start()` is called. 
- `setFacing` is called to set the camera to front / 1. But, `onOpened()` has not yet been called as a result of the previous step. `setFacing()` checks `isCameraOpened()` in order to switch camera, but since the camera has not yet been opened it does nothing except setting mFacing to 1.

Below are some logs with prints I sprinkled around the code. In onOpened I added the following code to print the what the actual camera that was opened is using - front or back:

```
try {
    CameraCharacteristics characteristics = mCameraManager.getCameraCharacteristics(mCamera.getId());
    Integer internal = characteristics.get(CameraCharacteristics.LENS_FACING);
    Log.d(TAG, "onOpened: current camera facing: " + (internal == CameraCharacteristics.LENS_FACING_FRONT ? Constants.FACING_FRONT : Constants.FACING_BACK));
} catch(CameraAccessException e) {
    Log.d(TAG, "onOpened: CameraAccessException");
}
```

```
CameraView: setUsingCamera2Api calling onRestoreInstanceState():
CameraView: onHostResume calling start():
Camera2: start: Camera opening. mFacing = 0
Camera2: startOpeningCamera: mFacing = 0
Camera2: setFacing: setting mFacing from 0 to 1
Camera2: setFacing: Camera not yet opened, setFacing Nan.
RNCameraView: onHostResume calling start():
Camera2: start: Camera opening. mFacing = 1
Camera2: onOpened: camera opened. mFacing = 1
Camera2: onOpened: current camera facing: 0 <<< this is the actual facing of mCamera, even though mFacing is 1. See note above
Camera2: startOpeningCamera: mFacing 1
```

Possible solutions as I see them
1. Force `setFacing()` to be called after camera was opened.
2. Somehow make `onRestoreInstanceState()` default to the camera that was set in the props.
3. in `onOpened()`, set the actual facing value to mFacing

I've never developed a native RN component and am very rusty on Java. I'd be happy to hear thoughts/guidance.

cristianoccazinsp, 2020-03-13:
That looks like a bug, `onRestoreInstanceState` should set everything exactly as it was originally given.

cristianoccazinsp, 2020-03-13:
I see that both facing and cameraId are stored and restored, what is the problem there? Are you sure of those findings? 

https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/CameraView.java#L251

https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/CameraView.java#L272

noambonnie, 2020-03-13:
I may have a wrong phrasing but the facts are correct: `onRestoreInstanceState` holds 0 (back camera) when launching the app. Not sure what state is being restored here as I'm testing this on a fresh launch of the app.

So even though in React Native the camera is set to 1/front, when it first initializes it starts with 0, then calls setFacing with 1.

cristianoccazinsp, 2020-03-13:
I believe you are correct. Although the state is properly saved, it is saving react's default values (0).  This is how react works, your component is initialized with default values and then the setters are called.  This would probably discard your option number 2.

noambonnie, 2020-03-13:
Yes, I also wondered about the design decision to start the camera based on
a prop (`useCamera2`). Following what you said, that RN actually controls
setting the props - this leads to the camera being started before other
props like 'facing' were set. If we're able to control the order it might
resolve the issue.

NOTE: I have a misleading print in the log - the first `onHostResume
calling start():` is in fact `setUsingCamera2Api` calling `start`. Which
leads me to my next concern.

noambonnie, 2020-03-13:
Is there a way to force prop initialization order?

cristianoccazinsp, 2020-03-13:
Not that I'm aware of.  Is it possible to look at Camera1's implementation to see what's different? I would bet there's a lock/synchronized block missing there. Camera2 is still quite bad and I wouldn't be surprised. Most of the improvements I did myself were all Camera1 (faster loading and capturing, not blocking the UI, race conditions, crashes, etc.) so Camera2 is still quite broken.

noambonnie, 2020-03-13:
Yes, there's a lot of `synchronized(this)` in Camera1... I don't know that I have the ability to understand which points in Camera2 need to be sync'ed. I might take a stab at this.

While we're here though, maybe you would know a solution to the problem that got me into this in the first place... Is there a way to mute the shutter sound using Camera1?

cristianoccazinsp, 2020-03-13:
Not that I'm aware of. Since iOS forces you to have a shutter sound I just left it as a default behaviour. 

What prop are you using exactly to mute the shutter sound? I can't even see that feature in the JS side.

However, looks like Camera1 might be able to mute the shutter sound as well, but it is not implemented: https://developer.android.com/reference/android/hardware/Camera#enableShutterSound(boolean)

noambonnie, 2020-03-13:
There's an undocumented prop: https://github.com/react-native-community/react-native-camera/pull/1441/files

In any event, I think I found a solution and I don't think there's a problem with it. I simply removed the call to `start()` from `setUsingCamera2Api()`. So remove this line and everything works: https://github.com/react-native-community/react-native-camera/blob/fb2105c7ccb456e721058bf09f0347eeb492e98c/android/src/main/java/com/google/android/cameraview/CameraView.java#L314

Now `start()` is being called from `onHostResume()` after all props have been set, including `setFacing`.

I think this is fine and is actually the right thing to do. I don't think that setting this prop should directly start the camera. And this prop in particular is not something that I think should be modified after initial creation of RNCamera.

Are you part of the project? Do you think this fix would be accepted if I create a pull request?

cristianoccazinsp, 2020-03-13:
Interesting, I don't see how `playSoundOnCapture` works only in Camera2 and not in Camera1, neither implementations have specific code for that prop.  Are you 100% sure that prop is actually doing anything? I think what you're seeing is the default behaviour of Android 10 to always have a shutter sound by default (which for some reason is different with Camera2)

Also, the change you propose seems reasonable. However, I would make sure to test it thoroughly first since it could break Camera2 implementation for some devices that do not support it and then fallback to Camera1 again ( I believe that's the reason for the start call there).

Lastly, while you are at it, it could be good to include a fix to the Camera1 implementation so shutter sound is honored in Android 10 :)

noambonnie, 2020-03-13:
I see your point that the prop should work in either camera. I'm pretty sure I read it somewhere but I can't track it down right now. I tested and it doesn't work with Camera1 - maybe Camera1 _always_ plays a sound. I'm trying to mute it and the prop won't do with Camera1. I tested it on Camera2 and it works. 

I would see about testing on older API levels for fallback. Best I can do probably is emulator. Looking at the code though I don't think there will be impacted because the top of the function tests for API level and returns if Camera2 is not supported.

I may or may not get to the Camera1 sound, and if I do it would be a separate PR. 

Thanks for your help. If you have any further thoughts/ideas I'd be happy to hear.

cristianoccazinsp, 2020-03-13:
`maybe Camera1 always plays a sound.`  This is exactly my point, I think this is true for Android 10. If you test Camera1 on multiple devices, you will see some Android phones will play a sound, and some won't, regardless of the prop value.

I think the best "fix" would be to move the prop down the Camera1/2 implementation and have each class play (or mute) the sound.  Like I said above, for Camera1 the change should be rather easy https://developer.android.com/reference/android/hardware/Camera#enableShutterSound(boolean)  and for Camera2 I guess you are forced to play the sound by yourself like it currently happens. However, you are right about this change probably belonging to another separate PR.

closed 2020-05-20 by stale[bot]