react-native-camera/react-native-camera #2522: Camera Freezes on coming back to screen

devilabhi, 2019-10-04:
Steps to Reproduce

1. Screen A 5 tabs 3rd tab is having camera
2. Screen B 2 tabs 1st tab is having camera
3. so from screen A 3rd tab i.e camera I go to Screen B 1st tab camera. camera is displaying fine over here
4. when I come back to Screen A, camera is freezed. i want to rerender this tab 

could you tell how can I do this?

cristianoccazinsp, 2019-10-06:
Is this iOS or Android? Are you rendering multiple cameras at the same time? I'm not entirely sure the library was really well tested for such case, especially if recording video.

If this is iOS, you might want to give a try to my branch that has a pull request with multiple iOS fixes that might solve the issue: https://github.com/cristianoccazinsp/react-native-camera/tree/ios-cameraids-ultrawide

If this is Android, it would be good to get some logs from Android Studio / Logcat.

Lastly, if you are using react navigation or similar, I would really encourage you to not render multiple cameras at once. Not only that might have undesirable side effects, but it will also put the device under a lot of stress unnecessarily. Just disable/enable the camera on focus/blur of your scene.

devilabhi, 2019-10-07:
I resolved this by doing

`{isFocused
                ? (
                  <RNCamera
                    ref={(ref) => {
                      this.camera = ref;
                    }}
                    style={styles.camera}
                    captureAudio={false}
                    type={RNCamera.Constants.Type.back}
                  />
                )
                : null}`

closed 2019-12-13 by stale[bot]