react-native-camera/react-native-camera #2524: getCameraIdsAsync()

ReSword, 2019-10-08:
When I use getCameraIdsAsync(), I got a error this.camera.getCameraIdsAsync() is not a funciton

<RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={styles.preview}
          type={RNCamera.Constants.Type.front}
          flashMode={RNCamera.Constants.FlashMode.on}
        />

let data = await this.camera.getCameraIdsAsync();

So, How to use getCameraIdsAsync(), Thanks

cristianoccazinsp, 2019-10-08:
Hi @ReSword , is this on iOS, Android, both? I've been using it for a while now and it is definitely working for me.

sibelius, 2019-10-08:
check latest version

ReSword, 2019-10-09:
Hi @cristianoccazinsp ， this is on Android，I updated my version  now it's "react-native-camera": "^3.7.0",  but I also got the same error

sibelius, 2019-10-09:
it looks like this.camera is null

can you provide full stacktrace?

ReSword, 2019-10-09:
@sibelius I  can use  this.camera.takePictureAsync() ,but can't use getCameraIdsAsync(),
this is camera' _proto_
__proto__
:
Camera
arePermissionsGranted
:
function arePermissionsGranted()
componentDidMount
:
function componentDidMount()
componentWillUnmount
:
function componentWillUnmount()
constructor
:
function Camera(props)
getSupportedRatiosAsync
:
function getSupportedRatiosAsync()
isMounted
:
(...)
isRecording
:
function isRecording()
pausePreview
:
function pausePreview()
recordAsync
:
function recordAsync(options)
refreshAuthorizationStatus
:
function refreshAuthorizationStatus()
render
:
function render()
replaceState
:
(...)
resumePreview
:
function resumePreview()
stopRecording
:
function stopRecording()
takePictureAsync
:
function takePictureAsync(options)
_convertNativeProps
:
function _convertNativeProps(_ref9)
_convertProp
:
function _convertProp(value, key)

cristianoccazinsp, 2019-10-09:
Silly question, but can you check the file under node_modules/react-native-camera/src/RNCamera.js for the getCameraIdsAsync function definition?

Any chance your node_modules file didn't update properly? That error really tells me the function is not defined.

ReSword, 2019-10-09:
I can find the getCameraIdsAsync function ,

  async getCameraIdsAsync() {
    if (Platform.OS === 'android') {
      return await CameraManager.getCameraIds(this._cameraHandle);
    } else {
      return await CameraManager.getCameraIds(); // iOS does not need a camera instance
    }
  }

cristianoccazinsp, 2019-10-09:
There's either a case typo somewhere, or something cached. The function is clearly there, so why would it complain that it is not?

ReSword, 2019-10-09:
I examined the spelling carefully. and I find sometime it's work , but most of the time it was unsuccessful........

cristianoccazinsp, 2019-10-09:
This is probably not the issue since you said takePictureAsync works, but you most likely need to get access to the camera first before invoking this method.  Have you tried calling it on a callback assigned to onStatusChange while also checking that the status is == "READY" ?

The RNCamera internal implementation only sets the internal camera handle (required for Android) once the component successfully mounts. 

This would also bring an interesting point, there should probably be a camera handle check on the JS side before attempting to invoke it.

ReSword, 2019-10-09:
When I use this  onStatusChange prop to check camera status,  It return  Object {cameraStatus: "READY", recordAudioPermissionStatus: "AUTHORIZED"}......

cristianoccazinsp, 2019-10-09:
I meant, make sure you are calling the function once you're camera is ready and not before. I've been testing and I can't find a way to get the same error. Running out of ideas.

ReSword, 2019-10-09:
OK ，I'll try again. Thank you very much.

ReSword, 2019-10-09:
can you give  the  sample code ? Thanks

cristianoccazinsp, 2019-10-09:
Sure, here's a sample made to query the list of cameras and render a button to loop through all of them. It is just a sample so it won't compile / be perfect. Should give you an idea.

```
<RNCamera
  ref={ref => {
    this.camera = ref;
  }}
  cameraId={this.state.cameraId}
  onStatusChange={this.onCameraStatusChange}
>
    <Button onPress={this.loopCamera}><Text>Change Me!</Text></Button>
</RNCamera>

const BACK_TYPE = RNCamera.Constants.Type.back;

onCameraStatusChange = (s) => {
    if(s.cameraStatus == 'READY'){
      this.setState({cameraReady: true}, async () => {
        let ids = [];
        let cameraId = null;
        
        // Query the list of available cameras, and set the first available back camera.
        // Also set the list of cameras so we can loop through all of them afterwards.
        try{
          ids = await this.camera.getCameraIdsAsync();
          
          if(ids.length){
            cameraId = ids[0].id;
            
            // Search for the first available back camera
            for(let c of ids){
              if(c.type == BACK_TYPE){
                cameraId = c.id;
                break;
              }
            }
          }
          
        }
        catch(err){
          console.error("Failed to get camera ids", err.message || err);
        }
        this.setState({cameraIds: ids, cameraId: cameraId});
      });
    }
    else{
      if(this.state.cameraReady){
        this.setState({cameraReady: false});
      }
    }
}

loopCamera = () => {
    let {cameraId, cameraIds} = this.state;
    
    if(cameraId != null && cameraIds.length){
      let newIdx = (cameraIds.findIndex(i => i.id == cameraId) + 1) % cameraIds.length;
      this.setState({cameraId: cameraIds[newIdx].id});
    }
    // do nothing if no camera ids
}
```

ReSword, 2019-10-09:
Thank you very much （^_^）

ReSword, 2019-10-09:
@cristianoccazinsp Thank you I get the camera ids now, but  it seems that OTG is not supported

ReSword, 2019-10-09:
@cristianoccazinsp Thank you I get the camera ids now, but it seems that external camera is not supported

cristianoccazinsp, 2019-10-09:
What was the issue with camera IDs? I'm curious. Perhaps it can be improved.

How are you testing external cameras? I don't really think this will work
with usb/wifi cameras, and it is intended to use to fine additional
internal device cameras.

ReSword, 2019-10-09:
First, I create A new App and add some simple code , find it works correctly....., 
then I delete my view part and created a simple view find it also works correctly......，
at last I re-add my view part, It's works......, 
So I really don't know the reason too, Maybe it's because my equipment is running out of power.
I charged up a little before the first success.

I use OTG to  link the  external cameras ， but I can't find the external cameras, only can find the originally cameras. Maybe I'm going to write it myself.

cristianoccazinsp, 2019-10-09:
Have you used another app to confirm the otg camera works? According to
Google docs, the camera list API should be able to list external cameras as
well, but I highly doubt it will work.

"Returns the number of physical cameras available on this device. The
return value of this method might change dynamically if the device supports
external cameras and an external camera is connected or disconnected."

https://developer.android.com/reference/android/hardware/Camera.html#getNumberOfCameras()

ReSword, 2019-10-09:
Yes I used the other native app it works
[https://github.com/jiangdongguo/AndroidUSBCamera](url)

cristianoccazinsp, 2019-10-09:
Looks like that library connects to the USB directly and uses some native code (C/C++) to achieve that feature. It would be interesting to know if the same can be achieved using the camera API.

Have you tried using the camera2 API instead (`useCamera2Api={true}`) ? It's less stable and slower, but perhaps the USB camera is detected with that one.

According to this, the new Camera API should be able to access external cameras (https://source.android.com/devices/camera/external-usb-cameras). But there are also some people that say that the only way to access them is to manually implement the USB connection.

ReSword, 2019-10-09:
No, I haven't used the ( useCamera2Api={true} ). but I don't have any equipment right now. I will try again tomorrow. thanks (=・ω・=)

cristianoccazinsp, 2019-10-09:
You can read more info here: https://stackoverflow.com/questions/57846505/accessing-a-usb-camera-using-android-camera2-api

Basically, Camera2 might support USB devices, but it is there are not many devices that support it natively.  You might have luck with newer devices.

If you make it work, it would be good motivation to move from camera1 to camera2 implementation in order to support USB devices. Sadly, Camera2 still needs some fixes to match Camera1 performance and stability.

ReSword, 2019-10-09:
Ok

ReSword, 2019-10-10:
@cristianoccazinsp I chose a device with four USB interfaces to link the external cameras direct ，find （ getCameraIdsAsync() ）can work, It doesn't even need to be used ( useCamera2Api={true} ) 

but if I use OTG to link the external cameras by using ( useCamera2Api={true} ) or no,  
（ getCameraIdsAsync() ）neither can work.

cristianoccazinsp, 2019-10-10:
@ReSword Sorry, so do USB cameras work, but OTG ones doesn't? Aren't they the same?

ReSword, 2019-10-11:
@cristianoccazinsp No ，

ReSword, 2019-10-11:
[image]
[image]

closed 2019-10-23 by ReSword

Shuo-Mr, 2020-10-24:
Does it work? I also have an external camera, which is tested with Huawei tablet (Android 6.1), but I can't get the ID of the external camera. The Android usbcamera library mentioned in the appeal can obtain the ID.

I just used (usecamera2api = {true}) and couldn't get it @ReSword