react-native-camera/react-native-camera #2525: Android random crash on Camera1 setDisplayOrientation

cristianoccazinsp, 2019-10-08:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: Android 9
- react-native-camera: 3.7.0

**Description/Current Behaviour**
Our crash reporting have just reported an unhandled exception from Camera1.setDisplayOrientation. 
The reason of this is quite unknown and random, but I guess we should have a try/catch somewhere there. See stack trace below.

In addition, the error happened after continuously rotating the device in about 15 seconds intervals.

Any ideas why would Android's API refuse to set the display orientation? I'm sorry if the error is not that good, but that's all that was reported.

**Expected Behaviour**
No Crash

**Steps to Reproduce**
Steps to reproduce are unknown.

**Additionals**

Stack trace:
```
java.lang.RuntimeException set display orientation failed 
    Camera.java:-2 android.hardware.Camera.setDisplayOrientation
    Camera1.java:684 com.google.android.cameraview.Camera1.setDisplayOrientation
    CameraView.java:128 com.google.android.cameraview.CameraView$1.onDisplayOrientationChanged
    DisplayOrientationDetector.java:114 com.google.android.cameraview.DisplayOrientationDetector.dispatchOnDisplayOrientationChanged
    DisplayOrientationDetector.java:90 com.google.android.cameraview.DisplayOrientationDetector$1.onOrientationChanged
    OrientationEventListener.java:143 android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged
    SystemSensorManager.java:833 android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent
    MessageQueue.java:-2 android.os.MessageQueue.nativePollOnce
    MessageQueue.java:326 android.os.MessageQueue.next
    Looper.java:169 android.os.Looper.loop
    ActivityThread.java:6939 android.app.ActivityThread.main
    Method.java:-2 java.lang.reflect.Method.invoke
    RuntimeInit.java:493 com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run
    ZygoteInit.java:870 com.android.internal.os.ZygoteInit.main
```

Device: Motorola Moto Z3 Play / Android 9

sibelius, 2019-10-08:
try catch could help in this case

cristianoccazinsp, 2019-10-08:
It probably will. I wonder, however, if it would leave the camera in an inconsistent state. I haven't been able to reproduce this myself, but I have a feeling there might be a race condition on the orientation detection and calls to setDisplayOrientation, and camera release (as said above by google). Perhaps setDisplayOrientation should be called on the main thread, or make it a synchronized method. Hard to tell without being able to reproduce it.

sibelius, 2019-10-08:
synchronized could help

race condition could help if you try to setDisplayOrientation again before the last one completed

how to you solve this?

cristianoccazinsp, 2019-10-08:
Would probably put a synchronized call here: https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L166 so it is also consistent with stop() below.

And another synchronized call https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L674 and here https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L674 which are the two places that might be called on a different thread.

However, I still don't know if that would fix the issue or not since we don't even know if the race condition is the cause. 

Perhaps the best option is to add a silent try/catch like the method below (https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L700) and risk having that inconsistent state.

Will try to reproduce if I have some free time to see what's causing it.

cristianoccazinsp, 2019-10-08:
Another possibly related issue, although an ANR (we know the camera might end up with an ANR as well under some circumstances). This comes from the surface changed events. When querying the user, he described the freeze happening after accidentally locking the phone. So chances are that both stop and the the surface change event happened at the same time and ended up in an ANR.

I don't know for sure how does React Native fire these events or in which threads, but might be related to similar race conditions.

ANR Log:
```
ANR  executing service com.zinspector.foregroundservice.ForegroundServiceTask 
    Camera.java:-2 android.hardware.Camera.startPreview
    Camera1.java:247 com.google.android.cameraview.Camera1.startCameraPreview
    Camera1.java:847 com.google.android.cameraview.Camera1.adjustCameraParameters
    Camera1.java:156 com.google.android.cameraview.Camera1.updateSurface
    Camera1.java:45 com.google.android.cameraview.Camera1.access$000
    Camera1.java:135 com.google.android.cameraview.Camera1$1.onSurfaceChanged
    PreviewImpl.java:56 com.google.android.cameraview.PreviewImpl.dispatchSurfaceChanged
    TextureViewPreview.java:46 com.google.android.cameraview.TextureViewPreview$1.onSurfaceTextureAvailable
    TextureView.java:390 android.view.TextureView.getTextureLayer
    TextureView.java:339 android.view.TextureView.draw
    View.java:20733 android.view.View.updateDisplayListIfDirty
    View.java:21586 android.view.View.draw
    ViewGroup.java:4551 android.view.ViewGroup.drawChild
    ViewGroup.java:4326 android.view.ViewGroup.dispatchDraw
    View.java:21863 android.view.View.draw
    View.java:20733 android.view.View.updateDisplayListIfDirty
    View.java:21586 android.view.View.draw
    ViewGroup.java:4551 android.view.ViewGroup.drawChild
    ViewGroup.java:4326 android.view.ViewGroup.dispatchDraw
    ReactViewGroup.java:684 com.facebook.react.views.view.ReactViewGroup.dispatchDraw
    View.java:20719 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ThreadedRenderer.java:725 android.view.ThreadedRenderer.updateViewTreeDisplayList
    ThreadedRenderer.java:731 android.view.ThreadedRenderer.updateRootDisplayList
    ThreadedRenderer.java:840 android.view.ThreadedRenderer.draw
    ViewRootImpl.java:3932 android.view.ViewRootImpl.draw
    ViewRootImpl.java:3706 android.view.ViewRootImpl.performDraw
    ViewRootImpl.java:3014 android.view.ViewRootImpl.performTraversals
    ViewRootImpl.java:1873 android.view.ViewRootImpl.doTraversal
    ViewRootImpl.java:8478 android.view.ViewRootImpl$TraversalRunnable.run
    Choreographer.java:949 android.view.Choreographer$CallbackRecord.run
    Choreographer.java:761 android.view.Choreographer.doCallbacks
    Choreographer.java:696 android.view.Choreographer.doFrame
    Choreographer.java:935 android.view.Choreographer$FrameDisplayEventReceiver.run
    Handler.java:873 android.os.Handler.handleCallback
    Handler.java:99 android.os.Handler.dispatchMessage
    Looper.java:214 android.os.Looper.loop
    ActivityThread.java:6990 android.app.ActivityThread.main
    Method.java:-2 java.lang.reflect.Method.invoke
    RuntimeInit.java:493 com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run
    ZygoteInit.java:1445 com.android.internal.os.ZygoteInit.main
```

cristianoccazinsp, 2019-10-21:
Giving an update here.

I haven't seen the crash due to setDisplayOrientation anymore in production. However, I'm still seeing ANR's when the camera resumes from having the phone locked.

Same device (Samsung S9, model SM-G960U), Android 9. I'm also unable to reproduce this locally, 

Here's another stack trace from the ANR:
```
ANR  Input dispatching timed out (Waiting to send non-key event because the touched window has not finished processing certain input events that were delivered to it over 500.0ms ago.  Wait queue length: 7.  Wait queue head age: 55316.3ms.) 
    Camera.java:-2 android.hardware.Camera.startPreview
    Camera1.java:251 com.google.android.cameraview.Camera1.startCameraPreview
    Camera1.java:874 com.google.android.cameraview.Camera1.adjustCameraParameters
    Camera1.java:157 com.google.android.cameraview.Camera1.updateSurface
    Camera1.java:45 com.google.android.cameraview.Camera1.access$000
    Camera1.java:135 com.google.android.cameraview.Camera1$1.onSurfaceChanged
    PreviewImpl.java:56 com.google.android.cameraview.PreviewImpl.dispatchSurfaceChanged
    TextureViewPreview.java:46 com.google.android.cameraview.TextureViewPreview$1.onSurfaceTextureAvailable
    TextureView.java:390 android.view.TextureView.getTextureLayer
    TextureView.java:339 android.view.TextureView.draw
    View.java:20733 android.view.View.updateDisplayListIfDirty
    View.java:21586 android.view.View.draw
    ViewGroup.java:4551 android.view.ViewGroup.drawChild
    ViewGroup.java:4326 android.view.ViewGroup.dispatchDraw
    View.java:21863 android.view.View.draw
    View.java:20733 android.view.View.updateDisplayListIfDirty
    View.java:21586 android.view.View.draw
    ViewGroup.java:4551 android.view.ViewGroup.drawChild
    ViewGroup.java:4326 android.view.ViewGroup.dispatchDraw
    ReactViewGroup.java:684 com.facebook.react.views.view.ReactViewGroup.dispatchDraw
    View.java:20719 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ViewGroup.java:4535 android.view.ViewGroup.recreateChildDisplayList
    ViewGroup.java:4507 android.view.ViewGroup.dispatchGetDisplayList
    View.java:20688 android.view.View.updateDisplayListIfDirty
    ThreadedRenderer.java:725 android.view.ThreadedRenderer.updateViewTreeDisplayList
    ThreadedRenderer.java:731 android.view.ThreadedRenderer.updateRootDisplayList
    ThreadedRenderer.java:840 android.view.ThreadedRenderer.draw
    ViewRootImpl.java:3932 android.view.ViewRootImpl.draw
    ViewRootImpl.java:3706 android.view.ViewRootImpl.performDraw
    ViewRootImpl.java:3014 android.view.ViewRootImpl.performTraversals
    ViewRootImpl.java:1873 android.view.ViewRootImpl.doTraversal
    ViewRootImpl.java:8478 android.view.ViewRootImpl$TraversalRunnable.run
    Choreographer.java:949 android.view.Choreographer$CallbackRecord.run
    Choreographer.java:761 android.view.Choreographer.doCallbacks
    Choreographer.java:696 android.view.Choreographer.doFrame
    Choreographer.java:935 android.view.Choreographer$FrameDisplayEventReceiver.run
    Handler.java:873 android.os.Handler.handleCallback
    Handler.java:99 android.os.Handler.dispatchMessage
    Looper.java:214 android.os.Looper.loop
    ActivityThread.java:6986 android.app.ActivityThread.main
    Method.java:-2 java.lang.reflect.Method.invoke
    RuntimeInit.java:494 com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run
    ZygoteInit.java:1445 com.android.internal.os.ZygoteInit.main
```

cristianoccazinsp, 2019-10-21:
Now here's something interesting... Host pause and resume events seem to be firing twice when the phone is locked and unlocked (not when using the home button) while being in foreground.

> 2019-10-21 12:27:38.497 E/TEST: onHostResume
2019-10-21 12:27:38.497 E/TEST: onHostResume before start
2019-10-21 12:27:38.497 E/TEST: Camera1.start called
2019-10-21 12:27:38.542 E/TEST: Camera1 before adjustCameraParameters
2019-10-21 12:27:38.543 E/TEST: Camera1 after all set internals
2019-10-21 12:27:38.544 E/TEST: Camera1 after setParameters
2019-10-21 12:27:38.544 E/TEST: Camera1 before startPreview
2019-10-21 12:27:38.562 E/TEST: Camera1 after startPreview
2019-10-21 12:27:38.562 E/TEST: Camera1.start returning
2019-10-21 12:27:38.562 E/TEST: onHostResume after start
-- Duplicate pause/resume here, on the same lock/unlock event?? --
2019-10-21 12:27:38.571 E/TEST: onHostPause
2019-10-21 12:27:38.571 E/TEST: onHostPause before stop
2019-10-21 12:27:38.571 E/TEST: Camera1.stop called
2019-10-21 12:27:40.014 E/TEST: Camera1.stop returning
2019-10-21 12:27:40.014 E/TEST: onHostPause after stop
2019-10-21 12:27:40.026 E/TEST: onHostResume
2019-10-21 12:27:40.026 E/TEST: onHostResume before start
2019-10-21 12:27:40.026 E/TEST: Camera1.start called
2019-10-21 12:27:40.059 E/TEST: Camera1 before adjustCameraParameters
2019-10-21 12:27:40.059 E/TEST: Camera1 after all set internals
2019-10-21 12:27:40.060 E/TEST: Camera1 after setParameters
2019-10-21 12:27:40.060 E/TEST: Camera1 before startPreview
2019-10-21 12:27:40.075 E/TEST: Camera1 after startPreview
2019-10-21 12:27:40.075 E/TEST: Camera1.start returning
2019-10-21 12:27:40.075 E/TEST: onHostResume after start

closed 2019-12-27 by stale[bot]