react-native-camera/react-native-camera #2527: There is a way or it's even possible to separate camera face recognition by architecture?

witalobenicio, 2019-10-09:
# Question

**To Do First**
- [X] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [X] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [X] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
There is a way or it's even possible to separate camera face recognition by architecture?

When we use `enableSeparateBuildPerCPUArchitecture` on Android, it generates 4 different builds with more reduced size.
With `react-native-camera` our .apk size is very big. It would be nice if the libs had a separate build for different architectures.

Something like this exist?

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

witalobenicio, 2019-10-21:
@sibelius maybe using `bundleRelease`  (generating AAB) already does this for us.
My .apk was 45Mb on play store after upgrade react-native-camera, then I tried to use `enableSeparateBuildPerCPUArchitecture` which reduced it to 37Mb.
This days I manage to use `bundleRelease` and upload AAB file to Google Play. This make my app goes to 18Mb to 26Mb. This is a very nice reduction.

sibelius, 2019-10-21:
can you add a documentation section about this?

how to reduce final apk size?

witalobenicio, 2019-10-21:
@sibelius 

Here is the google doc about it:
https://developer.android.com/guide/app-bundle

And this is facebook doc about it (just scroll down to bundleRelease):
https://facebook.github.io/react-native/docs/signed-apk-android

witalobenicio, 2019-10-21:
But to explain quickly:

When you use bundleRelease, it generates an AAB file, which contains all different formats, architectures, whatever your app needs to work for all devices.
When you send this AAB file to play store, they process it, and creates different pieces of an APK.
When you try to download the apk, google play knows what parts your device needs, combine them, and generates an .apk that is optimized for your phone.

closed 2019-12-23 by witalobenicio