react-native-camera/react-native-camera #2528: Error while updating property 'cameraId'

dcalixto, 2019-10-09:
After update to react-native-camera: "^3.6.0" and pass RN version 59.9 to androidX using jetifier and adding 
android.useAndroidX=true
android.enableJetifier=true 

to gradle.properties and implementation "androidx.appcompat:appcompat:1.0.0"
to android/build.gradle

and  maven { url "https://jitpack.io" }
        maven { url "https://maven.google.com" } 

to android/app/build.gradle

the cam start tl show:

Error while updating property 'cameraId' of a view managed by: RNCamera

Someone facing the same issue?

sibelius, 2019-10-09:
cc @cristianoccazinsp

cristianoccazinsp, 2019-10-09:
Is there a stack trace? Can you check Android Studio / Logcat?

Also, shouldn't androidX be for RN >= 0.60?  If you are using RN < 0.60, you need the "reverse" jetifier to go from androidX to appcompat. Correct me if I'm wrong.

Also as written in the migration helper from RN (https://react-native-community.github.io/upgrade-helper/?from=0.59.9&to=0.60.6), you shouldn't be adding appcompat at all if using RN >= 0.60

gaykov, 2019-10-28:
I see the same crashes from time to time in Crashlytics. I wasn't able to reproduce them locally, but I noticed that they happen only on Android 4 and only on Samsung devices (maybe it will help to get some clue).

I'm on React Native `0.61.2` and `react-native-camera` is `3.8.0`.

I'm also able to attach a stack trace:

```
Fatal Exception: com.facebook.react.bridge.JSApplicationIllegalArgumentException: Error while updating property 'cameraId' of a view managed by: RNCamera
       at com.facebook.react.uimanager.ViewManagersPropertyCache$PropSetter.updateViewProp(ViewManagersPropertyCache.java:98)
       at com.facebook.react.uimanager.ViewManagerPropertyUpdater$FallbackViewManagerSetter.setProperty(ViewManagerPropertyUpdater.java:134)
       at com.facebook.react.uimanager.ViewManagerPropertyUpdater.updateProps(ViewManagerPropertyUpdater.java:54)
       at com.facebook.react.uimanager.ViewManager.updateProperties(ViewManager.java:48)
       at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:278)
       at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:179)
       at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:773)
       at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:888)
       at com.facebook.react.uimanager.UIViewOperationQueue.access$2200(UIViewOperationQueue.java:42)
       at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:948)
       at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:28)
       at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:174)
       at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:84)
       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:789)
       at android.view.Choreographer.doCallbacks(Choreographer.java:591)
       at android.view.Choreographer.doFrame(Choreographer.java:560)
       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:777)
       at android.os.Handler.handleCallback(Handler.java:730)
       at android.os.Handler.dispatchMessage(Handler.java:92)
       at android.os.Looper.loop(Looper.java:176)
       at android.app.ActivityThread.main(ActivityThread.java:5493)
       at java.lang.reflect.Method.invokeNative(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:525)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1225)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1041)
       at dalvik.system.NativeStart.main(NativeStart.java)

Caused by java.lang.reflect.InvocationTargetException
       at java.lang.reflect.Method.invokeNative(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:525)
       at com.facebook.react.uimanager.ViewManagersPropertyCache$PropSetter.updateViewProp(ViewManagersPropertyCache.java:83)
       at com.facebook.react.uimanager.ViewManagerPropertyUpdater$FallbackViewManagerSetter.setProperty(ViewManagerPropertyUpdater.java:134)
       at com.facebook.react.uimanager.ViewManagerPropertyUpdater.updateProps(ViewManagerPropertyUpdater.java:54)
       at com.facebook.react.uimanager.ViewManager.updateProperties(ViewManager.java:48)
       at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:278)
       at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:179)
       at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:773)
       at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:888)
       at com.facebook.react.uimanager.UIViewOperationQueue.access$2200(UIViewOperationQueue.java:42)
       at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:948)
       at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:28)
       at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:174)
       at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:84)
       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:789)
       at android.view.Choreographer.doCallbacks(Choreographer.java:591)
       at android.view.Choreographer.doFrame(Choreographer.java:560)
       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:777)
       at android.os.Handler.handleCallback(Handler.java:730)
       at android.os.Handler.dispatchMessage(Handler.java:92)
       at android.os.Looper.loop(Looper.java:176)
       at android.app.ActivityThread.main(ActivityThread.java:5493)
       at java.lang.reflect.Method.invokeNative(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:525)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1225)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1041)
       at dalvik.system.NativeStart.main(NativeStart.java)

Caused by java.lang.NoClassDefFoundError: java.util.Objects
[... cut ...]
       at android.os.Handler.dispatchMessage(Handler.java:92)
       at android.os.Looper.loop(Looper.java:176)
       at android.app.ActivityThread.main(ActivityThread.java:5493)
       at java.lang.reflect.Method.invokeNative(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:525)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1225)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1041)
       at dalvik.system.NativeStart.main(NativeStart.java)
```

cristianoccazinsp, 2019-10-28:
That stack trace seems to be first useful info provided!

`Caused by java.lang.NoClassDefFoundError: java.util.Objects`

Can you confirm this only happens on Android 4? This probably explains it.

> All static methods of java.util.Objects are available above API 19 (Android 4.4.+)

Maybe this issue is related? https://github.com/react-native-community/react-native-camera/issues/2559#issuecomment-546258380

Try checking the solution there for Android 4.

gaykov, 2019-10-28:
>Can you confirm this only happens on Android 4?

At least in my case, out of ~10k active devices, it's crashing only on Android 4.

It's interesting, why the app was compiled successfully. I'm going to try the solution of extending from `MultiDexApplication` and update on the progress here.

cristianoccazinsp, 2019-10-28:
It should definitely be crashing if you're using a target SDK < 19. This is also probably a bug since `Objects.equals` seems to be only included in SDK 19 or above.

cristianoccazinsp, 2019-10-28:
I've compiled the example app just fine, so the error will most likely not be detected at compile time, even with SDK 16, which is odd.

However, I can't seem to actually run the example app. Everything builds fine but fails to run due to unresolved modules, jest-haste-map issues, watchman issues, etc. I'm most likely doing something wrong  though.

gaykov, 2019-10-28:
I have `minSdk` set to 16 and the app compiles successfully all the time.

Do I understand correctly that the app crashes on this line?
https://github.com/react-native-community/react-native-camera/blob/7abc7c2e01b1c1bb71646d8e1ec8957fa2491062/android/src/main/java/com/google/android/cameraview/Camera1.java#L294

After some research I've found that it's possible to replace it with something like:

```Java
public static boolean objectEquals(Object o1, Object o2) {
    if (o1 == null && o2 == null) return true;
    if (o1 == null) return false;
    return o1.equals(o2);
}
```

Which makes sense for me.

cristianoccazinsp, 2019-10-28:
That line and the one below which also uses Objects.equals are most likely the issue. I have no idea why that wouldn't be detected at compile time (it's Java, it should!). Is this change something you could test with Android 4 (not 4.4)?

gbalduzzi, 2019-10-29:
Hi, I'm the author of #2559 , after fixing that issue I run into this one (with way fewer devices though)

I can confirm this is not a multiDex-related issue and that it does only happen on devices with android below 4.4.

If it is a compatibility issue, it is really very strange that the compiler does not detect it

cristianoccazinsp, 2019-12-09:
Should this be re-opened? Not sure why it was closed, but it doesn't look like we ever added the Object.equals replacement wrote by @gaykov .

closed 2019-12-10 by sibelius

n1ru4l, 2019-12-10:
:tada: This issue has been resolved in version 3.14.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v3.14.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket: