react-native-camera/react-native-camera #2529: Error on catalyst

djschilling, 2019-10-09:
# Bug Report

**Platforms**

MacOS Catalina

The library does not compile when trying to use the new build target `Mac` in XCode.

The error is:

```
node_modules/react-native-camera/ios/RN/RNCamera.m:688:31: 'setHighResolutionStillImageOutputEnabled:' is unavailable: not available on macCatalyst
```

It is not important for me that the lib works on catalyst but it should compile so that i can build the project for catalina.

Screenshot from `RNCamera.m`

[image]

djschilling, 2019-10-09:
Also `#import <AssetsLibrary/ALAssetsLibrary.h>` in `RCTCameraManager.m` gives an error:

```
node_modules/react-native-camera/ios/RCT/RCTCameraManager.m:9:9: AssetsLibrary is deprecated and is not available when building for Mac Catalyst. Consider migrating to Photos instead, or use `#if !TARGET_OS_MACCATALYST` to conditionally import this framework, first importing TargetConditionals.h if necessary.
```

cristianoccazinsp, 2019-10-09:
Have you tried following the instructions from here? https://developer.apple.com/documentation/xcode/creating_a_mac_version_of_your_ipad_app

Can you exclude the whole RNCamera library instead? Is this really the only library that doesn't compile for MacOS? Considering it is a camera library meant for iOS/ipad, there are probably a lot of caveats when building for MacOS.

djschilling, 2019-10-10:
Sure when i exclude it from my build it works.
But this is not a nice build step that i want to do everytime i want to build for the mac.

I think a first fix for this library could be something like this: https://github.com/facebook/react-native/pull/25427/files

And then just showing maybe a black screen instead of the camera, telling that it is not working on Catalina.

This would be nice.

cristianoccazinsp, 2019-10-10:
Thats a good idea. The black screen already happens for the simulator. If
you look at the code there are a bunch of macros to handle the simulator
case, so that probably can be extended to Catalina as well.

djschilling, 2019-12-09:
still an issue

djschilling, 2020-02-07:
still an issue

djschilling, 2020-04-08:
i guess still nothing happened here.

sibelius, 2020-04-08:
feel free to send a PR fixing it

closed 2020-06-14 by stale[bot]

mouhsinelonly, 2020-07-12:
any solution for this

marcosrdz, 2021-02-11:
We apply these two patches to get it to compile on Catalyst:

https://github.com/BlueWallet/BlueWallet/blob/master/scripts/maccatalystpatches/RCTCameraManager.patch
https://github.com/BlueWallet/BlueWallet/blob/master/scripts/maccatalystpatches/RNCamera.patch