react-native-camera/react-native-camera #2530: Usage of `rectOfInterest` prop to limit barcode scan area.

abdelmagied94, 2019-10-09:
Please, Could you tell me how to set `rectOfInterest` so I can limit the barcode scan area?

sibelius, 2019-10-09:
try to rebase this PR https://github.com/react-native-community/react-native-camera/pull/874

computerjazz, 2019-11-08:
I just played around with `rectOfInterest` in a barcode scanner. Seems to work, but there are a few oddities/gotchas which may or may not be bugs:

The values in the rect object you pass in range from 0.0 - 1.0, corresponding to percentage of the screen (0.5 ===  half of the screen). So, to create a rect that is half screen width/height and centered, you would pass:

```js
rectOfInterest={{ x: 0.25, y: 0.25, width: 0.5, height: 0.5 }}
```

Now, my gotcha was that the coordinate system seems to be oriented in landscape, even though my app is locked to portrait. So if my rect was:

```js
rectOfInterest={{ x: 0, y: 0, width: 0.5, height: 1.0 }}
```

What I ended up with was a barcode scanner that ignored the _bottom half_ of the screen, even though I was expecting it to ignore the _right half_. I haven't dug into the code enough to understand why this was the case.

iOS docs here: https://developer.apple.com/documentation/avfoundation/avcapturemetadataoutput/1616291-rectofinterest

abdelmagied94, 2019-11-09:
@sibelius For iOS, I fixed it by transforming the `_rectOfInterest` before setting it.
```swift
CGRect transformedRectOfInterest = [_previewLayer metadataOutputRectOfInterestForRect: _rectOfInterest];
[_metadataOutput setRectOfInterest: transformedRectOfInterest];
```

sibelius, 2019-11-09:
Care to send a pull request?

closed 2020-01-15 by stale[bot]