react-native-camera/react-native-camera #2531: Display both front and back camera in the same screen.

Quantum-35, 2019-10-10:
# Question
    Is it possible to split the android screen into 2 views where one view uses the `type` back and the other view uses the front type to display both front and back camera?

Below is a screenshot of what I was trying to implement.
[image]

sibelius, 2019-10-10:
cc @cristianoccazinsp

cristianoccazinsp, 2019-10-10:
I think the support for this depends heavily on the hardware, and most android phones do not support it. Also, if possible, most likely needs the Camera2 implementation. Camera 1 states `Your application should only have one Camera object active at a time for a particular hardware camera.` (https://developer.android.com/reference/android/hardware/Camera.html#open(int))

> The Android camera APIs generally allow multiple cameras to be used at the same time, but most devices do not have enough hardware resources to support that in practice - for example, there's often only one camera image processor shared by both cameras.
https://stackoverflow.com/questions/30377081/access-both-back-and-front-cameras-simultaneously/30444841

Nevertheless, have you tried checking what's the output from logcat/android studio? I want to believe that since the implementation always creates it's own camera instance/object for each view, multiple views should work _fine_ as long as the API and hardware support it.

cristianoccazinsp, 2019-10-10:
Testing it myself, this is the error I'm getting:

> W/CameraBase: An error occurred while connecting to camera 1: Status(-8): '8: connectHelper:1481: Too many cameras already open, cannot open camera "1"'

So the above seems to be right, you will need some hardware support for this.

closed 2019-12-16 by stale[bot]

Jahb, 2022-12-21:
Has anyone got these workings for Expo/React Native?
I think IOS and Android now support it:
https://developer.apple.com/videos/play/wwdc2019/249/
https://developer.android.com/training/camera2/multi-camera