react-native-camera/react-native-camera #2533: The camera view flickers after video start recording.

igor-lemon, 2019-10-11:
The camera view flickers after video start recording. Added `defaultVideoQuality` param, it didn't solve the issue.

cristianoccazinsp, 2019-10-11:
Is this on iOS? I've noticed this as well, although I haven't really given it too much importance. This is mostly related to the photo to video switch, and I guess why most native camera apps have two completely different previews for video and photo. Not entirely sure how this can be improved, adding a slight delay between configuration changes and record start maybe?

If this is Android, the reason is similar, but with the addition that some code runs on the UI thread so under cheap devices the "freeze" might be noticeable.

cristianoccazinsp, 2019-10-11:
If you set the prop `pictureSize` to match either the `quality` prop of `recordAsync` or `defaultVideoQuality`, you won't have a flicker when the video recording starts (except maybe the first time after a camera mount, I don't know why). However, you will end up also capturing lower quality/resolution photos.

cristianoccazinsp, 2019-10-12:
@sibelius maybe you have some more ideas about this. 

The flicker/first frame of the video being darker issue seems more obvious after one of the fixes I did that would set the photo preset back to normal after each video recording. Before that fix, the camera preset could remain at a value different than the default one (e.g., 4k), causing subsequent records to not flicker (because recording and photo would use the same preset). However, that was also leaving the camera in a bad state where switching to the front camera would then fail if it did not support that preset, and subsequent pictures would be taken with a preset that was not "Photo", possibly resulting in a different quality and even aspect ratio.

The flicker of the preview can't really be prevented if photo and video are captured with different presets, but it would be great for the flicker to not end up as part of the first frame of the recorded video. I believe the current behaviour matches as close as possible the native camera app (Photo preset for Photos, video preset for videos), but currently has this issue.

Maybe a tiny delay before the preset change to the record start call could help?

Related discussion on SO:
https://stackoverflow.com/questions/39229477/avcapturesession-captures-black-dark-frames-after-changing-preset

Related issues: 
https://github.com/react-native-community/react-native-camera/issues/1934
https://github.com/react-native-community/react-native-camera/pull/1886/files

nachumFreedman, 2019-10-15:
add this to the camera component tag made the flicker shorted. 

defaultVideoQuality={RNCamera.Constants.VideoQuality["720"]}

igor-lemon, 2019-10-15:
@nachumFreedman It's maybe strange, but with 720p screen flickering longer than with 1080p

cristianoccazinsp, 2019-10-15:
Hey @igor-lemon ,

Here's what currently happens if you record a video at let's say 720p, but your picture size (preview and photo capture) is set to the default value.
https://zinspectordev2.s3.amazonaws.com/usi/2/1571146241966052ed85284419802b8367a8f70dae.mp4

You can see the camera zooms in and flickers. This is also what happens with the native camera app and the reason you have a video and photo mode. The default "photo" quality uses a different internal resolution and quality setup which does not work for videos.

However, if you make the picture size match the video resolution (e.g., set pictureSize={'1280x720'}, iOS only), then both the preview and video recording will happen at the same resolution and there will be no switch. However, if you take a picture like this, you will end up with a 16:9 picture. Even worse, if your preview dimensions are 4:3, it will mean that you will take the original camera input in 4:3, convert it to 16:9, and then back to 4:3 on the view, losing a good bunch of width and height of what the camera captured. So you need to be extremely careful with your preview dimensions if you're taking pictures. I found this out the hard way from users using the Iphone 11 which is extremely "long" and has a very odd aspect ratio.

Video of no flickering:
https://zinspectordev2.s3.amazonaws.com/usi/2/1571145556049f172137c04aa98aff3884af8bd3c8.mp4

I think the tiny flickering is a good trade-off from the original RCT camera library that you had to manually set video or photo modes. But, if you only want videos, you can still control it with the pictureSize prop (which I guess that's why it was added there and never documented).

I wish this could be less confusing with just the defaultVideoQuality prop, but using only that would mean forcing pictures to be taken in video mode, possibly losing quality and dimensions.

cristianoccazinsp, 2019-10-15:
Adding to the above, here's the same test with RNCamera 3.6.0.

You can see that not only the video flickers, but it also captures a dark first frame, and even worse, totally messes up the photo capture preview (resolution is never adjusted back to photo capture).

Flickering on the preview and messed up preview after video recording:
https://zinspectordev2.s3.amazonaws.com/usi/2/157115147808d8e28dc49246a6a2f770fc388f7eaf.mp4

Actual recording with a black first frame:
https://zinspectordev2.s3.amazonaws.com/usi/2/1571151462d824d6ab5233497da3635b38df1ea218.mp4

I would say this update improved video recording significantly if you are trying to support both video and photo with the same preview.

calebmcelroy, 2019-11-01:
@cristianoccazinsp Thanks for the details! Adding and matching the pictureSize prop with the quality of video stopped the flickering for me.

snobear, 2019-12-03:
Hey @cristianoccazinsp I'm having the same issue on an iphone 7; haven't tested other devices yet.

I've tried matching `defaultVideoQuality` and `pictureSize` per some of this discussion, but I still get the flicker and some of the zoom.

My RNCamera setup:

```
      <RNCamera
        ref={ref => {this.camera = ref}}
        style={cameraStyles.preview}
        type={RNCamera.Constants.Type.front}
        flashMode={RNCamera.Constants.FlashMode.on}
        defaultVideoQuality={RNCamera.Constants.VideoQuality['720p']}
        pictureSize={'1280x720'}
        androidCameraPermissionOptions={{
          title: 'App needs access to your camera',
          message: 'App needs permission to use your camera',
          buttonPositive: 'OK',
          buttonNegative: 'Cancel',
        }}
        androidRecordAudioPermissionOptions={{
          title: 'App needs access to your microphone',
          message: 'App needs permission to use your microphone',
          buttonPositive: 'OK',
          buttonNegative: 'Cancel',
        }}
      />
```

Styles, fyi:

```
const cameraStyles = StyleSheet.create({
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center',
  }
});
```

and the record call looks like:

```
const { uri, codec = 'mp4' } = await this.camera.recordAsync();
```

Sounds like the preview vs. video recording modes are still rendering different sizes. Any ideas?

**Versions**
react-native-camera: 3.11.1
react-native: 0.61.4
react: 16.8.1
iOS: 13.1.3 (iphone 7 plus)

snobear, 2019-12-03:
Ah figured it out right after posting this of course 😄. I had to turn off flash mode:

```
        flashMode={RNCamera.Constants.FlashMode.off}
```

cristianoccazinsp, 2019-12-03:
> Ah figured it out right after posting this of course 😄. I had to turn off flash mode:
> 
> ```
>         flashMode={RNCamera.Constants.FlashMode.off}
> ```

That's interesting, have you tried with torch mode instead?  Flash on doesn't really do anything to video, not sure why it would cause that effect. Not even cycling through different flash modes causes a flicker.

snobear, 2019-12-05:
I actually don't need the flash/torch mode, but had it enabled for some reason, probably from a copy and paste. Not sure why it was causing the flicker, but disabling it fixed it for me 🤷‍♂

cristianoccazinsp, 2019-12-05:
Right, Torch mode does make the camera flicker. I don't really know why, but when the recording starts, the camera shuts down any flash/torch, so it has to be re-enabled, which causes the camera to flicker!

d3vhound, 2019-12-19:
Still having this issue regardless of config params.

jackiewung, 2020-01-23:
```
defaultVideoQuality={RNCamera.Constants.VideoQuality['720p']}
flashMode={RNCamera.Constants.FlashMode.off}
pictureSize={'1280x720'}
```

adding these props removed the flicker when calling `recordAsync`

RZulfikri, 2020-02-07:
any update related this issue, how to handle zoom in / out once calling `takePictureAsync` and `recordAsync` in the same screen?

ck86, 2020-03-10:
When I set the `defaultVideoQuality` and `pictureSize` to 720p (1280x720) The back camera flickers while the front camera does not on iPhone 6s and both on an iPhone XS max. 
You can see that after the flicker the preview has changed, it looks like it zoomed in.

This behaviour happens at least on 3.18.0 and 3.19.0, on 3.11.1 (which I used before) not. I didn't checked the versions in between.

ck86, 2020-03-10:
In addition: 3.16.0 where the videoStabilizationMode was integrated it flickers, too.

ck86, 2020-03-10:
Sorry for the spamming, but now I can confirm that the flickering belongs to the videoStabMode. If I turn it off then there is no flickering at all.

CTres, 2020-03-10:
Flicker is occurring on all videoStabilizationMode presets.

wolverineks, 2020-03-24:
@igor-lemon would you mind filling out the bug report template?

closed 2020-05-30 by stale[bot]

enaluz, 2020-07-15:
For anybody viewing this issue in the future, setting `keepAudioSession` to `true` fixed this for me. See [this](https://react-native-community.github.io/react-native-camera/docs/rncamera#ios-keepaudiosession) doc reference.

```
<RNCamera 
  { ...props }
  keepAudioSession
/>
```

bortolilucas, 2020-11-16:
Matching pictureSize={'1280x720'} and defaultVideoQuality={RNCamera.Constants.VideoQuality['720p']} solved the flickering on ios for me

ThanhNguyen140797, 2021-06-11:
any update? 
front camera is fine. But back camera still flickering.

robert-go, 2021-07-30:
- Use right `pictureSize` with `defaultVideoQuality`
"3840x2160" : "2160p"
"1920x1080" : "1080p"
"1280x720" : "720p"

- Don't change the video quality when start recording
```cameraRef.current?.recordAsync().then((res) => {})```