react-native-camera/react-native-camera #2537: Zooming not functional on Android(?)

Jacob-Finn, 2019-10-11:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android: Samsung Galaxy S9+
iOS: iPhone 7

**Versions**

Android: 9
iOS: 12.4
react-native-camera: 3.7.1
react-native: 0.60.5
react: 16.8.6
react-navigation: 3.11.1

**Description/Current Behaviour**
Incrementing the zoom prop of the RNCamera component via a state change doesn't change the zoom level on Android but works perfectly fine on iOS.

**Expected Behaviour**

Incrementing the zoom prop of the RNCamera should cause the zoom to change on both iOS and Android

**Steps to Reproduce**
- Assign the zoom prop to a state variable
- Increase/decrease the zoom prop in a state change
- Android will not be affected, iOS will either zoom in or zoom out.

**Additionals**

Related code snippets:
`          <RNCamera
            ref={(ref) => this.cameraRef = ref}
            zoom={zoom}
            style={{ height: Platform.OS === 'android' ? '10%' : '100%' }}
            captureAudio={false}
            flashMode={flashOn ? 'torch' : 'none'}
            onCameraReady={() => this.setState({ scanning: true })}
            onMountError={() => this.setState({ scanning: false })}
            barCodeTypes={[RNCamera.Constants.BarCodeType.code128, RNCamera.Constants.BarCodeType.pdf417]}
            onBarCodeRead={this.handleRead}
            androidCameraPermissionOptions={{
              title: 'Permission to use camera',
              message: 'We need your permission to use your camera to scan barcodes',
              buttonPositive: 'OK',
              buttonNegative: 'CANCEL',
            }}
          />
          {!userStoppedScanning
            ? (
              <>
                <TouchableWithoutFeedback onPress={this.handleZoom}>
                  <Icon
                    allowFontScaling={false}
                    style={[styles.cameraIcon, { position: 'absolute', top: 10, left: 10 }]}
                    name={zoom === 0.002 ? 'magnify-minus' : 'magnify-plus'}
                  />
                </TouchableWithoutFeedback>
                <TouchableWithoutFeedback
                  onPress={() => this.setState({ flashOn: !flashOn })}
                >
                  <Icon
                    allowFontScaling={false}
                    style={[styles.cameraIcon, { position: 'absolute', top: 10, right: 10 }]}
                    name={flashOn ? 'flashlight' : 'flashlight-off'}
                  />
                </TouchableWithoutFeedback>
              </>
            ) : null}`

`  handleZoom = () => {
    const { zoom } = this.state;
    if (zoom === 0.002) {
      this.setState({ zoom: 0 });
    } else this.setState({ zoom: zoom + 0.001 });
  }`

cristianoccazinsp, 2019-10-11:
A 0.001 zoom increment won't do much on Android. The value must be between 0 and 1, being 1 100% zoom and 0 0% zoom.
On iOS, zoom behaves very oddly (because the zoom values can be extremely high), so you will definitely see a zoom change even with 0.001.

For both iOS and Android consistency, I recommend setting the `maxZoom` prop to 8 or 16 (this is for iOS only) which is the standard digital zoom value. Then, increment your zoom value by either 0.1 or 0.01, both should definitely work depending on how "smooth" you want the zooming in to happen if you are using pinch to zoom.

Jacob-Finn, 2019-10-11:
@cristianoccazinsp Thanks for the advice! Unfortunately, in order to get a smooth experience I ended up keeping track of zoom separately depending on if the os was Android or iOS.

closed 2019-10-11 by Jacob-Finn