react-native-camera/react-native-camera #2538: Video is not recording warning

igor-lemon, 2019-10-12:
If I start to record video and after fast stop recording video I have `Video is not recording.` warning. It's strange because video recording continues. 

How to reproduce
1) Set `maxDuration`
2) Start recording video
3) Fast stop recording video

igor-lemon, 2019-10-12:
Very interesting. So I found that video start recording after ~1-1.5 sec after call `recordAsync` method. How can I catch the start recording event? I see only 1 way, it's call `isRecording` method. But
1) It method only for iOS platform
2) Need to call this method

I think good way to add some start video recording callback

cristianoccazinsp, 2019-10-12:
I wonder if moving the stop recording call to the session queue would solve the issue (https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L876) 

That would mean that stop recording gets enqueued rather than stopping immediately which might end up causing other issues.

Something like:

```
dispatch_async(self.sessionQueue, ^{
    if ([self.movieFileOutput isRecording]) {
        [self.movieFileOutput stopRecording];
    } else {
        RCTLogWarn(@"Video is not recording.");
    }
})
```

Is this something you can test? Should be pretty straight forward to update from XCode.

n1ru4l, 2019-10-14:
:tada: This issue has been resolved in version 3.7.2 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v3.7.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

igor-lemon, 2019-10-15:
@n1ru4l It didn't solve anything. The issue is still actual.

cristianoccazinsp, 2019-10-15:
Can you tell us what are you trying to do exactly? Video should definitely be stopping. I've added even further checks to stop video here (https://github.com/react-native-community/react-native-camera/pull/2542) in case you are stopping it before it even starts to record. Can you try that?

n1ru4l, 2019-10-19:
:tada: This issue has been resolved in version 3.8.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v3.8.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

igor-lemon, 2019-10-20:
LOL. I don't want to upset but nope, the issue is still actual.
[image]

cristianoccazinsp, 2019-10-20:
But does the video stop with 1 call? You will still get the message if you
call it more than once while the video setup process is happening, but it
should definitely be stopping the video.

igor-lemon, 2019-10-20:
Yes. I started to record a video and stopped after 1 sec.
I didn't do any actions with the camera during recording time.
If I record video more than 1 sec the problem does not reproduce.

cristianoccazinsp, 2019-10-20:
If it stops, then there's really nothing to worry about (it would
previously not stop, so you would need more than 1 call). The video setup
is slow ( especially if the capture preset or quality needs to be updated),
and objective C does not provide built in features to cancel enqueued tasks
(like cancelling a promise), so there's really not a lot to do about it
than waiting for the video setup to finish and then fire the cancel / error
promise.

If the message bothers you, then just check if recording is happening
before calling stop (or just prevent stopping for the first second). When
stopping, check video duration, and if it failed to record or duration is
less than what you would want, delete the file.

Also, don't test with debug mode. Debug is 10 times slower than an actual
release, so you probably also have a very slow JS component making things
worse.

igor-lemon, 2019-10-20:
I propose to implement some callback like `onStartRecordVideo` which will be called when the camera starts to record video.

cristianoccazinsp, 2019-10-20:
That would be good and not too hard to implement.

igor-lemon, 2019-10-20:
this would greatly simplify development

wawawoom, 2019-12-12:
Hello,

Same problem here with 
"react-native": "0.61.3",
"react-native-camera": "^3.9.0"

Any news ?

Thank you

cristianoccazinsp, 2020-02-07:
I've started work on this here: https://github.com/cristianoccazinsp/react-native-camera/tree/onRecordingStart

Basically, implemented `onRecordingStart` and `onRecordingEnd` callbacks. I'm waiting for some other PRs to get merged before doing a new PR for this since I had to work on top of another branch. 

@sibelius  can you review the changes from this branch before submitting the PR? I had to add the changes from an existing branch (the one that has some android changes) so I cannot make a PR yet until those changes are merged.

sibelius, 2020-02-07:
I think this 2 news events make sense, so it can update properly ui based on them

closed 2020-04-14 by stale[bot]

jaroslav009, 2021-04-22:
Still same issue. Someone has answer?

jaroslav009, 2021-04-22:
I just rebuild project and all okay, but i think in future this issue will