react-native-camera/react-native-camera #2540: App Crashed when took pictures while options contain fixOrientation: true

januarsmad, 2019-10-14:
# Bug Report

**Platforms**
Android

**Versions**

- Android: 6.0.1
- react-native-camera: 2.11.2
- react-native: 0.59.6
- react: 16.8.3

**Description/Current Behaviour**

the app always crash when i took picture while options contain fixOrientation: true.
here's the option list that i used in my project.
`const options = {
quality: this.props.backgroundImage ? 1 : 0.8,
width: this.props.backgroundImage ? 1080 : 698,
base64: true,
forceUpOrientation: true,
orientation: 'portrait',
fixOrientation: true,
mirrorImage: !this.props.cameraTypeBack,
pauseAfterCapture: true,
};`

when using ios, it works perfectly fine. 
The device i use is Samsung J5.

**Expected Behaviour**
success taking picture

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2019-10-14:
Can you upgrade to master?

januarsmad, 2019-10-14:
i already use master since i installed it. i just upgraded from 2.6.0 to 2.11.2 and tested it before post this.

chakrihacker, 2019-10-14:
Any other code changes?

Try unlink and let react-native do auto-link if you are at 0.60 or above

januarsmad, 2019-10-14:
i just found the solution. 
after checking node_modules, i found that there's another property called writeExif that accept object. So i add this to my options config, and this is what my config looks like:
`
const options = {
          width: 1080,
          base64: true,
          orientation: 'portrait',
          forceUpOrientation: true, // IOS only
          exif: true,
          writeExif: {
            Orientation: 6,
          },
          mirrorImage: !this.props.cameraTypeBack,
          pauseAfterCapture: true,
        };
`

closed 2019-10-14 by januarsmad