react-native-camera/react-native-camera #2541: Honor the `captureAudio` property on iOS

cristianoccazinsp, 2019-10-15:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android
iOS

**Versions**

- react-native-camera: 3.7.2

**Description/Current Behaviour**

PR https://github.com/react-native-community/react-native-camera/pull/2539 introduced a change that would break apps without audio recording permission on iOS. 

In order to fix this, code should correctly honor the `captureAudio` (default true) prop and do not create an audio recording connection by default if set to false.

On Android, this prop doesn't seem to be used, but doesn't seem to be needed neither since the JS code will set mute to true by default if this is set or no permission has been given; thus, preventing any crash related to permissions.

cristianoccazinsp, 2019-10-15:
Can someone give me the rationale behind only requesting audio permission if DEBUG? https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCameraManager.m#L452

Since the captureAudio prop is not even passed down to native, I'm trying to avoid it altogether by just requesting audio if there's permission. Since the camera only renders when all permissions have been checked, this should work. However, the above line seems to only request audio permission if debug, which is something I don't quite get.

Passing down the captureAudio prop might be problematic since it might not be set when the camera is initializing.

SmartArray, 2019-10-24:
@cristianoccazinsp Thanks for your contribution. Finally it works perfect on my iOS 13 device

closed 2019-12-30 by stale[bot]