react-native-camera/react-native-camera #2544: Different video file size on different phones

olinations, 2019-10-16:
I am testing video recording using an iphone 6 plus and iphone 8 plus.  I have the following recordingOptions...

````
maxDuration: 30,
videoBitrate: 2*1024*1024, // 2 Mbps
quality: RNCamera.Constants.VideoQuality['720p'],
codec: RNCamera.Constants.VideoCodec['H264'],
````

When I record for the max duration the size of the video is double on the iphone 8 than it is in the iphone 6.  For instance, the iphone 6 video comes in at 8 MB, while the iphone 8 video comes in at 18 MB.  By these standards it is likely the iPhone X and 11 are even bigger.

Is there a way to make sure the size is the same no matter what phone is being used? I know they all won't be exactly the same as it depends on what is being recorded, etc, but they should be pretty close when recording under almost identical circumstances, no?

Is there a setting I am missing or some way to achieve what I am trying to achieve?

cristianoccazinsp, 2019-10-16:
Does the same happen with the default settings, and with other qualities? Can you try not setting quality, and also with 480p and 1080p? Does not using a bitrate also result in different sizes?

cristianoccazinsp, 2019-10-16:
I was also looking at  the different recording options iOS has, perhaps bitrate is not enough to guarantee the same settings.
Please check https://developer.apple.com/documentation/avfoundation/avassetwriterinput/video_settings_dictionaries?language=objc
You could try manually adding some of those settings here https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L833 through xcode.

olinations, 2019-10-16:
> Does the same happen with the default settings, and with other qualities? Can you try not setting quality, and also with 480p and 1080p? Does not using a bitrate also result in different sizes?

It looks like the same situation if I remove the quality property, as well if I remove the default defaultVideoQuality={RNCamera.Constants.VideoQuality['720p']}. The sizes are still relatively the same 8MB and 18MP.

When I remove quality and make defaultVideoQuality={RNCamera.Constants.VideoQuality['480p']} the file size stays the same for iPhone 6 at 8 MB and it comes down to 10MB on the iPhone 8.

If I change the bitrate to 1*1024*1024 instead of 2*1024*1024 and make the default quality 720p, the iPhone 6 size gets cut in half to 4.5MB but the iPhone 8 stays at 18MB.

olinations, 2019-10-16:
I'm wondering if this is inevitable because the better the camera the higher the ppi, thus it  will be a bigger file size on better phones, even though the bitrate and resolution are the same.

cristianoccazinsp, 2019-10-16:
I would definitely try with the other native iOS properties to see if
something changes. Shouldn't be hard to test by changing the code in xcode

xHeinrich, 2019-10-16:
Try use hvec as the codec for anything higher than an iPhone 7, it's possible the iPhone 8 is ignoring the bitrate setting due to not being able to record using h264

olinations, 2019-10-16:
> Try use hvec as the codec for anything higher than an iPhone 7, it's possible the iPhone 8 is ignoring the bitrate setting due to not being able to record using h264. You should get a higher quality output with the same file size.

Awesome @xHeinrich - that appears to have worked!

xHeinrich, 2019-10-16:
@olinations this is what we use to detect if h264 or hvec should be used depending on device: https://gist.github.com/xHeinrich/6303d453277bc46b55891e138355dff2

cristianoccazinsp, 2019-10-17:
What if you always want h264?

olinations, 2019-10-17:
@xHeinrich - I was just about to get started building something like that, but way less awesome. Thanks again!

xHeinrich, 2019-10-17:
@cristianoccazinsp I'd suggest against it since hvec is superior in quality and filesize to h264 afaik.

cristianoccazinsp, 2019-10-17:
> @cristianoccazinsp I'd suggest against it since hvec is superior in quality and filesize to h264 afaik.

Issue is browser's compatibility.

closed 2019-12-23 by stale[bot]