react-native-camera/react-native-camera #2545: Uploading Video to Firebase Storage as mp4

ChristopherJohnson25, 2019-10-17:
# Question

**Ask your Question**
Does anyone have any experience uploading a video to Firebase storage as an mp4? I'm streaming videos with VideoJS and it will not play .mov. Nor will react native video stream as far as I can tell. Any ideas?

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

olinations, 2019-10-17:
In most cases you can simply change the .mov extension to the .mp4 extension . No need to convert the file because .mov is a wrapper for what is already a .mp4 so it will play just the same with the .mp4 extension.

I have done this and it seems to have worked on my devices, but I have not tested on all devices, may be worth a shot before diving into converting video formats.

ChristopherJohnson25, 2019-10-17:
Unfortunately, simply changing the extension does not work (the file still works as .mp4, but it cannot get played by Chrome or Video.JS). I suspect that Video.JS is referencing the  mime type? I'm currently working on trying to change that via blob, but having some troubles.

igor-lemon, 2019-10-21:
@ChristopherJohnson25 Yes. Just use ffmpeg library for processing videos
https://github.com/tanersener/react-native-ffmpeg

sibelius, 2019-10-21:
can someone add a section on our documentation about this?

ChristopherJohnson25, 2019-10-21:
Thanks @igor-lemon, I will look into this.

Bilal112, 2019-12-23:
!
[image]
   its done

closed 2020-02-28 by stale[bot]