react-native-camera/react-native-camera #2549: RNCamera crash on Android when Expo Unimodules installed

sezginriggs, 2019-10-17:
# Bug Report

**To Do First**

- [x ] Did you try latest release?
- [ ] Did you try master?
- [ x] Did you look for existing matching issues?

**Platforms**

Android version crashes after Unimodules installed for other Expo packages.

iOS version crashes after Unimodules installed for other Expo packages.

**Versions**

- Android: 9
- iOS: 12.4.1
- react-native-camera: 3.7.2
- react-native: 0.61.2
- react: 19.9.0
- react-navigation: 3.12.1

**Description/Current Behaviour**

After installing react-native-unimodules (0.7.0-rc.1) for other Expo libraries, RNCamera started to crash the App.

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

Igor-Boldt, 2019-10-17:
im having the same issue

sezginriggs, 2019-10-17:
it looks like this bug can be related with React Native 0.61.*

the packages below works okay for iOS (not tested on Android)

react-native-camera: 3.6.0
react-native: 0.60.5
react: 16.8.6
react-navigation: 3.12.1
react-native-unimodules: 0.7.0-rc.1

sezginriggs, 2019-10-21:
This bug seems solved for iOS on 3.8.0 but on Android it still leads to crash.

sezginriggs, 2019-11-04:
is there any news about this bug?

closed 2020-01-10 by stale[bot]

wiredmatrix, 2021-03-08:
Am still faced with same issue with android. After I installed unimodules my android app crashed with error react-native-camera.
Here's the snippet of the error message:
FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:compileDebugJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':app:debugCompileClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > Cannot choose between the following variants of project :react-native-camera:
          - generalDebugApiElements
          - mlkitDebugApiElements
        All of them match the consumer attributes:
          - Variant 'generalDebugApiElements' capability vocajam:react-native-camera:unspecified:
              - Unmatched attributes:
                  - Found com.android.build.api.attributes.VariantAttr 'generalDebug' but wasn't required.
                  - Found react-native-camera 'general' but wasn't required.
              - Compatible attributes:
                  - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
                  - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
                  - Required org.gradle.usage 'java-api' and found compatible value 'java-api'.
          - Variant 'mlkitDebugApiElements' capability vocajam:react-native-camera:unspecified:
              - Unmatched attributes:
                  - Found com.android.build.api.attributes.VariantAttr 'mlkitDebug' but wasn't required.
                  - Found react-native-camera 'mlkit' but wasn't required.
              - Compatible attributes:
                  - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
                  - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
                  - Required org.gradle.usage 'java-api' and found compatible value 'java-api'.