react-native-camera/react-native-camera #2552: Autolinking is not working on 0.61.1

oktapodia, 2019-10-18:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android 10 emulator & android 9.0 device

**Versions**

- Android: 10.0
- iOS:
- react-native-camera: 3.6.0, 3.7.2 & master
- react-native: 0.61.1
- react: 16.9.0
  

**Description/Current Behaviour**

If I am using the autolinking of the module, I am getting an error after allowing the access to camera ```requireNativeComponent: "RNCamera" was not found in the UIManager.```

This is working with a manual link `react-native link react-native-camera`

**Expected Behaviour**

The autolink should work

**Steps to Reproduce**

Using the autolinking of the react native version 0.61.1

**Additionals**

[image]

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2019-10-18:
Does it only happens with RN 0.61? Does it work with 0.60?

oktapodia, 2019-10-19:
I can't even build the app in a full new project created via `react-native init test060 --version react-native@0.60.6`

Trace:
```
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:preDebugBuild'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > Cannot choose between the following variants of project :react-native-camera:
          - generalDebugRuntimeElements
          - mlkitDebugRuntimeElements
        All of them match the consumer attributes:
          - Variant 'generalDebugRuntimeElements' capability test060:react-native-camera:unspecified:
              - Unmatched attributes:
                  - Found com.android.build.api.attributes.VariantAttr 'generalDebug' but wasn't required.
                  - Found react-native-camera 'general' but wasn't required.
              - Compatible attributes:
                  - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
                  - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
                  - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
          - Variant 'mlkitDebugRuntimeElements' capability test060:react-native-camera:unspecified:
              - Unmatched attributes:
                  - Found com.android.build.api.attributes.VariantAttr 'mlkitDebug' but wasn't required.
                  - Found react-native-camera 'mlkit' but wasn't required.
              - Compatible attributes:
                  - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
                  - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
                  - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.

```

sibelius, 2019-10-19:
Check docs about variant selection

oktapodia, 2019-10-19:
I tried in a full new project and only ran `react-native run-android` like usual, the `--variant=release` or `--variant=debug` does not work neither

sibelius, 2019-10-19:
Read the docs

Agungkes, 2019-10-21:
You need to add to your android/app/build.gradle. Read the docs :dancing_men: 

`
android {
  ...
  defaultConfig {
    ...
    missingDimensionStrategy 'react-native-camera', 'general' <-- insert this line
  }
}`

oktapodia, 2019-10-21:
I guess that the purpose of the autolinking is to avoid manual steps before using a lib in react-native, this confirm that the autolinking is not there/working isn't it?

Agungkes, 2019-10-21:
#2297

fabriziobertoglio1987, 2019-11-01:
Yes, maybe we could include [a link to the docs folder on the homepage](https://github.com/react-native-community/react-native-camera/tree/master/docs) to avoid this issues

skaptox, 2019-11-05:
> Read the docs

You are a genious, that solved the problem.

mieszko4, 2019-11-22:
@Agungkes @fabriziobertoglio1987 
In https://github.com/react-native-community/react-native-camera/blob/master/docs/installation.md#requirements adding `missingDimensionStrategy` is under `Setting up Firebase->Android` section which is confusing because firebase is optional.
For ios we have `Mostly automatic install with CocoaPods` section which explains how to add pod :+1: 

So I think we are missing here section `Mostly automatic install with Auto-link` where this `missingDimensionStrategy` should be required to set manually.

sibelius, 2019-11-22:
can somebody improve documentation to make this more explicit?

mieszko4, 2019-11-22:
I did a (minimum) proposal here https://github.com/react-native-community/react-native-camera/pull/2602

But I think the whole doc could have an improved structure by spliting:
1. using rn 60 vs previous
2. doing automatic vs manual
3. using only generic vs with firebase

sibelius, 2019-11-22:
please do

mayconmesquita, 2019-12-10:
nice, waiting for this! :)

mieszko4, 2019-12-10:
@mayconmesquita It's been already live for some time https://github.com/react-native-community/react-native-camera/blob/master/docs/installation.md
@sibelius I will close this issue.

closed 2019-12-10 by mieszko4

mortdatherobot, 2020-04-20:
I faced the same situation. 
uninstall apk 
run nxp react-native run-android 
it will works