react-native-camera/react-native-camera #2553: RNCamera 3.7.2 -- iOS crashes unless microphone permission is given even though captureAudio=false

acoutts, 2019-10-18:
# Bug Report

Platform affected: iOS

**Versions**
3.7.2

``` typescript
<RNCamera
  style={{flex: 1}}
  type={RNCamera.Constants.Type.back}
  flashMode={RNCamera.Constants.FlashMode.off}
  captureAudio={false}
  onGoogleVisionBarcodesDetected={this.checkBarcodes}>

  // ...
  
</RNCamera>
```

On iOS as soon as the RNCamera component is rendered the app crashes unless I add the following to Info.plist:

```
	<key>NSMicrophoneUsageDescription</key>
	<string>.......................</string>
```

Why does RNCamera need access to the microphone if audio is disabled? I only want to scan QR codes so there's no need to record video or audio.

cristianoccazinsp, 2019-10-19:
Can you try with 3.8.0?

acoutts, 2019-10-21:
Upgrading to 3.8.0 fixed it! Thanks.

closed 2019-10-21 by acoutts