react-native-camera/react-native-camera #2556: onFacesDetected breaks app on Android (device-only)

KittisakChitmun, 2019-10-23:
# Bug Report

RNCamera

Steps to reproduce
1. Install react-native-camera with face detection using only manual procedures 
2. Import { RNCamera, FaceDetector }
3. Add component to page with onFacesDetected callback

**Actual behaviour**
The app completely crash when using onFacesDetected but when not using it working normally.
On 'IOS' can face detect 

Environment
React Native version: 0.61.2
react-native-camera version : 3.8.0

Version: npm version

KittisakChitmun, 2019-10-23:
When remove onFacesDetected line from RNCamera I working normally

KittisakChitmun, 2019-10-23:
Repoduce
‘
react-native init test
npm install react-native-camera
‘
And follow installation doc for android step by step and then it happen this bug

sibelius, 2019-10-23:
@KittisakChitmun without a stack trace is hard to debug

KittisakChitmun, 2019-10-23:
Ok I will try to find its error and will return

KittisakChitmun, 2019-10-23:
I found error in android studio 
it happen after when callback onFacesDetected
```
FATAL EXCEPTION: AsyncTask #2
    Process: com.rncamera, PID: 15059
    java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$3.done(AsyncTask.java:365)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
        at java.util.concurrent.FutureTask.run(FutureTask.java:271)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:257)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
        at java.lang.Thread.run(Thread.java:784)
     Caused by: java.lang.IllegalStateException: Default FirebaseApp is not initialized in this process com.rncamera. Make sure to call FirebaseApp.initializeApp(Context) first.
        at com.google.firebase.FirebaseApp.getInstance(com.google.firebase:firebase-common@@16.0.2:240)
        at com.google.firebase.ml.vision.FirebaseVision.getInstance(Unknown Source:1)
        at org.reactnative.facedetector.RNFaceDetector.createFaceDetector(RNFaceDetector.java:95)
        at org.reactnative.facedetector.RNFaceDetector.getDetector(RNFaceDetector.java:46)
        at org.reactnative.camera.tasks.FaceDetectorAsyncTask.doInBackground(FaceDetectorAsyncTask.java:76)
        at org.reactnative.camera.tasks.FaceDetectorAsyncTask.doInBackground(FaceDetectorAsyncTask.java:22)
        at android.os.AsyncTask$2.call(AsyncTask.java:345)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:257) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636) 
        at java.lang.Thread.run(Thread.java:784) 
```

closed 2019-12-31 by stale[bot]