react-native-camera/react-native-camera #2558: RNCamera captureAudio prop does not work

jbburf, 2019-10-23:
# Bug Report

**To Do First**

- [ ] Did you try latest release? Yes
- [ ] Did you try master? Yes
- [ ] Did you look for existing matching issues? None Found

**Platforms**
Running iOS 13.1 on an iPhone XS

**Versions**
- Android: API 28
- iOS: 13.1
- react-native-camera: 3.8.0
- react-native:  0.60.5
- react: 16.8.6
- react-navigation: 3.13.0

**Description/Current Behavior**

When using RNCamera, the `captureAudio` prop does not work. The application crashes once I navigate to the video screen unless I add a line in the Info.plist to prompt for microphone access.

2019-10-23 10:46:01.736989-0500 xxxxxxxxxxxx[9177:2890410] [access] This app has crashed because it attempted to access privacy-sensitive data without a usage description.  The app's Info.plist must contain an NSMicrophoneUsageDescription key with a string value explaining to the user how the app uses this data.

**Expected Behaviour**

The prop should allow the camera to launch and capture images without asking for microphone access.

**Steps to Reproduce**

Build application and navigate to scan screen.

cristianoccazinsp, 2019-10-23:
I'm sure this was fixed with 3.8.0 (https://github.com/react-native-community/react-native-camera/pull/2542). Can you make sure you are using that version?

jbburf, 2019-10-25:
@cristianoccazinsp you're right I thought I was running 3.8.0 but when manually updating via yarn it fixed it.

closed 2019-10-25 by jbburf