react-native-camera/react-native-camera #2564: Cammera stutters for a second when changing route

espipj, 2019-11-01:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

Checked on iOS but might happen as well on Android

**Versions**

- Android:
- iOS: 12
- react-native-camera: 3.8.0
- react-native: 0.60.3
- react: 16.8.6

**Description/Current Behaviour**
In the Docs' recipe:
https://react-native-community.github.io/react-native-camera/docs/recipes#events-continue-if-screen-is-mounted-but-not-on-top-of-stack
it is suggested to enable/disable any of  `onFaceDetected, onGoogleVisionBarcodesDetected, onTextRecognized, onBarCodeRead ` by checking if the camera is on top of the stack (navigation -> navigation.isFocused()) and consequently switching the `handleFaceDetected`  method for null. This seems sensible to me, buy when applying it and switching back and forward between different routes coming back to the camera results in a little *stutter* and a blue mask (see #1934 ) for about 0.5sec.
In my case, this behaviour is fixed by adding the `navigation.isFocused()` check inside the handler method in an if clause.
I think docs should be updated about this matter.

espipj, 2019-11-01:
I'll open a PR over the weekend about it

cristianoccazinsp, 2019-11-02:
Stutter may be related to this? https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1243

Basically, when detecting faces, and according to that comment (I really haven't tested what happens if the comment is ignored), the camera cannot detect faces while in photo capture mode, so it has to switch its preset, causing a stutter/flicker.

If you want to avoid this fully, at the expense of capturing photos of possibly lower resolution, you may try setting the `pictureSize` prop to "High" (iOS only). I haven't tried this, but I have a feeling that doing so would prevent the camera from switching presets, and hence removing the flickering.

I would still use your suggestion though, keep the camera open in background but just ignore events if not focused. The quality difference between "High" (which is made for video), and "Photo", is quite noticeable.

closed 2020-01-08 by stale[bot]