react-native-camera/react-native-camera #2575: Autofocus didn't work on some device

anggaand123, 2019-11-11:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
Autofocus didn't work on some device
My Device is Samsung Galaxy Tab A SM-T280

[image]

**Tags**
camera, barcode, autofocus

cristianoccazinsp, 2019-11-11:
Can you connect the device and get some logs from logcat / android studio? Make sure to filter by your application and to get logs near the time when you open the camera. If there's anything wrong with setting autofocus, it will probably show there.

anggaand123, 2019-11-15:
> 11-14 20:30:06.579 12688-12688/? E/CAMERA_1::: true; Exposure: 0
11-14 20:30:06.579 12688-12688/? E/CAMERA_1::: -4
11-14 20:30:06.579 12688-12688/? E/CAMERA_1::: 4

I got this @cristianoccazinsp

anggaand123, 2019-11-15:
also when i touch the screen i got this

> E/unknown:ReactNative: Got DOWN touch before receiving UP or CANCEL from last gesture

cristianoccazinsp, 2019-11-15:
The touch event is most likely unrelated.

As for the other 3 lines of logs, are you sure there isn't more? Can you share your <RNCamera> component?

If setting the autofocus value failed for some reason, you should be seeing messages like the ones written here: https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L484

anggaand123, 2019-11-15:
im not installed version 3
i used 2.11
cause i got some error about manifest with version 3 
@cristianoccazinsp

closed 2019-11-19 by anggaand123