react-native-camera/react-native-camera #2578: Is it possible to stream real-time base64 to server?

suraneti, 2019-11-12:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
I want to stream base64 from devices camera to the server via web-socket for the detection process, so after I've tried to implement a lot of methods it seems no way to stream real-time base64 via this library?

**Tags**
question

cristianoccazinsp, 2019-11-12:
Why base64? The fact that you are getting base64 images from native to JS in real time (and for every frame?), means you will be blowing up the react bridge.

I don't think you can do this easily, especially since the frames are only available on the native side.

sibelius, 2019-11-12:
check this https://medium.com/react-native-training/build-youtube-alike-livestreams-with-react-native-8dde24adf543

cristianoccazinsp, 2019-11-12:
Looks like the "recording started" event is something that's still needed :D

sibelius, 2019-11-12:
easy to add

suraneti, 2019-11-13:
@sibelius I've followed this blog yesterday, it seems this blog present with an old version of `react-native-camera` and I stuck on this line `RNCameraViewHelper.emitRecordingStartedEvent(this, path); // <= This line is new` with an error about `this` parameter cannot convert to ViewGroup.

suraneti, 2019-11-13:
@cristianoccazinsp do you have any plan to implement `recording started` event in the future?

cristianoccazinsp, 2019-11-13:
I definitely do if someone doesn't do it before me! I have been focusing mostly on the pictures side of RNCamera since that's what have been giving me the most problems, but that's on my todo list in order to get perfect video time counters.

suraneti, 2019-11-13:
@cristianoccazinsp Thanks 😄, I hope this feature gonna merge soon and I think many people need to use this feature too.

fabriziobertoglio1987, 2019-11-13:
https://github.com/blackuy/react-native-twilio-video-webrtc

cybercoder, 2019-12-06:
Did these [patches](https://github.com/cybercoder/react-native-video-upload-demo/tree/master/streaming/frontend/patches) added ?

sibelius, 2019-12-06:
send PR

cybercoder, 2019-12-06:
> send PR

I don't know about Native programming either Android or IOS, these patch is written by : @DanielMSchmidt who wrote [this](https://medium.com/react-native-training/build-youtube-alike-livestreams-with-react-native-8dde24adf543)

iosvanyd1, 2020-04-14:
Live-stream would definitely be a good addition to the package. Rather than sending the video stream to a cache or file on the device it would be great to have access to it and send it over RTMP to a server.

cristianoccazinsp, 2020-04-14:
For me, live streaming looks like should be implemented 100% on the native side.  Recording should receive the URL, method, and headers, (and perhaps a "template for the post request format), and native should make the requests accordingly.  Seems like a lot of work!

closed 2021-01-13 by stale[bot]