react-native-camera/react-native-camera #2580: Why isn't pictureSize documented?

maxschmeling, 2019-11-13:
# Question

Is there a reason the `pictureSize` prop isn't included in the TypeScript types file (or the documentation)? I'm using it so that the camera doesn't flicker when you start recording video, but it gives a TypeScript warning because the prop doesn't exist on the `RNCamera` type.

I'm happy to submit a PR if there isn't a reason it is being left off, but since it isn't in the type or the API documentation, I'm wondering if it's intentional.

**Tags**
typescript

sibelius, 2019-11-13:
Please do

cristianoccazinsp, 2019-11-14:
I have always wondered the same. My guess would be that it is not really useful since you can control aspect ratio on android with the `ratio` prop, or the actual picture ratio on iOS by simply setting that ratio on the preview / RNCamera styles.
`pictureSize` really changes some of the internals of the camera, and actually trying to take a picture with anything but the default on iOS could lead to slightly lower quality photos 

> If you enable video stabilization by setting preferredVideoStabilizationMode to YES for any output, the high resolution still images emitted by AVCaptureStillImageOutput may be smaller by 10% or more.

Source: https://developer.apple.com/documentation/avfoundation/avcapturestillimageoutput/1616305-highresolutionstillimageoutputen?language=objc

https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1132

cristianoccazinsp, 2019-11-24:
@maxschmeling  if you do document this property, you might also might want to take a look at a possible bug on the Camera1 Android implementation.

Camera1 sets the picture size if given here (https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L417); however, it looks like it will be reseted/not used at all after any change to the camera properties in here https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L947

The above line probably needs a conditional so the currently set picture size is used if set.

closed 2020-01-31 by stale[bot]