react-native-camera/react-native-camera #2582: Flow error in RNCamera.js for View.props

James-E-Adams, 2019-11-14:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

 N/A

**Versions**

- Android: 
- iOS:
- react-native-camera:3.9.0
- react-native:0.61.2
- react:16.9
  

**Description/Current Behaviour**

[image]

Flow error on `View.props`

Run flow on a codebase using react-native-camera

Haven't tried in a basic repo but if no one else is having this problem it might just be our flowconfig. Can post it if hard to repro.

James-E-Adams, 2019-11-14:
Similarly: 
[image]

sibelius, 2019-11-14:
Can you send a pull request upgrading Flow to latest version and fixing this issues?

James-E-Adams, 2019-11-17:
Sure - I'll look into it when I have better internet to clone the repo!

NonkelDaniel, 2019-12-03:
I am also having this issue, how can we move past this?

closed 2020-02-08 by stale[bot]