react-native-camera/react-native-camera #2583: 'ratio' does not apply in type 'back' when useCamera2Api is true

robotan0921, 2019-11-14:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 9
- iOS:
- react-native-camera: 3.9.0
- react-native: 0.61.4
- react: 16.8.6
- react-navigation: 4.0.0

**Description/Current Behaviour**

I tried to get Full HD (1920x1080) picture using RNCamera.
My option is, 
* useCamera2Api={true}
* ratio={16:9}
* width: 1080
so, I expected a picture of 1920x1080 size.

But, the actual picture size is "1080x1440", it may be "4:3" ratio (maybe default).
I think useCamera2Api apply ratio option...

My code is below:
```
<RNCamera
        ref={ref => this.camera = ref }
        ratio={16:9}
        type={this.props.type}
        playSoundOnCapture={true}
        useCamera2Api={true}
        ....
```

```
const options: TakePictureOptions = {
        width: 1080,
        quality: 1.0,
        fixOrientation: true, // for Android
        orientation: 'portrait',
      };
      const data = await this.camera.takePictureAsync(options);
     ....
```

sibelius, 2019-11-14:
Can you send a pull request fixing it?

cristianoccazinsp, 2019-11-14:
Does it happen with camera1?

robotan0921, 2019-11-15:
> Does it happen with camera1?

No, it only happened with camera2api. camera1 returns expected photo size.

cristianoccazinsp, 2019-11-15:
I don't personally see anything wrong with the Camera2 code. I would suggest sticking with camera1 if this is urgent (it really works much better anyways), otherwise, try adding some debug/logs in the Java code and test it again to see why it is failing:

Some lines that might be failing:
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera2.java#L457

https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera2.java#L457

I have a feeling that if you change the ratio after the camera has mounted, it might work.

robotan0921, 2019-11-15:
cristianoccazinsp, thank you for your advice. ok, I'll use camera1 :)

closed 2020-01-21 by stale[bot]