react-native-camera/react-native-camera #2584: onFacesDetected breaks app on Android (device-only)

KittisakChitmun, 2019-11-14:
Bug Report
To Do First

[x] Did you try latest release?
[x] Did you try master?
[x] Did you look for existing matching issues?

RNCamera

Steps to reproduce

Install react-native-camera with face detection using only manual procedures
Import { RNCamera, FaceDetector }
Add component to page with onFacesDetected callback
Actual behaviour
The app completely crash when using onFacesDetected but when not using it working normally.
On 'IOS' can face detect

Environment
React Native version: 0.61.2
react-native-camera version : 3.8.0

Version: npm version

*Edit
Found this error in Android Studio

FATAL EXCEPTION: AsyncTask #2
    Process: com.rncamera, PID: 15059
    java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$3.done(AsyncTask.java:365)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
        at java.util.concurrent.FutureTask.run(FutureTask.java:271)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:257)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
        at java.lang.Thread.run(Thread.java:784)
     Caused by: java.lang.IllegalStateException: Default FirebaseApp is not initialized in this process com.rncamera. Make sure to call FirebaseApp.initializeApp(Context) first.
        at com.google.firebase.FirebaseApp.getInstance(com.google.firebase:firebase-common@@16.0.2:240)
        at com.google.firebase.ml.vision.FirebaseVision.getInstance(Unknown Source:1)
        at org.reactnative.facedetector.RNFaceDetector.createFaceDetector(RNFaceDetector.java:95)
        at org.reactnative.facedetector.RNFaceDetector.getDetector(RNFaceDetector.java:46)
        at org.reactnative.camera.tasks.FaceDetectorAsyncTask.doInBackground(FaceDetectorAsyncTask.java:76)
        at org.reactnative.camera.tasks.FaceDetectorAsyncTask.doInBackground(FaceDetectorAsyncTask.java:22)
        at android.os.AsyncTask$2.call(AsyncTask.java:345)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:257) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636) 
        at java.lang.Thread.run(Thread.java:784)

cristianoccazinsp, 2019-11-14:
`Caused by: java.lang.IllegalStateException: Default FirebaseApp is not initialized in this process com.rncamera. Make sure to call FirebaseApp.initializeApp(Context) first.`

That should tell you something. I haven't used this at all so I can't really help, but looks like a configuration issue? Did you follow all the steps correctly?

KittisakChitmun, 2019-11-19:
1. `npm install react-native-camera --save`
2. Open up `android/app/src/main/java/[...]/MainApplication.java`
 - Add `import org.reactnative.camera.RNCameraPackage;` to the imports at the top of the file
 - Add `new RNCameraPackage()` to the list returned by the `getPackages()` method. Add a comma to the previous item if there's already something there.
3. Append the following lines to `android/settings.gradle:`
```
include ':react-native-camera'
project(':react-native-camera').projectDir = new File(rootProject.projectDir, 	'../node_modules/react-native-camera/android')
```

4.Insert the following lines in `android/app/build.gradle:`
```gradle
implementation project(':react-native-camera')
```
 if using MLKit for text/face/barcode recognition:
```
android {
  ...
  defaultConfig {
    ...
    missingDimensionStrategy 'react-native-camera', 'mlkit' <-- insert this line
  }
}
```
5.Declare the permissions in your Android Manifest (required for video recording feature)
```
<uses-permission android:name="android.permission.RECORD_AUDIO"/>
<uses-permission android:name="android.permission.CAMERA" />
<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
```
6.Add jitpack to android/build.gradle
```
allprojects {
    repositories {
        maven { url "https://www.jitpack.io" }
        maven { url "https://maven.google.com" }
    }
}
```
7. Additional steps for using MLKit for text/face/barcode recognition

   7.1. Using Firebase MLKit requires seting up Firebase project for your app. If you have not already 
   added Firebase to your app, please follow the steps described in getting started guide. In short, 
   you would need to

- Register your app in Firebase console.
- Download google-services.json and place it in `android/app/`
- add the folowing to project level build.gradle:
```
buildscript {
      dependencies {
      // Add this line
      classpath 'com.google.gms:google-services:4.0.1' <-- you might want to use different version
      }
    }
```
add to the bottom of `android/app/build.gradle` file
```
apply plugin: 'com.google.gms.google-services'
```
   7.2. Configure your app to automatically download the ML model to the device after your app is 
   installed from the Play Store. If you do not enable install-time model downloads, the model will be 
   downloaded the first time you run the on-device detector. Requests you make before the 
   download has completed will produce no results.
```
<application ...>
...
  <meta-data
      android:name="com.google.firebase.ml.vision.DEPENDENCIES"
      android:value="ocr, face, barcode" />
</application>
```

KarlHeitmann, 2019-12-01:
How did you write the callback in your app? I've also had the same issue, I wrote the callback on onFacesDetected and the app crashed. This was the wrong code:

`
class OcularCam extends PureComponent {
  handleFaceDetected = (argumentos) => {
    console.log(argumentos)
  }
  ........
  render() {
    return(
    ....
        <RNCamera
          .......
          onFacesDetected={this.handleFaceDetected}
        />
    ....
    )
  }
}
`

I've decided to put an arrow function directly into the brackets of the props onFacesDetected, like this:
`
<RNCamera
  ....
  onFacesDetected={( argumento ) => {
    console.log(argumento)
    console.log(argumento.faces[0].bounds)
  }
/>
`

This way, the problem was solved.

closed 2020-02-06 by stale[bot]