react-native-camera/react-native-camera #2586: is there a way to detect if it's pop-up camera?

carlmagumpara, 2019-11-14:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2019-11-14:
Maybe such info would be better requested at https://github.com/react-native-community/react-native-device-info ? Is it even possible to query for that?

carlmagumpara, 2019-11-18:
> Maybe such info would be better requested at https://github.com/react-native-community/react-native-device-info ? Is it even possible to query for that?

just post the issue in react-native-device-info thanks

closed 2019-11-18 by carlmagumpara