react-native-camera/react-native-camera #2587: [Android] RNCamera crashes/freezes occasionally

nartc, 2019-11-14:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android:
- react-native-camera: 3.9.0
- react-native: 0.61.4
- react: 16.11.0
- react-navigation: 4.0.10

**Description/Current Behaviour**
I have an app that uses `react-navigation-tabs` (picture shown below). One of the tab displays a `RNCamera` and used as a barcode scanner. The problem I'm running into is the tab with `RNCamera` will occasionally freezes/crashes the app completely. 

I ran `adb logcat` to see what's going on but I'm not very efficient with Android native to tell so I'm going to post here.

When the `RNCamera` is working, here's what shown in the `adb logcat`
[image]
and here's the screen of a simulator:
[image]

When the `RNCamera` isn't working:
[image]
and here's the screen:
[image]

More detail:
1. The screen (1) that contains the `RNCamera` is a StackNavigator.
2. Upon mount, (1) will check for CAMERA permission and request CAMERA permission if appropriate.

[image]
`<Scanner/>` is a regular component (not a React Navigation Component) that houses the `RNCamera`. I tried leaving just the `RNCamera` there without any parent or child, the issue is still happening. 

[image]
This is the Navigation Screen that houses the `Scanner` (which in turns render `RNCamera`). 

**Expected Behaviour**
Should not crash/freeze

**Steps to Reproduce**
TBA

cristianoccazinsp, 2019-11-14:
Do you know precisely if it is an ANR or an actual crash? 

Also, can you check if the same happens when not using the barcode scanner? I have been battling with ANRs for some time already, and provided a few updates so various calls happen outside of the UI thread in order to prevent ANRs. However, I haven't been using barcodes (yet,), so I can't tell for sure if those updates are also required for barcode scanning.

Does it happen while the app is open, or when pausing/resuming?

nartc, 2019-11-14:
@cristianoccazinsp it is an ANR since ADB LogCat provides No crash stacktrace. 

1. What's the easiest way to check "when not using barcode scanner"?
2. What are your updates? to the RNCamera module?
3. It happens when the app is opened. When I pause (put the app on background) and reopen the app, it does not ANR but I got a black screen in place of the RNCamera. 

Both Device and Simulator behave the same in this scenario.

cristianoccazinsp, 2019-11-14:
1. You mentioned you are using this to scan barcodes, but I do not see any event related to bar codes, are you even using it at all? I know RNCamera internals change when you add the barcode events.
2. If you are using 3.9.0, you already have those updates. Basically, some heavy operations (camera startup) are moved to a non UI thread. I noticed that this change removed all my reported ANRs when resuming from the home screen or a locked phone.

3. So you are not getting an ANR, nor a crash, just the camera is not properly starting up. Can you try without navigation tabs? I recall someone having issues with tabs as well when losing focus. You may need to unmount and mount your RNCamera component when going to background or the tab loses focus.

nartc, 2019-11-14:
1. True. I haven't hooked up any event specific to Barcode yet since I noticed this problem so I haven't proceeded. 
3. I have followed the React Navigation tips on RNCamera Docs site (about wilFocus and willBlur). Is there any additional things that I should be aware of?

cristianoccazinsp, 2019-11-14:
I can't tell exactly since I don't use tabs navigation, but perhaps this is related? https://github.com/react-native-community/react-native-camera/issues/2564

I haven't had issues with the camera resuming from home/lock screen since 3.9.0, but yet again I do not use any tab navigation library that might interfere with the mounted components.

Can you try without tab navigator?

nartc, 2019-11-14:
Just tested with just StackNavigator. Still happens. 

I'm using `react-native-permissions` to manage CAMERA permission. Do I need to use the permissions provided by `react-native-camera`?

cristianoccazinsp, 2019-11-14:
As far as I know, RNCamera manages all the permissions for you, why do you need to double check for permissions?

Just use something like the following:
```
androidCameraPermissionOptions={{
  title: 'Permission to use camera',
  message: 'We need your permission to use your camera',
  buttonPositive: 'Ok',
  buttonNegative: 'Cancel',
}}
androidRecordAudioPermissionOptions={{
  title: 'Permission to use audio recording',
  message: 'We need your permission to use your audio',
  buttonPositive: 'Ok',
  buttonNegative: 'Cancel',
}}
onStatusChange={this.onCameraStatusChange}
pendingAuthorizationView={
  <SafeAreaView style={styles.cameraLoading}>
    <Spinner color={style.brandLight}/>
  </SafeAreaView>
}
notAuthorizedView={
  <View>
    {cameraNotAuthorized}
  </View>
}

onCameraStatusChange = (s) => {
    if(s.cameraStatus == 'READY'){
    
    }
    else{
    
    }
}
```

You definitely don't need to check permissions manually

nartc, 2019-11-14:
Got rid of `react-native-permissions` usage seems to make it a little bit more stable, I guess?! I got the Camera tab to work up to ~5+ times until it goes crazy again. (Earlier, it's been 1-2 times then it stopped working). 

`onCameraReady` never ran if it stopped working.

cristianoccazinsp, 2019-11-14:
I can't definitely make it happen here, the camera always resumes just fine when coming from background.

Are you doing anything else with the component that wraps RNCamera?

Any chance this PR might be related? https://github.com/react-native-community/react-native-camera/pull/2581

nartc, 2019-11-14:
Nothing really specific. I just wrap the RNCamera in a parent View to position it then add some SubView to the RNCamera to show a barcode Icon.

cristianoccazinsp, 2019-11-14:
If you are seeing the spinner it means the camera is waiting for authorization. If it was just the component freezing, you would be seeing the preview screen frozen, or a black screen ( I think).

nartc, 2019-11-14:
I've just turned on `resetOnBlur` (createBottomTabNavigator's option) and currently on a 20+ streaks without the Camera being frozen. `resetOnBlur` supposedly reset the state of the tab on navigating away but I don't see how it plays into "fixing" the bug. I'm just saying it's "more stable" for now so I'm going to close the Issue. 

I'll keep this one updated if I run into more problems. Thanks for the guide/help so far.

cristianoccazinsp, 2019-11-14:
Interesting, let me know if you find the actual reason of the freezes.

nartc, 2019-11-14:
@cristianoccazinsp I ran into the issue again. And looking into this PR of yours #2560 . Then I tried to downgrade `react-native-camera` down to 3.8.0 and the Camera is initialized 100% of the time. I did notice it's a little bit slower than in 3.9.0 because of the UI Thread problem but it's stable.

You think that you could play with Tab Navigation to see what's really going on? Sorry man I'm not very efficient with Java to tell if anything's wrong from your PR :(

nartc, 2019-11-14:
#2560 This PR might have had an issue in how the Camera is initialized when using with `react-navigation`.

I've tried SwitchNavigator, StackNavigator with Modal, all ran into some unstable initialization of the camera on navigation.

Current solution: downgrade `react-native-camera` to 3.8.0

cristianoccazinsp, 2019-11-14:
Interesting, can you share a small piece of code on how you're using tab navigation so I can test it?

If I had to guess, the issue might come from here: https://github.com/react-native-community/react-native-camera/pull/2560/files#diff-44cab38e4c94fe3c649ce206d577570eR488
That's what changed and moved the camera startup to the non UI thread.

nartc, 2019-11-14:
@cristianoccazinsp I'm in the middle of creating a repro actually. I can update it real quick (given I'm not distracted)

cristianoccazinsp, 2019-11-15:
Hi @nartc , any luck with providing some reproducible example? I'm using stack navigator (same version of react-navigation), and I can't seem to reproduce the issue at all. Tested on Android 8.1 and 10, both resume the camera just fine. Note that I do not pause the preview or unmount the component in any time.

nartc, 2019-11-15:
@cristianoccazinsp https://github.com/nartc/rncamera-navigation-android-test Sorry for taking so long but here's the repo. The issue isn't reliably reproducible since it's a pretty empty app. But I started noticing the issue if I add a parent around RNCamera. Take a look. It might take many tries.

cristianoccazinsp, 2019-11-15:
Thanks, I will let you know if I find something.

cristianoccazinsp, 2019-11-15:
I'm starting to think this might be some issue with different android versions. I couldn't reproduce the issue on the test device I have (Android 8.1). Is this what you're testing (see video below)?

https://zinspectordev2.s3.amazonaws.com/usi/2/1573840141d44adf79781543d0bc90d2f7734c45dd.mp4

Camera seems to resume just fine.

Can you test the example app from this PR with your device to see if you have the same issue? https://github.com/react-native-community/react-native-camera/pull/2588

AnagramEngineering, 2019-11-15:
I see a somehow similar problem on my side. The loading spinner freezes (as well as the complete app). 

I'm using: 
```
react: 16.9.0 => 16.9.0
react-native: 0.61.4 => 0.61.4
react-native-camera": "3.9.0"
```
After downgrading to version 3.8.0 the bug is gone 

"react-native-camera": "3.8.0"

I'm on Android7 but some testers mentioned the same bug on Android 9.

nartc, 2019-11-15:
@cristianoccazinsp I'll try the example app later today, if not over the weekend. I tested on my simulator as well as on my device (Android 5.1)

nartc, 2019-11-15:
@AnagramEngineering did you use React Navigation, namely TabNavigator (`createBottomTabNavigator`)? I only have this issue when used with React Navigation

AnagramEngineering, 2019-11-15:
@nartc yes, even though the when going up in the view tree there would be first a stacknavigator and then a tabnavigator.

cristianoccazinsp, 2019-11-15:
> @cristianoccazinsp I'll try the example app later today, if not over the weekend.
> 
> I tested my repro on my simulator as well as on my device (Android 5.1)
> 
> EDIT: Just watched the clip you included. Did you try going back and forth the tabs multiple times?

I can't really seem to have the issue happen with the code you provided. I don't have any device with Android 5 though.

nartc, 2019-11-15:
@AnagramEngineering Yeah. My navigation config is as follow:
```
- MainSwitch
----Login
----AuthLoading
----MainTab
--------HomeStack
--------CameraStack
----------Camera
```

@cristianoccazinsp I'll see if I could reproduce it and record.

cristianoccazinsp, 2019-11-15:
Update: Now withe android studio debugger I think I managed to get the issue to happen. Oddly, it only happened after react's auto refresh.

cristianoccazinsp, 2019-11-16:
Now this is quite interesting, and could use some extra brains figuring out why it happens!

I said above I wasn't able to reproduce the issue; however, I can get the issue to happen every time when switching from home to cameraStack after the first time if I disable yellow boxes with `console.disableYellowBox = true;`

Tracking the ANR further, the code completely stops when trying to emit the event though RN's own event system here: https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java#L172
 fired from https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraView.java#L78

It is also beyond me why would the even dispatch work the first time, but not afterwards, and only if yellow boxes are disabled.

cristianoccazinsp, 2019-11-16:
@nartc and @AnagramEngineering  can you guys try this PR? I believe I've found the issue. At least, I couldn't make the ANR happen anymore with @nartc 's code.

https://github.com/react-native-community/react-native-camera/pull/2591

nartc, 2019-11-16:
@cristianoccazinsp Damnnn man you're a beast lol. I'll give the PR a try in couple of hours. Thanks for the hard work

joshuafarray, 2019-11-16:
Not sure about the specifics, but I can vouch for having consistently experienced an ANR once every 4-6 instantiations of RNCamera using the 3.9.0 release and have experienced 0 ANRs with @cristianoccazinsp's PR.  

react-native: 0.61.4
react: 16.9.0
react-navigation: 4.0.10

nartc, 2019-11-17:
@cristianoccazinsp Looks good so far. Haven't had ANRs with with your PR. I think we can call it a fix. 
1. Jump back and forth navigation tabs many many times.
2. Bring app to background and open back up
3. Bring up multitask screen then go back to app

All 3 scenarios worked.

walidsahli, 2019-11-19:
i faced the same issue mentioned here and i don't think that i found the solution but i found a workaround, in the RNCamera.js in the render method i added that true in the if condition and i check permission when the app fires and it works as expected :) 

`render() {
    const { style, ...nativeProps } = this._convertNativeProps(this.props);

    if (this.state.isAuthorized || this.hasFaCC() || **_true_**) {
      return (
        <View style={style}>
          <RNCamera
            {...nativeProps}
            style={StyleSheet.absoluteFill}
            ref={this._setReference}
            onMountError={this._onMountError}
            onCameraReady={this._onCameraReady}
            onAudioInterrupted={this._onAudioInterrupted}
            onAudioConnected={this._onAudioConnected}
            onGoogleVisionBarcodesDetected={this._onObjectDetected(
              this.props.onGoogleVisionBarcodesDetected,
            )}
            onBarCodeRead={this._onObjectDetected(this.props.onBarCodeRead)}
            onFacesDetected={this._onObjectDetected(this.props.onFacesDetected)}
            onTextRecognized={this._onObjectDetected(this.props.onTextRecognized)}
            onPictureSaved={this._onPictureSaved}
            onSubjectAreaChanged={this._onSubjectAreaChanged}
          />
          {this.renderChildren()}
        </View>
      );`

nartc, 2019-11-19:
@walidsahli #2591 PR has been merged and it seemed to fix the issue. I'm waiting for a new release to test it :)

sibelius, 2019-11-19:
@nartc can you try to fix the CI https://circleci.com/gh/react-native-community/react-native-camera/1098?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link

I think upgrading to latest react native 61, it will solve this

nartc, 2019-11-19:
@sibelius I think it sounds like an issue with Docker being out of memory because of Java on CircleCI. 

https://medium.com/@stevewestwood/building-android-projects-on-circleci-using-their-machine-executor-b731b386a12d

nartc, 2019-11-20:
Latest release (3.11.x) does not have the changes from the PR. Just FYI

cristianoccazinsp, 2019-11-20:
Are you sure? Looks like it does: https://github.com/react-native-community/react-native-camera/blob/v3.11.1/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java#L184

Modules dispatch is there.

nartc, 2019-11-20:
I’m going to double check. I updated the package version. Reinstall. Go to Android Studio -> Clean -> Sync and Rebuild

nartc, 2019-11-21:
@cristianoccazinsp OK I lied. I didn't know how to read Java code lol. It's there. and the issue is gone. But I'm noticing some unstable behaviors. I will create new issue if the behaviors are in fact issues. For now, this issue should be closed and marked fixed by 3.11.1

closed 2019-11-21 by nartc

cristianoccazinsp, 2019-11-21:
What kind of unstable behaviors?  Did you observe the same when you tried
just the PR? Or do you think it's new?

I have been using the changes from all the PRs on Android for a week or so
without any reported crash or ANR.

nartc, 2019-11-21:
@cristianoccazinsp 
1. Put the app on multitask screen then go back to the app, the camera preview is frozen. No ANR no crash, the preview just freezes. Going to a different tab then back to the Camera tab makes it work again.
2. Put the app on multitask screen then go back to the app, the camera is just a black rectangle. No ANR no crash. Going to a different ab then back to the Camera tab WON'T make it work again. 

However, I only saw 1 incident of these two behaviors and haven't been able to reproduce it all night. I'll keep an eye out for them though. Again, remember I have a very old device :)

cristianoccazinsp, 2019-11-21:
Is the other app when multitasking also using the camera? Multitasking or
split screen is quite odd to be honest, I don't really expect it to work
100%.  The camera pauses when the activity pauses, and resumes when it
resumes, when multitasking those events are probably different.

cyeksan, 2019-11-22:
@AnagramEngineering saved my day, thank you so much. Downgrading to version 3.8.0 solved the problem.

cristianoccazinsp, 2019-11-22:
@cyeksan  can you try with the latest version? Should be fixed, and you will benefit from faster camera opening and captures.

nartc, 2019-11-22:
@cyeksan 3.11.1 should fix the problem :)

cyeksan, 2019-11-22:
Thanks a lot @cristianoccazinsp and @nartc, I upgraded react-native-camera to 3.11.1 and it works like a charm. As I understand, the problem is directly related to version 3.9.0. Thank you so much for your help.

rizvanrzayev, 2020-08-20:
> I see a somehow similar problem on my side. The loading spinner freezes (as well as the complete app).
>
> I'm using:
>
> ```
> react: 16.9.0 => 16.9.0
> react-native: 0.61.4 => 0.61.4
> react-native-camera": [...]

ureysen