react-native-camera/react-native-camera #2592: Audio device could not set inactive -[RNCamera removeAudioCaptureSessionInput]: Error Domain=NSOSStatusErrorDomain Code=560030580 "(null)"

lucastonon, 2019-11-16:
I am trying to use react-native-camera 3.9.0 together with react-native-video
on react-native 0.59.9.

So I am recording a video then displaying it on a react-native-video, and it plays for 1-2 seconds the video until it freezes and shows this warning Audio device could not set inactive -[RNCamera removeAudioCaptureSessionInput]: Error Domain=NSOSStatusErrorDomain Code=560030580 "(null)"
`

                  : videoUri ?
                    <View style={{flex:1, justifyContent: 'flex-end'}}>
                        <Video style={styles.videoTaken}
                               source={{uri: videoUri}}
                               repeat={false}
                               resizeMode={"cover"}
                               fullscreen={false}
                               controls={false}
                        />
                    </View>

                    : <RNCamera
                            ref={this.camera}
                            style={styles.camera}
                            type={cameraFacingDirection}
                            flashMode={flashMode}
                            zoom={cameraZoom}
                            maxZoom={7}
                            defaultVideoQuality={RNCamera.Constants.VideoQuality["720p"]}
                            captureAudio={true}
                            androidCameraPermissionOptions={{
                                title: 'Permission to use camera',
                                message: 'We need your permission to use your camera',
                                buttonPositive: 'Ok',
                                buttonNegative: 'Cancel',
                            }}
                            androidRecordAudioPermissionOptions={{
                                title: 'Permission to use audio recording',
                                message: 'We need your permission to use your audio',
                                buttonPositive: 'Ok',
                                buttonNegative: 'Cancel',
                            }}
                        >
                            {({ camera, status, recordAudioPermissionStatus }) => {
                                if (status !== 'READY') return <Text>Pending camera</Text>;
                                return (
                                    <CameraControls />

                                );
                            }}
                        </RNCamera>`

[image]

Does anyone have any idea?
Thank you

cristianoccazinsp, 2019-11-16:
You can't play a video with audio, and have the camera record with audio at the same time. Have you tried with `captureAudio={false}`?
You most likely need to have mounted the video player, or the camera, but not both.

lucastonon, 2019-11-16:
@cristianoccazinsp thanks for your reply.

I tried what you said and yes, it works with captureAudio={false}, but I need to capture audio and to playback with audio.

Without autoplay it works too. But I also need autoplay.

So what I tried after you pointed me this unmounting problem is to set captureAudio with a state variable. captureAudio={this.state.captureAudio}. Before I start recording, I set captureAudio to true, then when it finishes recording, I set it to false, wait 1000ms (durations less than that didn't work) then I finally set the videoUri to mount the <Video> component.

In a nutshell, I needed to re-render the camera without capturing audio and wait until the camera component releases the audio session to finally mount the video.

 `

    startRecording = async(camera: RNCamera) => {
        if (camera) {
            this.setState({
                captureAudio: true,
                recording: true
            })

            ...
            
            const data = await camera.recordAsync(options)
            this.setState({
                captureAudio: false,
                recording: false
            })
            setTimeout(() => {
                this.setState({
                    videoUri: data.uri,
                })
            }, 1000)
        }
     }

    ...
    
    <RNCamera
                captureAudio={this.state.captureAudio}

    ...`

It works, but it is a hell of an ugly solution, and for slower devices I don't believe 1000ms would be enough.
Do you have any alternative in mind?

Thank you

cristianoccazinsp, 2019-11-16:
You can try without unmounting the camera. Simply playing the video on a modal, or another screen if you are using a stack navigator.

I haven't had issues with playing video with audio while the camera is in background (on a previous screen from the stack), even with captureAudio=true.  I can't see precisely if the camera actually stops it's preview when the video starts to play or not, because it runs in background on another screen, but video definitely plays fine from start to end.

I agree that the timeout is an ugly solution, just wanted to confirm the audio from the camera was the issue.

Rather than a timeout, have you just tried with two renders (using the callback param)? Something like. Not sure if this gives time to the camera to unmount and only fires the callback after the state has fully updated the UI.
```
this.setState({showCamera: false, showVideo:false}, () => { this.setState({showVideo: true})});
```

lucastonon, 2019-11-16:
So you gave me an idea. Turned out that I didn't need to release the audio session, because it looked like when I was not rendering the camera, it started releasing the audio while the video was grabbing it and then the conflict arose.

So I just did this instead
`
                    <RNCamera
                            style={[styles.camera, {display: videoUri ? 'none' : 'flex'}]}`

I added a conditional display style prop instead of not unmounting the camera.

Thank you @cristianoccazinsp

cristianoccazinsp, 2019-11-16:
So I was right about the video and camera being able to co-exist together?

That probably has the advantage that you can resume the camera pretty much instantly after the video playback is over. Let me know if you find any more conflicts with it. Truth is, I have been fighting a lot with the audio issues on iOS, it's not just video, but phone calls, background music, speech-to-text, etc, it is all connected.

lucastonon, 2019-11-16:
> So I was right about the video and camera being able to co-exist together?
>
> That probably has the advantage that you can resume the camera pretty much instantly after the video playback is over. Let [...]

Yes, I think the problem was when attaching and detaching sessions simultaneously. But yes, they can coexist and it is a lot faster to resume it now. Working with hardware and drivers are always a pain. I don't know if I have the experience to help you with that, but let me know if I can help you.

antonmyrberg, 2019-12-03:
Hi, I'm struggling with an issue that I believe is similar to this, given that I also get the same error code. On iOS (12+), the video drops all sound after about 20 secs and removes the audio from the recording. 

I've tried the conditional display of camera, no go. Also completely removed the Video preview from my component, the sound still gets cut at around 20-30 s. Video keeps rolling and sound is there up until 10 s into the video.

Experiencing this on several devices, iOS only.

cristianoccazinsp, 2019-12-03:
Is the video recorded without sound, or just the player that fails to play it with sound?

The second error tells me you are destroying the camera component while audio is still running somewhere.

BenCrooksWeBill, 2019-12-05:
After taking a video I am navigating to another screen and then displaying the video. The same issue is occurring. The video only playing for about 2 seconds, then shows this error. 
Doesn't seem to work with captureAudio={false} set either. (also setting the audio track to null on react-native-video doesn't work)
Wondering if there is an easier fix? Moving across to another screen makes this a little more difficult to check the component unmount. Is there a way to check when the audio has loaded? 
Even a link with a clearer solution would be a great help!
Thanks

cristianoccazinsp, 2019-12-05:
@BenCrooksWeBill  so removing the audio release line fixed your issue? I would really try to find a better work around, since removing that might help your app, but cause issues to other apps (e.g., others apps won't be notified that the audio device can be acquired again).

Are you using react-navigation? If so, you shouldn't be needing to unmount RNCamera when moving to a new screen, unless you are "going back", then you could just play the video on mount/focus events?

BenCrooksWeBill, 2019-12-05:
Yes, problem solved :). 
> removing that might help your app, but cause issues to other apps
Perhaps a warning error that does not stop the video would work best. (I am not sure if this is 100% possible, only been on rn for a couple months). So it would play the vid without audio, and give a warning about the error. 

I am using react-navigation and technically i am 'going back' but it is calling the screen anew so that it reloads (reload needed for the video to take effect). You're correct in saying that I don't need to unmount the component, although what I meant was that I also cannot detect the unmount from the next screen, I dont think it is possible to pass when the component has dismounted to the next screen as i have already navigated there. I assumed because of my problem that the RNCamera component had not unmounted before showing the video, I may be mistaken. 
I just tried on the video component mounting and it has the same issue. (obviously after returning the code above)

cristianoccazinsp, 2019-12-05:
@sibelius  what are your thoughts on this? Looks like audio management is a real pain and causing issues when RNCamera is used with the video player component. Would it be a good idea to remove the audio release call altogether, or perhaps a setting/property to skip that line and a docs update?

@BenCrooksWeBill can't really think think of any other work around, the RNCamera "cleanup" code happens asynchronously so it will rarely cleanup at the same time as the unmount event happens. I also don't think it would be a good idea to add an event to know when this happens or anything, would make code quite messy.

antonmyrberg, 2019-12-05:
I would love for this to be explained as a common gotcha in the docs.
Caused me some really painful days this week :) I don't think I understand
the problem enough to explain it, otherwise I'd happily contribute to the
docs.

Thanks for all your efforts!

cristianoccazinsp, 2019-12-10:
@BenCrooksWeBill  and others that were having trouble. Can you test this PR https://github.com/react-native-community/react-native-camera/pull/2636 and set `keepAudioSession={true}` ? Should have the same effect as removing the audio release line you manually did.

BenCrooksWeBill, 2019-12-10:
@cristianoccazinsp how do i test a PR? :P

cristianoccazinsp, 2019-12-10:
> @cristianoccazinsp how do i test a PR? :P

Change react-native-camera in your packages.json to install directly from github instead of the version you're using. 
For example: `"react-native-camera": "github:cristianoccazinsp/react-native-camera#ios-audio-release"`

Not entirely sure if there's an easier way to do it :)

BenCrooksWeBill, 2019-12-10:
Thanks for the help, although, it doesn't seem to work, doing the same thing as always unfortunately

cristianoccazinsp, 2019-12-10:
So your work around was to comment the line where the line is released. This change basically adds a flag to do the same thing (skip that line). Can you double check the code is indeed doing that now? Should be working if that was the reason...

BenCrooksWeBill, 2019-12-11:
Well I deleted my node-modules and the install worked the same as the original change. So it worked fine.
Thanks again!!

rodrigowpl, 2020-08-19:
I'm having the same issue commented here:
https://github.com/react-native-community/react-native-camera/issues/2592#issuecomment-561139835

My details:
```
System:
    OS: macOS 10.15.5
    CPU: (12) x64 Intel(R) Core(TM) i7-8850H CPU @ 2.60GHz
    Memory: 1.02 GB / 16.00 GB
    Shell: 5.7.1 - /bin/zsh
  Binaries:
    Node: 12.18.2 - /usr/local/bin/node
    Yarn: 1.22.4 - ~/.yarn/bin/yarn
    npm: 6.14.5 - /usr/local/bin/npm
    Watchman: 4.9.0 - /usr/local/bin/watchman
  Managers:
    CocoaPods: 1.9.1 - /usr/local/bin/pod
  SDKs:
    iOS SDK:
      Platforms: iOS 13.6, DriverKit 19.0, macOS 10.15, tvOS 13.4, watchOS 6.2
    Android SDK:
      API Levels: 23, 25, 26, 27, 28, 29
      Build Tools: 23.0.1, 23.0.2, 25.0.0, 25.0.2, 25.0.3, 26.0.1, 26.0.2, 26.0.3, 27.0.3, 28.0.2, 28.0.3, 29.0.2
      System Images: android-25 | Google APIs Intel x86 Atom, android-26 | Google APIs Intel x86 Atom_64, android-26 | Google Play Intel x86 Atom, android-27 | Google APIs Intel x86 Atom
      Android NDK: Not Found
  IDEs:
    Android Studio: 4.0 AI-193.6911.18.40.6626763
    Xcode: 11.6/11E708 - /usr/bin/xcodebuild
  Languages:
    Java: 1.8.0_181 - /usr/bin/javac
    Python: 3.6.6 - /Users/rodrigowpl/anaconda3/bin/python
  npmPackages:
    @react-native-community/cli: Not Found
    react: ^16.13.1 => 16.13.1 
    react-native: ^0.63.2 => 0.63.2 
```

What happens:
- When record a video with more than 10, 20 seconds, the audio don't work when watching it
- The recorded video file worked outside, for example, I've uploaded to a server and play in the browser, and the audio is there, just playing in `react-native-video` isn't working

any idea?

Versions:

react-native-camera: ^3.35.0
react-native-video: ^4.4.5"

gsavvid, 2020-09-15:
This is still causing problems unfortunately. I tried the solution that was mentioned above with the conditional `captureAudio`. To be more specific, I set it to true only during recording and back to false when the recording is finished so that it doesn't conflict with the player (which is actually in another screen). This works fine the first time I open the camera to record but the second time I return to this camera screen, I get the following error as soon I start recording again:

`You attempted to set the key 'mute' with the value 'true' on an object that is meant to be immutable and has been frozen.`

Any ideas?

cristianoccazinsp, 2020-09-15:
@gsavvid that's strange.  I'm able to record with sound (no need to do the mute/unmute step) and then play video on a separate screen while the camera is still mounted. I'm also using react-native-video.  What happens if you don't use the `mute` flag? Are you using any other audio related prop?

gsavvid, 2020-09-15:
@cristianoccazinsp By `mute` flag do you mean `captureAudio` or the `mute` property of the recording options that you pass to `camera.recordAsync(recordingOptions);`? I don't use the latter at all but if I don't use `captureAudio`, then I get the warning that was mentioned in the original post:

`Audio device could not set inactive -[RNCamera removeAudioCaptureSessionInput]: Error Domain=NSOSStatusErrorDomain Code=560030580 "(null)"`.

But if I do toggle it, then I get the error I posted above 🤷‍♂️ 

 I, too, use `react-native-video` as well as `react-navigation` and as I mentioned my camera component is in a different screen than my video component.

cristianoccazinsp, 2020-09-15:
@gsavvid  I think your error is different then.  `Audio device could not set inactive` means you tried to unmount the camera right before the video player started, which makes sense since you are trying to release the app's audio while taking it with the video player.

My guess is that you are unmounting the camera when moving to the video player screen, and it happens so fast that the camera library doesn't have time to release the audio session.

You can try either of these options, without using `captureAudio` at all:
- Do not unmount the camera, just leave it running in background
- Try with `keepAudioSession`, this will make the native camera not release the audio session at all, which should reduce the chances of conflict with the video component trying to take it.

gsavvid, 2020-09-15:
@cristianoccazinsp thanks again for your reply.

> Audio device could not set inactive means you tried to unmount the camera right before the video player started, which makes sense since you are trying to release the app's audio while taking it with the video player.

That was my thought as well but I figured that setting `captureAudio` to `false` as soon as the recording finishes and way before navigating back to the player, would do the trick. But for some reason this only works the first time and every subsequent attempt leads to the error I mentioned in my first comment:

`You attempted to set the key 'mute' with the value 'true' on an object that is meant to be immutable and has been frozen.`

Back to your suggestions:

> Do not unmount the camera, just leave it running in background

I'm not sure how to achieve this really since the camera is in a different screen that will eventually get popped. In addition, I'm not sure what this means in terms of memory consumption.

> Try with keepAudioSession={true}, this will make the native camera not release the audio session at all, which should reduce the chances of conflict with the video component trying to take it.

This seems to be one step closer to the solution but there's still one problem: the very first time I navigate back to the player, the playback resumes (automatically by default) but about 1 second later it pauses. This only happens the first time - afterwards it works fine both with the camera and the player.

cristianoccazinsp, 2020-09-15:
So keeping the camera mounted is not an option since it is in the top of the stack and is unmounted when going back, got it!

With `keepAudioSession={true}`, looks like you're almost there.  I don't really know why it would only work after the first time, it could be that the video component is doing something odd with the audio session?

You may need to try with adding some delays between screen transitions to leave time for the components (both camera and video) to settle with their audio session handling.  Can you try with react-native-video again just in case?

gsavvid, 2020-09-16:
The player's not doing anything odd with the audio session as far as I can see. Adding a delay between screen transitions works but it's an ugly workaround. I'm wondering if it'd be possible to have some event from the camera that's triggered after it released all the resources.

cristianoccazinsp, 2020-09-16:
Considering all resources releasing happen during the unmount process, I don't really think it can be that simple to add such a callback.  

Have you tried unmounting the camera component first (e.g., with a conditional render), and only after it has been unmounted (i.e., on the `setState` callback) do the screen transition? Lastly, you should also only mount the video component after the screen transition has finished; that will make both the transition smooth and give the app the necessary time to clean everything up.

closed 2021-01-13 by stale[bot]

elleryfamilia, 2021-06-18:
Running into the same issue as original poster.. the proposed solutions don't seem to be working. Anyone get anywhere with this?

jonitef, 2021-09-17:
We have this issue too.

We are recording video with: 

```
            <RNCamera
              ref={props.reference}
              onCameraReady={onCameraReady}
              type={'front'}
              keepAudioSession={false}
              style={[styles.cameraStyle]}
              pictureSize={'480x640'}
              flashMode={RNCamera.Constants.FlashMode.off}
              videoStabilizationMode={RNCamera.Constants.VideoStabilization.off}
              defaultVideoQuality={RNCamera.Constants.VideoQuality['480p']}
              onRecordingStart={onRecordingStart}
              onRecordingEnd={onRecordingEnd}
            />
```

If you play music in the background with other app like Spotify and start recording video, recording does not work and will throw:
`Audio device could not set inactive: -[RNCamera removeAudioCaptureSessionInput]: Error Domain=NSOSStatusErrorDomain Code=560030580 "(null)"`
It also stops the background music.

When there is no background music, recording works perfectly.

We want the recording to stop background music, record video, when recording stops continue to play the background music.

cristianoccazinsp, 2021-09-17:
As far as I know, keepAudioSession only releases the audio when the component unmounts. The audio session is also obtained when the component mounts, but if you are getting that error, it most likely means Spotify takes exclusive access to the audio session and it cannot be “stolen”. 

If it isn’t working as above, perhaps something broke with some iOS updates and the code needs some reviewing.

jonitef, 2021-09-17:
With other libraries like react-native-video and react-native-audio-recorder-player Spotify releases the audio session and gets it back, so I don't think that's it. So I think the problem is somewhere in react-native-camera audio session handling.