react-native-camera/react-native-camera #2595: AVCaptureSessionPresetPhoto not supported during face detection. Falling back to AVCaptureSessionPresetHigh

noor94Test, 2019-11-18:
the FaceDetector.detectFacesAsync is not working on the following devices :
iPhone Xr Red / iOS 12 and iPhone XS max /iOS 13.2

const checkIfPictureContainFace = await FaceDetector.detectFacesAsync(
      data.uri
    );
error:
AVCaptureSessionPresetPhoto not supported during face detection. Falling back to AVCaptureSessionPresetHigh

sibelius, 2019-11-18:
Can u send a pull request fixing it?

noor94Test, 2019-11-18:
@sibelius I am asking for a fix for this issue.

sibelius, 2019-11-18:
let us know when you fix it

noor94Test, 2019-11-18:
We are not able to fix this issue, we are asking for someone smart to help us!!

cristianoccazinsp, 2019-11-18:
The log is just a warning/log from here https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1407

While I don't know the specific details about why that constraint is needed, it should be just a warning. 

I haven't used face detection myself so I'm not sure what's the issue. If you want to prevent the warning from happening, you could try with `pictureSize="High"` (iOS only). That would put the camera into the right capture mode (at the risk of losing photo quality) and silence the warning, but I don't think it would change much since that's what the native code already does.

Have you followed everything from the install instructions for face detection?

closed 2020-01-24 by stale[bot]

lucastonon, 2020-05-19:
Setting pictureSize="High" stops the warning but face detector still won't work

lucastonon, 2020-05-19:
> Closing this issue after a prolonged period of inactivity. Fell free to reopen this issue, if this still affecting you.
Setting pictureSize="High" stops the warning but face detector still won't work