react-native-camera/react-native-camera #2596: App crash only in android release builld

mikasaloliz, 2019-11-18:
# Bug Report

**To Do First**

- [yes] Did you try latest release?
- [yes] Did you try master?
- [yes] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:8.1
- react-native-camera:3.9.0
- react-native:61.0
here logcat
`E/AndroidRuntime: FATAL EXCEPTION: mqt_native_modules
    Process: com.packet, PID: 8955
    java.lang.NullPointerException: Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference
        at com.facebook.react.bridge.ReadableNativeArray.getDouble(ReadableNativeArray.java:91)
        at com.facebook.react.bridge.JavaMethodWrapper$4.extractArgument(JavaMethodWrapper.java:63)
        at com.facebook.react.bridge.JavaMethodWrapper$4.extractArgument(JavaMethodWrapper.java:59)
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:355)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:150)
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:26)
        at android.os.Looper.loop(Looper.java:164)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:225)
        at java.lang.Thread.run(Thread.java:764)`

cristianoccazinsp, 2019-11-18:
Can you share your <RNCamera> component?  I recall seeing this somewhere when trying to use a camera method (with ref.someMethod) before it is ready.

Are you calling any camera method before the mount/ready event is fired?

jmy10241024, 2019-11-25:
> > Can you share your component? I recall seeing this somewhere when trying to use a camera method (with ref.someMethod) before it is ready.
> > Are you calling any camera method before the mount/ready event is fired?
> 
> thank you for response but i accidentally fixed it by downgrade from react-native 0.61 to 0.60
I had the same issue, only crash on release build, work well on debug,   i use react-native 0.61.4

LaoZhangAndroid, 2019-12-16:
> Can you share your component? I recall seeing this somewhere when trying to use a camera method (with ref.someMethod) before it is ready.
> Are you calling any camera method before the mount/ready [...]

same issue

jordenchang55, 2019-12-18:
I have the same problem and I use barcode scan function only. When a barcode is read, the exception will popup.

```javascript
<RNCamera
      style={styles.preview}
      type={RNCamera.Constants.Type.back}
      captureAudio={false}
      onGoogleVisionBarcodesDetected={({barcode}) => {
        ToastAndroid.show(barcode);
      }}
/>
```
[image]

closed 2020-02-23 by stale[bot]