react-native-camera/react-native-camera #2598: After capture Image Preview is not showing.

IbrahimSheikh24, 2019-11-19:
```
I am getting Image Uri after capture image, but that uri is not binding to Image View to show preview. 
I am getting uri like below
"file:///data/user/0/com.CC.FF/cache/Camera/f96cbb26-8e4b-47b4-9339-3b02d1b24b75.jpg"

In Manifest I have added.
<uses-permission android:name="android.permission.CAMERA" />
<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

My Code is as below.
import React from 'react';
import { View, StyleSheet, Text, TouchableOpacity, Image,  } from 'react-native'
import { RNCamera } from 'react-native-camera';

export default class Camera extends React.Component {
  constructor(props) {
    super(props);
    this.state = {
      path: null,
      data: null
    }
  }

  renderCamera() {
    return (
      <RNCamera
        ref={ref => {
          this.camera = ref;
        }}
        style={styles.preview}
        type={RNCamera.Constants.Type.back}
        flashMode={RNCamera.Constants.FlashMode.on}
        captureAudio={false}
        androidCameraPermissionOptions={{
          title: 'Permission to use camera',
          message: 'We need your permission to use your camera',
          buttonPositive: 'Ok',
          buttonNegative: 'Cancel',
        }}
      >
        <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
          <TouchableOpacity
            onPress={() => {
              this.takePicture()
            }}
            style={styles.capture}>
            <Text style={{ fontSize: 14 }}> SNAP</Text>
          </TouchableOpacity>
        </View>
      </RNCamera>
    )
  }
  
  renderImage() {
    return (
      <View>
        <Image
          source={{ uri: this.state.path }}
          style={styles.preview}
        />
        <Text
          style={styles.cancel}
          onPress={() => this.setState({ path: null })}
        >Cancel
        </Text>
      </View>
    );
  }
  render() {
    return (
      <View style={styles.container}>
        {this.state.path ? this.renderImage() : this.renderCamera()}
      </View>
    );
  }
  takePicture = async () => {
    try {
      const options = { quality: 0.5, base64: true, pauseAfterCapture:true };
      const data = await this.camera.takePictureAsync(options);
        this.setState({ path: data.uri, data : data });
        console.log('Path to image: ' + data.uri);
      CameraRoll.saveToCameraRoll(data.uri,"photo")
    } catch (err) {
      console.log('err: ', err);
    }
  };
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    flexDirection: 'column',
    backgroundColor: 'blue',
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center',
  },
  capture: {
    flex: 0,
    backgroundColor: '#fff',
    borderRadius: 5,
    padding: 15,
    paddingHorizontal: 20,
    alignSelf: 'center',
    margin: 20,
  },
  cancel: {
    position: 'absolute',
    right: 20,
    top: 20,
    backgroundColor: 'transparent',
    color: '#FFF',
    fontWeight: '600',
    fontSize: 17,
  },
});

```

closed 2019-11-19 by IbrahimSheikh24

akanksha-agarwal93, 2021-03-10:
What was the resolution for this ?

ag185366, 2021-07-20:
I faced a similar issue, I'm showing the border to the camera wrapper when the picture was taken. As soon as the border width shifts from 0 to 1 I lost the camera preview. Instead of updating the border width, I update the border color, it works just fine, 
`
borderColor: takenPicture ? theme.gradient.style.bar.start : 'transparent', // Don't add this condition to borderWidth, if you add due to change in width pauseAfterCapture is breaking
        opacity: takenPicture ? 0.5 : 1,
`